themactep/thingino-firmware #551: 360 Camera motor control failure after firmware flashing - potential firmware bug or hardware issue?

mr-chenwu, 2025-03-09:
I'm a newbie and was thrilled to successfully flash the firmware to revive my 360 camera. However, I can't get the motor to function now. I'm wondering if this is due to a firmware issue or a hardware malfunction, and would greatly appreciate your help. Thank you.

[image]

gtxaspec, 2025-03-09:
theres a bug on the S09motor script, for now the fix right now is to edit S09motor, and comment out line 87.

like so: 
                `#set_motor_phases "v" "$gpio_motor_v"`

themactep, 2025-03-11:
```
gpio_motor_h=49 50 57 58
gpio_motor_limit_h_1=6
gpio_motor_limit_h_2=48
gpio_motor_switch=62
motor_maxstep_h=6300
motor_maxstep_v=2500
motor_speed=700
```

i don't see vertical motor pins. is it a camera with one set of pins for both directions?

gtxaspec, 2025-03-11:
Yes

oopuuu, 2025-03-14:
> Yes

May I ask what are the Tilt motor goio pins  of this 360ap1pa3 camera?

[image]

themactep, 2025-03-19:
try with the new build tomorrow, would it work out of the box?

gtxaspec, 2025-04-01:
@mr-chenwu this should be fixed, can you verify?

closed 2025-09-06 by themactep