themactep/thingino-firmware #555: exfat driver is buggy with vfat cards.

kad, 2025-03-11:
I'm observing filesystem errors on the card where timelapses/recording are happening.
This card, if moved back to Linux box, can be read/written correctly after fsck repair.

```shell
root@wc2-c712 mmcblk0p1# rm -rf wc2-8586/

rm: can't remove 'wc2-8586/timelapses/20250309': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250302': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250303': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250304': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250305': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250306': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250307': Directory not empty
rm: can't remove 'wc2-8586/timelapses/20250308': Directory not empty
root@wc2-c712 mmcblk0p1# cd wc2-8586/timelapses/20250307
root@wc2-c712 20250307# ls
ls: ./20250307T000007.jpg: No such file or directory
ls: ./20250307T000107.jpg: No such file or directory
ls: ./20250307T000207.jpg: No such file or directory
ls: ./20250307T000307.jpg: No such file or directory
ls: ./20250307T000407.jpg: No such file or directory
ls: ./20250307T000508.jpg: No such file or directory
ls: ./20250307T000607.jpg: No such file or directory
ls: ./20250307T000707.jpg: No such file or directory
ls: ./20250307T000807.jpg: No such file or directory
```

Card is 32gb, vfat formatted originally.

kad, 2025-03-12:
Potentially related: https://github.com/dorimanx/exfat-nofuse/issues/123 and https://github.com/dorimanx/exfat-nofuse/issues/112
It looks like exfat driver is taken from repository not maintained by past 7-8 years.

kad, 2025-03-14:
Tried to re-build firmware with vfat enabled, all files are readable on camera without any errors.

```patch
diff --git a/board/ingenic/xburst1/kernel/3.10/t20.generic.config b/board/ingenic/xburst1/kernel/3.10/t20.generic.config
index 91ad158fd..97f09532b 100644
--- a/board/ingenic/xburst1/kernel/3.10/t20.generic.config
+++ b/board/ingenic/xburst1/kernel/3.10/t20.generic.config
@@ -1706,8 +1706,11 @@ CONFIG_GENERIC_ACL=y
 #
 # DOS/FAT/NT Filesystems
 #
-# CONFIG_MSDOS_FS is not set
-# CONFIG_VFAT_FS is not set
+CONFIG_FAT_FS=y
+CONFIG_MSDOS_FS=y
+CONFIG_VFAT_FS=y
+CONFIG_FAT_DEFAULT_CODEPAGE=437
+CONFIG_FAT_DEFAULT_IOCHARSET="iso8859-1"
 # CONFIG_NTFS_FS is not set

 #
```

gtxaspec, 2025-03-14:
how many files are on the sd card? i am unable to reproduce locally

kad, 2025-03-14:
```
root@thingino-t20x-8586 mmcblk0p1# find . -type f | wc -l
12824
root@thingino-t20x-8586 mmcblk0p1#
```
Timelapses are enabled in my setup, with retention of 1 week. Errors started to be seen after first several runs of cleanup background tasks. With records enabled, errors started to be seen sooner.

gtxaspec, 2025-03-17:
looks like this only happens when theres a really big number of files on the card.  i'll follow up with more testing.

gtxaspec, 2025-04-13:
his should help
https://github.com/themactep/thingino-firmware/commit/d2ad530b5e9a8955588e7f6747ab08cf8ddc2167

kad, 2025-04-14:
> this should help [d2ad530](https://github.com/themactep/thingino-firmware/commit/d2ad530b5e9a8955588e7f6747ab08cf8ddc2167)

I'm still in the opinion it is better to use well maintained upstream driver from Linux kernel instead of unmaintained out-of-tree driver with no activity in past 7 years.

gtxaspec, 2025-04-23:
do you know of a newer driver that works with our kernel version? `Linux 3.10.14`.  If there is a newer driver that compiles with out kernel, I'm all for moving to it.

virmaior, 2025-04-30:
just encountered the same issue. dmesg is full of
```
[   41.611795] [EXFAT] sector_read: out of range error! (sec = 2783015996)
[   41.612147] [EXFAT] sector_read: out of range error! (sec = 935892156)
[   41.612260] [EXFAT] sector_read: out of range error! (sec = 1464412924)
[   41.612342] [EXFAT] sector_read: out of range error! (sec = 1464412924)
[   42.877944] convert_uni_to_ch: fail to use nls
[   42.877960] convert_uni_to_ch: fail to use nls
[   42.877966] convert_uni_to_ch: fail to use nls
[   42.877971] convert_uni_to_ch: fail to use nls
[   42.877977] convert_uni_to_ch: fail to use nls
[   42.877982] convert_uni_to_ch: fail to use nls
[   42.877988] convert_uni_to_ch: fail to use nls
[   42.877993] convert_uni_to_ch: fail to use nls
[   43.301441] convert_uni_to_ch: fail to use nls
[   43.301453] convert_uni_to_ch: fail to use nls
[   43.301459] convert_uni_to_ch: fail to use nls
[   43.301465] convert_uni_to_ch: fail to use nls
[   43.301470] convert_uni_to_ch: fail to use nls
[   43.301475] convert_uni_to_ch: fail to use nls
[   43.301481] convert_uni_to_ch: fail to use nls
[   43.301486] convert_uni_to_ch: fail to use nls
[   43.851078] convert_uni_to_ch: fail to use nls
[   43.851091] convert_uni_to_ch: fail to use nls
[   43.851096] convert_uni_to_ch: fail to use nls
[   43.851102] convert_uni_to_ch: fail to use nls
[   43.851108] convert_uni_to_ch: fail to use nls
```

reformatting the card seems to have worked for now.
had just seen this on a different camera and assumed the cause was the SD card flaking but this is a new SD card.

kad, 2025-04-30:
> do you know of a newer driver that works with our kernel version? `Linux 3.10.14`. If there is a newer driver that compiles with our kernel, I'm all for moving to it. I'm not aware of anyone backporting the newer samsung exfat driver to (our) older kernel version.

Just enable plain vfat driver in 3.x kernel series, as it was in the comment from March. It is good enough for even big cards (fat32 there supports up to 2tb).
If somebody interested in specifically exfat formatted cards and accept the risk - let them keep using it.
But for majority, use standard supported stable in-tree driver.