themactep/thingino-firmware #556: Onvif not working

PeterSz-RV, 2025-03-13:
Where are the config files of Onvif? The default user name is thingino, I need to change this in order to get it to work with my NVR.

Jorgi-1, 2025-03-13:
https://github.com/themactep/thingino-firmware/wiki/ONVIF

virmaior, 2025-03-13:
`onvif_config=/etc/onvif.conf`

themactep, 2025-03-13:
you cannot change the username, only password. the username is tied to other services to make onvif work.

PeterSz-RV, 2025-03-13:
It would be suuuuuper cool, if devs could incorporate editable filed to be able to change username within the control panel for onvif. A lot of cheap hardware NVR's would work with this and they usually set the username to "admin"

themactep, 2025-03-14:
the field is disabled intentionally. because we have to pin the username.

closed 2025-03-14 by themactep