themactep/thingino-firmware #560: Remote Logging - Configuring from WebUI creates wrong variable name in web.conf

Omagerrrhd, 2025-03-13:
Noticed that remote logging had stopped working recently, and did some poking around. 

649 root      1956 S    /sbin/syslogd -n -C64 -t -D
2746 root      1892 S    grep sysl

Found that in the syslog init script (/etc/init.d/S01syslog), the script references the following:

. /usr/share/common

if [ -n "$rsyslog_host" ]; then

Which uses web.conf as the init script. In web.conf, if a server is configured in the UI, the variable $rsyslog_host doesn't exist - it's named $rsyslog_ip

/snip
rsyslog_ip="IP_SET_FROM_GUI"
rsyslog_port="514"   
rsyslog_local="false"

If I change the web.conf variable name to match the variable name of $rsyslog_host used in /etc/init.d/S01syslog and restart, syslog starts up correctly and logs to the remote host:

/etc/init.d/S01syslogd restart
- Adding remote logging to IP_SET_FROM_GUI:514
Stopping syslogd...
- Killing syslogd by PID
stopped syslogd (pid 651)
Starting syslogd...

Verified running correctly:

ps -w | grep sysl
12080 root      1956 S    /sbin/syslogd -n -C64 -t -D -R IP_SET_FROM_GUI:514
12131 root      1892 S    grep sysl

I'm not sure where the best place to make the change is in the code - if S01syslog is a static file, that would probably be it so the variable name isn't overwritten at condig change time - but I'm not sure and didn't want to make a code change that breaks something else....

Omagerrrhd, 2025-03-13:
Just tested on another camera by changing the S01syslogd variable to match the web.conf variable name, and remote logging started as expected. So either workaround works

themactep, 2025-03-19:
fixed in #c080009d2c249f7d143da5cdf15509364ec1a9a7
please confirm it works.

Omagerrrhd, 2025-03-25:
Updated to latest build, confirmed that syslog starts with remote logging enabled as expected:

`  638 root      1956 S    /sbin/syslogd -n -C64 -S -t -D -R remote_syslogd_IP:514
 2685 root      1892 S    grep sys`

closed 2025-03-25 by Omagerrrhd