themactep/thingino-firmware #562: default hostname convention changed

nicolasroche898, 2025-03-14:
before the great refactoring (mid-2024),
a camera would get this default hostname:

ing-jooan-a6m-12ab.lan

with latest firmware release (2025-03-14), the default hostname has changed to:
thingino-t23n-12ab.lan

notice the old default convention had the full camera model in the hostname (good)
while the new naming convention only has a generic "t23n" uninformative model (for users), instead of the actual camera model.

it seems to be determined at compile time when /etc/os-release file is generated(?)

themactep, 2025-03-19:
restored. please check and confirm it works for you.

nicolasroche898, 2025-03-23:
tested on release master+47fca50, 2025-03-22, fixed (restored to previous hostname scheme). fixed.

closed 2025-03-23 by nicolasroche898