themactep/thingino-firmware #563: webui wipe overlay, wipes the wrong partition, wiping the entire rootfs instead

nicolasroche898, 2025-03-14:
tested on latest: (fully newly flashed, from github latest release files)
jooan_a6m_atbm6012bx
master+e9088ea, 2025-03-14 09:09:31 +0000

bug report:
web ui-> reset -> wipe overlay

expected: wipe /mtdblock2 aka overlay, about 224kB

actual (bug): seems to wipe /mtd4, about 5.8MB,

result: it wiped almost the netire flash instead of overlay, resulting in brick
need to recover using sdcard.

log (partial):
```
Erase overlay
# flash_eraseall -j /dev/mtd4

Erasing 32 Kibyte @ 0 -  0% complete. Cleanmarker written at 0.
Erasing 32 Kibyte @ 8000 -  0% complete. Cleanmarker written at 8000.
Erasing 32 Kibyte @ 10000 -  1% complete. Cleanmarker written at 10000.
Erasing 32 Kibyte @ 18000 -  1% complete. Cleanmarker written at 18000.
...
Erasing 32 Kibyte @ 568000 - 98% complete. Cleanmarker written at 568000.
Erasing 32 Kibyte @ 570000 - 99% complete. Cleanmarker written at 570000.
Erasing 32 Kibyte @ 578000 - 100% complete.
# 
After the reset is completed, you will need to reconfigure the camera. For detailed instructions, please consult the relevant article in our wiki.
```

note: this is the log printed from the webui. i didn't manually type flasherase mtd4, it was done automatically by webui.

themactep, 2025-03-17:
fixed.

closed 2025-03-17 by themactep

nicolasroche898, 2025-03-23:
tested on release master+47fca50, 2025-03-22, fixed.