themactep/thingino-firmware #564: Dollar sign in WiFi password

Vlad-Star, 2025-03-14:
I just upgraded one of my Wyze v3 cameras with "sysupgrade -f" using the following method:
```
curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade && chmod +x /sbin/sysupgrade
/sbin/sysupgrade -f
```

but following the reboot and the initial re-configuration (via Thingino-XXXX AP), after clicking Proceed button, not was not able to connect camera back to the WiFi router. Until I tried to prefix the dollar sign in my WiFi password with the backslash (i.e. "\$" instead of "$"). 

Never had this problem with the previous version, seems to be a newly introduced bug. Please fix.

**Old firmware info** (pulled from another Wyze v3 camera, installed the same version across them all):
```
ARCHITECTURE=mips
IMAGE_ID=wyze_c3_t31x_rtl
BUILD_ID="master+78b6e29, 2024-11-11 02:58:50 -0500"
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-ge08fe70f
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2024.11-git
BUILDROOT_PRETTY_NAME="Buildroot 2024.11-git"
```

**New firmware info**:
```
ARCHITECTURE=mips
IMAGE_ID=wyze_c3_t31x_rtl
BUILD_ID="master+e9088ea, 2025-03-14 08:58:21 +0000"
BUILD_TIME="2025-03-14 08:58:21 +0000"
COMMIT_ID="master+e9088ea, 2025-03-14 04:17:44 +0000"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-gb4758c95
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"
```

Vlad-Star, 2025-03-16:
Confirmed the same issue upgrading the rest of Wyze v3 cameras

themactep, 2025-03-24:
Please test with the latest release and see if it works now.

themactep, 2025-04-04:
closing as stale.

closed 2025-04-04 by themactep

Vlad-Star, 2025-04-04:
Sorry, didn't have a chance to test until now, but was not able to proceed, something else is broken
```
root@wyze-cam ~# curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrad
e/files/sysupgrade
root@wyze-cam ~# ls -l /sbin/sysupgrade
-rwxr-xr-x    1 root     root         10719 Apr  4 13:42 /sbin/sysupgrade
root@wyze-cam ~# date
Fri Apr  4 13:42:17 EDT 2025
root@wyze-cam ~# /sbin/sysupgrade -f
Run as /sbin/sysupgrade -f
/mnt/mmcblk0p1 is a mountpoint
Create /tmp/sysupgrade directory
Update self
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade to /tmp/sysupgrade/sysupgrade
####################################################################################################################################################### 100.0%
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade-stage2 to /sbin/sysupgrade-stage2
####################################################################################################################################################### 100.0%
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -f
/mnt/mmcblk0p1 is a mountpoint

                             ____ _____ ___  ____
                            / ___|_   _/ _ \|  _ \
                            \___ \ | || | | | |_) |
                             ___) || || |_| |  __/
                            |____/ |_| \___/|_|

        You have requested to perform a full firmware upgrade,
        which is a risky operation replacing the existing bootloader.

        In the event of an error, the camera will become inoperable,
        and you will need to perform a recovery restore the system.

        You have ten seconds to cancel the upgrade by pressing Ctrl-C.
        If you would like to continue now, press Enter.

            10 .. 9 .. 8 .. 7 .. 6 .. 5 .. 4 .. 3 .. 2 .. 1 .. 0
Check connection to https://github.com
Change directory to /mnt/mmcblk0p1
Upgrade from GitHub
Cannot get remote file size
Clean up temporary files.
root@wyze-cam ~#
```

themactep, 2025-04-04:
name change has happened since. run `gestalt` and select your camera, then rerun `sysupgrade`.

Vlad-Star, 2025-04-04:
Confirmed to be working as expected. Thank you!