themactep/thingino-firmware #566: No wifi AP after install on eufy

Rainz3, 2025-03-16:
I have 3 eufy C120s, I was able to successfully install thingino on one using the otg cloner method without any issues, the second one however, will not broadcast an ssid for me to connect to. I haven't tried on the third as I'm afraid it'd be useless if I couldn't get it working. I did back up the firmware but haven't had luck getting it restored. There are two versions of thingino for this camera, I've tried both with the same results. Also, after I flash thingino after a while I can press the setup button and it'll chime like it has on all other cameras when checking the IP address so I think it's flashed correctly just not broadcasting an AP. Is there anyway to modify a .conf to manually add my WiFi settings?

gtxaspec, 2025-03-16:
what is the exact model on the label of the camera?

Rainz3, 2025-03-16:
> what is the exact model on the label of the camera?

T8400X

Rainz3, 2025-03-17:
Read on discord someone having similar issue though his was a t8400 not t8400x and was asked to create a .diag file in the root folder of an SD Card, boot it for 20 seconds and pull the card. I've tried that but nothing is written to the .diag file.

themactep, 2025-03-21:
hook up an uart adapter and get a boot log

Rainz3, 2025-03-24:
I'll order a uart adapter, I did finally realize the one that worked was actually a t8400 not an t8400x. Also digging into my amazon purchase histroy the 2 t8400x arrived 2 months before the 8400. Or atleast I am assuming it did. Can you recommend a uart adapter?

themactep, 2025-03-24:
any 3.3v adapter should do. https://www.amazon.com/Atnsinc-CP2102-USB-2-0-TTL/dp/B08ZS6H9VS

pieterdevriesch, 2025-08-02:
Ok most of the diagnostic commands I would use on a normal Linux install don't work in this embedded version. 
I'll be the first to admit I'm not an expert, even if I'm pretty handy on the CLI.

I did manage to run the diagnostics [command.](https://gist.github.com/pieterdevriesch/f242ec902db1e523fa109c52c74f1a49)

pieterdevriesch, 2025-08-02:
Also I've left the UART wires connected so I'm available for additional troubleshooting. The SOC is covered in an RF shield and I'm hoping I won't need to remove it, but if it's the only way I will, I have two of these things anyway and they are basically ewaste if I can't get thINGino working on them.

themactep, 2025-08-02:
How did you flash it? For starters, try to reflash from an sd card with a new firmware as autoupdate-full.bin.

pieterdevriesch, 2025-08-02:
Ok, done. Flash succesfull but I don't see any change.
[log](https://gist.github.com/pieterdevriesch/c584db735db55267592cfb5419167967)

wltechblog, 2025-08-03:
You can burn this image to SD and pop it in the running cam, it will create a diagnostic file you can share. 

https://github.com/wltechblog/thingino-installers/tree/main/diagnostics

pieterdevriesch, 2025-08-03:
Looks like that runs the same diagnostics command I posted before but [here you go](https://gist.github.com/pieterdevriesch/31f7e6b4f6c752a829538a9684564409)

pieterdevriesch, 2025-08-03:
It did not output anything, when checking manually I see no devices in sdio:

`root@ing-eufy-t8400x-acdc sbin# ls -lah /sys/bus/sdio/devices`
`drwxr-xr-x    2 root     root           0 Mar 17 17:56 .`
`drwxr-xr-x    4 root     root           0 Mar 17 17:53 ..`

pieterdevriesch, 2025-08-03:
No luck so far. Using the SD card autoupdate-full.bin just hangs and does nothing(waited 30 minutes to be sure). Manually trying gives me this:

`root@ing-eufy-t8400x-acdc sysupgrade# sysupgrade -x fw.bin`
`Run as /sbin/sysupgrade -x fw.bin`
`Not enough free space in /tmp to download firmware!`
`/mnt/mmcblk0p1 is a mountpoint`
`Mountpoint /mnt/mmcblk0p1 is writable, using it for download.`
`Upgrade from a local file /mnt/mmcblk0p1/fw.bin`
`Found /mnt/mmcblk0p1/fw.bin`
`Skip self-update`
`Re-run script from /tmp/sysupgrade`
`Run as /tmp/sysupgrade/sysupgrade -x fw.bin`
`Not enough free space in /tmp to download firmware!`
`/mnt/mmcblk0p1 is a mountpoint`
`Mountpoint /mnt/mmcblk0p1 is writable, using it for download.`
`Upgrade from a local file /mnt/mmcblk0p1/fw.bin`
`Found /mnt/mmcblk0p1/fw.bin`
`Change directory to /mnt/mmcblk0p1`
`Upgrade from a provided file`
`/tmp/sysupgrade`
`Bootloader detected`
`File /mnt/mmcblk0p1/fw.bin (33657694) is larger than mtd7 partition (33554432)!`
`Clean up temporary files.`

themactep, 2025-08-03:
33657694 is larger than 32MB (33554432), you back up is borked. did you make it with cloner on windows?

pieterdevriesch, 2025-08-03:
I did, I've still got the individual 1MB chunks, I guess I didn't combine them correctly.

themactep, 2025-08-03:
make sure the chunks are exactly 1MB (1,048,576 bytes) each

aaronadamsCA, 2026-01-18:
Not OP but also struggling with a couple of T8400 cameras, possibly the same problem.

I took backups from both cameras, then flashed one camera with a few different firmware files; nothing worked. Then when I tried restoring its backup I got an overflow error.

It turns out most of the backup chunks from both cameras have a few extra bytes:

```
-a---l         1/17/2026   4:20 PM        1053043 fw_backup_0.bin
-a---l         1/17/2026   4:20 PM        1052576 fw_backup_1.bin
-a---l         1/17/2026   4:20 PM        1052633 fw_backup_10.bin
-a---l         1/17/2026   4:20 PM        1052724 fw_backup_11.bin
-a---l         1/17/2026   4:20 PM        1052637 fw_backup_12.bin
-a---l         1/17/2026   4:20 PM        1052705 fw_backup_13.bin
-a---l         1/17/2026   4:20 PM        1052820 fw_backup_14.bin
-a---l         1/17/2026   4:20 PM        1052850 fw_backup_15.bin
-a---l         1/17/2026   4:20 PM        1052610 fw_backup_16.bin
-a---l         1/17/2026   4:20 PM        1052707 fw_backup_17.bin
-a---l         1/17/2026   4:20 PM        1052831 fw_backup_18.bin
-a---l         1/17/2026   4:20 PM        1052656 fw_backup_19.bin
-a---l         1/17/2026   4:20 PM        1050019 fw_backup_2.bin
-a---l         1/17/2026   4:20 PM        1049590 fw_backup_20.bin
-a---l         1/17/2026   4:20 PM        1052762 fw_backup_21.bin
-a---l         1/17/2026   4:20 PM        1052730 fw_backup_22.bin
-a---l         1/17/2026   4:20 PM        1052652 fw_backup_23.bin
-a---l         1/17/2026   4:20 PM        1052782 fw_backup_24.bin
-a---l         1/17/2026   4:20 PM        1051254 fw_backup_25.bin
-a---l         1/17/2026   4:20 PM        1048576 fw_backup_26.bin
-a---l         1/17/2026   4:20 PM        1048576 fw_backup_27.bin
-a---l         1/17/2026   4:20 PM        1048576 fw_backup_28.bin
-a---l         1/17/2026   4:20 PM        1048576 fw_backup_29.bin
-a---l         1/17/2026   4:20 PM        1052720 fw_backup_3.bin
-a---l         1/17/2026   4:20 PM        1049095 fw_backup_30.bin
-a---l         1/17/2026   4:20 PM        1048723 fw_backup_31.bin
-a---l         1/17/2026   4:20 PM        1052694 fw_backup_4.bin
-a---l         1/17/2026   4:20 PM        1052834 fw_backup_5.bin
-a---l         1/17/2026   4:20 PM        1052195 fw_backup_6.bin
-a---l         1/17/2026   4:20 PM        1052694 fw_backup_7.bin
-a---l         1/17/2026   4:20 PM        1052819 fw_backup_8.bin
-a---l         1/17/2026   4:20 PM        1052671 fw_backup_9.bin
-a---l         1/17/2026   4:27 PM       33655330 fw_backup_full_combined.bin
```

The results are exactly repeatable: if I pull a new backup from the camera that still has its OEM firmware, the chunk sizes are consistent (and consistently wrong).

Since OP's backup file is also too big, I'm wondering if there's some shared root cause that's causing both of our backups and flashes to go bad. I don't know enough about the Ingenic Cloner to know which settings to try playing with, though.