themactep/thingino-firmware #570: xiaomi_mjsxj03hl_t31n_jxq03p is keeping crash every few hours

unijiang, 2025-03-19:
Build number: 

- xiaomi_mjsxj03hl_t31n_jxq03p  master+1d31073, 2025-03-17 08:57:23 +0000

Diagnostic log:

- http://tb.thingino.com/826n

- http://tb.thingino.com/uc3o

gtxaspec, 2025-04-01:
does this still happen on the latest buildS?

unijiang, 2025-04-02:
> does this still happen on the latest buildS?

I just upgraded to the latest build and will monitor it for a day or two.

unijiang, 2025-04-04:
The issue persists, the latest log info: https://tb.thingino.com/1s9r/, a crash should happen around 10:37

```
# logread
Mar 18 01:53:31 odhcp6c[1662]: fd1e:bce5:916d::3b3 preferred 43200 valid 43200
Apr  4 10:37:11 odhcp6c[1662]: Starting REQUEST transaction (timeout 4294967295s, max rc 10)
Apr  4 10:37:11 odhcp6c[1662]: Send REQUEST message (elapsed 0ms, rc 0)
Apr  4 10:37:11 odhcp6c[1662]: Got a valid REPLY after 18ms
Apr  4 10:37:11 odhcp6c[1662]: IA_NA 0001 T1 21600 T2 34560
Apr  4 10:37:11 odhcp6c[1662]: 2408:823d:4c12:2d7c::3b3 preferred 43200 valid 43200
Apr  4 10:37:11 odhcp6c[1662]: fd1e:bce5:916d::3b3 preferred 43200 valid 43200
Apr  4 10:37:11 odhcp6c[1662]: T1 21600s, T2 34560s, T3 43200s
Apr  4 10:37:11 odhcp6c[1662]: entering stateful-mode on wlan0
Apr  4 10:37:11 odhcp6c[1662]: Starting <POLL> transaction (timeout 21600s, max rc 0)
Apr  4 10:37:11 : [INFO:Config.cpp]: Loaded configuration from /etc/prudynt.cfg
Apr  4 10:37:11 : [INFO:main.cpp]: PRUDYNT-T Next-Gen Video Daemon: Mar 17 2025 17:53:20_0e665be0
Apr  4 10:37:11 prudynt[1914]: [DEBUG:Logger.cpp]: Logger Init.
Apr  4 10:37:11 prudynt[1914]: [INFO:main.cpp]: Starting Prudynt Video Server.
Apr  4 10:37:11 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMPSystem::init()
Apr  4 10:37:11 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_OSD_SetPoolSize(1048576)
Apr  4 10:37:11 prudynt[1914]: [INFO:IMPSystem.cpp]: LIBIMP Version IMP-1.1.6
Apr  4 10:37:11 prudynt[1914]: [INFO:IMPSystem.cpp]: SYSUTILS Version: SYSUTILS-1.1.6
Apr  4 10:37:11 prudynt[1914]: [INFO:IMPSystem.cpp]: CPU Information: T31-N
Apr  4 10:37:11 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Open() = 0
Apr  4 10:37:11 prudynt[1914]: [INFO:IMPSystem.cpp]: Sensor: jxq03p
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_AddSensor(&sinfo) = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_EnableSensor() = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_System_Init() = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_EnableTuning() = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetContrast(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSharpness(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSaturation(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetBrightness(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetContrast(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSharpness(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSaturation(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetBrightness(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSinterStrength(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetTemperStrength(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetISPHflip(1)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetISPVflip(1)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetISPRunningMode(0)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetISPBypass(1)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetAntiFlickerAttr(2)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetAeComp(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetMaxAgain(160)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetMaxDgain(80)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: Set white balance. Mode: 0, rgain: 0, bgain: 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetBcshHue(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetDefog_Strength(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetDPC_Strength(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetDRC_Strength(128)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: ISP Tuning Defaults set
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetSensorFPS(25, 1) = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Tuning_SetISPRunningMode(0) = 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPSystem.hpp]: IMP_System_RebaseTimeStamp(15525644);
Apr  4 10:37:12 prudynt[1914]: [DEBUG:main.cpp]: create audio thread
Apr  4 10:37:12 prudynt[1914]: [DEBUG:worker.cpp]: Start audio_grabber thread for device 1 and channel 0 and encoder 0
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMPAudio::init()
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AENC_RegisterEncoder(&handle, &enc)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:WS.cpp]: Generated token length: 32, Token: 'QBVzqFDvyKLpdP0JJFbfWNWN8HW4cP2f'
Apr  4 10:37:12 prudynt[1914]: [INFO:Opus.cpp]: Encoder bitrate: 40000
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AENC_CreateChn(0, &encattr)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AI_SetPubAttr(1)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AI_GetPubAttr(1)
Apr  4 10:37:12 prudynt[1914]: [INFO:WS.cpp]: Server started on port 8089
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AI_Enable(1)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AI_SetChnParam(1, 0)
Apr  4 10:37:12 prudynt[1914]: [DEBUG:IMPAudio.cpp]: IMP_AI_GetChnParam(1, 0)
[... cut ...]
Apr  4 10:37:28 prudynt[1914]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Apr  4 10:37:28 prudynt[1914]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Apr  4 10:37:28 prudynt[1914]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:0
Apr  4 10:37:28 prudynt[1914]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Apr  4 10:37:38 crond[1785]: time disparity of 25004 minutes detected
Apr  4 10:38:00 crond[1785]: USER root pid 2304 cmd daynight
```

gtxaspec, 2025-04-04:
I dont see a crash indicated? is the stream dropping?

unijiang, 2025-04-04:
> I dont see a crash indicated? is the stream dropping?

I saw the `up` time is refreshed around 10:37, `11:18:39 up 41 min`.

gtxaspec, 2025-04-04:
OK, i'll try to find similar hardware to test and reproduce.

rorymurtagh, 2025-04-20:
Maybe related ... My Waze v3s crashed at random, more frequently when switching day and night. Nothing in the logs, CPU/uptime and memory were unremarkable. The problem is the power supply. I had a small 'brick' with two USBA outs I was using and seems like it didn't provide enough juice. It seems that  the little extra draw for night caused the cameras to do down. TLDR; check your power supply is beefy enough.

closed 2025-04-24 by unijiang