themactep/thingino-firmware #571: NexHT 86336 1080P Pan/Tilt/Zoom camera porting.

xromansx, 2025-03-19:
Hi,
 I would like to try your FW on this camera I was able to run OpenIPC on it but it lacks some features and with closed source part hard to adopt.
 
HW of the device:
 CPU: T21 (ide)
 Camera: jxf37
 Wifi: 8188fu
 IRCUT pins 79 80
   
USB only for power so Serial or chip flash method. as I already payed with another FW I'll try direct flash.

[image]

[image]

xromansx, 2025-03-19:
[image]

[image]

after uboot no output from console, I have dump of firmware from programmer if needed.

xromansx, 2025-03-19:
Compiled and boot: thingino-t21n_jxf37_rtl8188ftv.bin
dmesg
`[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_turkey_1.0__ (buildroot@buildroot) (b                                                                                                             uildroot-gcc-14.2.0) #1 PREEMPT Sun Dec 8 10:16:40 UTC 2024
[    0.000000] CPU0 RESET ERROR PC:058A09CC
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:864MHz L2CLK:432Mhz H0CLK:225MHz H2CLK:225Mhz PCLK:112Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0043b000 @ 00010000 (usable)
[    0.000000]  memory: 00035000 @ 0044b000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02900000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x028fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x028fffff]
[    0.000000] On node 0 totalpages: 10496
[    0.000000] free_area_init_node: node 0, pgdat 80443e90, node_mem_map 8100000                                                                                                             0
[    0.000000]   Normal zone: 82 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 10496 pages, LIFO batch:1
[    0.000000] Primary instruction cache 16kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 8-way, VIPT, no aliases, linesize 32 byt                                                                                                             es
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 64kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pa                                                                                                             ges: 10414
[    0.000000] Kernel command line: mem=41M@0x0 rmem=23M@0x2900000 console=ttyS1                                                                                                             ,115200n8 panic=10 root=/dev/mtdblock4 rootfstype=squashfs init=/init mtdparts=j                                                                                                             z_sfc:256k(boot),32k(env),224k(config),1472k(kernel),5504k(rootfs),-(rootfs_data                                                                                                             ),7680k@0x80000(upgrade),8192k@0(all)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 36368k/41984k available (3620k kernel code, 5616k reserve                                                                                                             d, 710k data, 212k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:422
[    0.000000] clockevents_config_and_register success.
[    0.000026] Calibrating delay loop... 858.52 BogoMIPS (lpj=4292608)
[    0.039991] pid_max: default: 32768 minimum: 301
[    0.040272] Mount-cache hash table entries: 512
[    0.040932] Initializing cgroup subsys debug
[    0.040959] Initializing cgroup subsys freezer
[    0.043717] devtmpfs: initialized
[    0.046395] regulator-dummy: no parameters
[    0.046898] NET: Registered protocol family 16
[    0.083366] bio: create slab <bio-0> at 0
[    0.096211] jz-dma jz-dma: JZ SoC DMA initialized
[    0.098778] usbcore: registered new interface driver usbfs
[    0.099140] usbcore: registered new interface driver hub
[    0.099574] usbcore: registered new device driver usb
[    0.100702]  (null): set:280  hold:281 dev=112500000 h=562 l=562
[    0.101145]  (null): set:69  hold:70 dev=112500000 h=140 l=140
[    0.104812] cfg80211: Calling CRDA to update world regulatory domain
[    0.106768] Switching to clocksource jz_clocksource
[    0.112401] dwc2 otg probe start
[    0.112446] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.112475] DWC IN OTG MODE
[    0.113438] dwc2 dwc2: Keep PHY ON
[    0.113459] dwc2 dwc2: Using Buffer DMA mode
[    0.113474] dwc2 dwc2: Core Release: 3.00a
[    0.113524] dwc2 dwc2: DesignWare USB2.0 High-Speed Host Controller
[    0.113558] dwc2 dwc2: new USB bus registered, assigned bus number 1
[    0.115281] hub 1-0:1.0: USB hub found
[    0.115326] hub 1-0:1.0: 1 port detected
[    0.115506] dwc2 dwc2: DWC2 Host Initialized
[    0.116107] dwc2 dwc2: enter dwc2_gadget_plug_change:2588: plugin = 1 pullup_                                                                                                             on = 0 suspend = 0
[    0.116137] dwc2 otg probe success
[... 45 lines cut ...]
[    0.744053] 0x000000080000-0x000000800000 : "upgrade"
[    0.751489] 0x000000000000-0x000000800000 : "all"
[    0.758475] SPI NOR MTD LOAD OK
[    0.762422] usbcore: registered new interface driver asix
[    0.768461] usbcore: registered new interface driver cdc_ether
[... cut ...]
[    1.560356] jffs2: notice: (421) jffs2_build_xattr_subsystem: complete buildi                                                                                                             ng xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0                                                                                                              orphan) found.
[    5.717090] gpio:jz->reg = 0xb0011000
[    5.717107] gpio pin: 0x4000000
[    5.717115] jz->dev_map[0]: 0x7900000
[   11.541502] set sensor gpio as PA-high-10bit
[   11.877641] jxf37 chip found @ 0x40 (i2c0)
[   11.877657] sensor driver version H20200416a
`

xromansx, 2025-03-19:
some variables:

wlan_module=8188fu
gpio_ircut=79 80
also how you add sufix to gpio_wlan with gpio_wlan 50 0 it checks gpio 50 and gpio 0

to be able to use wifi I needed to add below lines in interface config: 
pre-up echo 50 > /sys/class/gpio/export
pre-up echo out > /sys/class/gpio/gpio50/direction
pre-up echo 0 > /sys/class/gpio/gpio50/value

dmesg
[  145.176935] usb 1-1: new high-speed USB device number 2 using dwc2
[  145.387343] usb 1-1: device 0bda:f179 detected via OTG
[  145.471109] RTW: HW EFUSE
[  145.471134] RTW: 0x000: 29 81 00 FC  0B 00 00 00  00 0C 04 4C  10 07 00 00
[  145.471187] RTW: 0x010: 2C 2C 2C 2C  2C 2D 2E 2E  2E 30 30 11  FF FF FF FF
[  145.471240] RTW: 0x020: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471293] RTW: 0x030: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471345] RTW: 0x040: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471397] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471449] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471502] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471554] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471606] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471659] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471711] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  23 21 24 00  00 00 00 FF
[  145.471763] RTW: 0x0C0: FF 02 00 10  00 FF 00 FF  00 00 FF FF  FF FF FF FF
[  145.471814] RTW: 0x0D0: DA 0B 79 F1  47 66 40 F0  C8 14 70 6E  F1 09 03 52
[  145.471866] RTW: 0x0E0: 65 61 6C 74  65 6B 09 03  38 30 32 2E  31 31 6E 00
[  145.471918] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.471971] RTW: 0x100: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472023] RTW: 0x110: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472075] RTW: 0x120: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472127] RTW: 0x130: C1 B6 FF FF  FF FF FF FF  FF FF 00 11  FF FF FF FF
[  145.472231] RTW: 0x140: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472285] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472337] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472389] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472442] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472494] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472547] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472599] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472651] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472703] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472755] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472808] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  145.472874] RTW: hal_com_config_channel_plan chplan:0x23
[  145.530475] RTW: [RF_PATH] ver_id.RF_TYPE:RF_1T1R
[  145.530500] RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x11, rf_reg_path_avail_num:1, max_tx_cnt:1
[  145.530511] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[  145.530524] RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
[  145.530535] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x11, max_tx_cnt:1
[  145.530644] RTW: [RF_PATH] HALDATA's rf_type:RF_1T1R, NumTotalRFPath:1
[  145.530661] RTW: [TRX_Nss] HALSPEC - tx_nss:1, rx_nss:1
[  145.530672] RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
[  145.530683] RTW: [TRX_Nss] HALDATA - tx_nss:1, rx_nss:1
[  145.532784] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  145.578707] RTW: rtw_ndev_init(wlan0) if1 mac_addr=f0:c8:14:70:6e:f1
[  145.609959] RTW: rtw_ndev_init(wlan1) if2 mac_addr=f2:c8:14:70:6e:f1

I don't know why but it get's now only IPv6 and I see wlan0 and wlan1 in networks

xromansx, 2025-03-19:
GPIO 81 IRLED

gtxaspec, 2025-03-19:
suffix for gpios is:
XXo or O (lower case letter "o" or upper case letter "o")

example:
gpio_wlan=58O
would mean gpio 58 set high

xromansx, 2025-03-19:
I already figure out thanks took from other config )

xromansx, 2025-03-19:
Now problem is that I don't see the image it's pink. on motor section I see "Your camera does not seem to support motors"
and doesn't recognize sd card.

xromansx, 2025-03-19:
I created config for nexht_86336
could compile FW with working functions, but most important camera sensor doesn't work. 
for now not working: sensor, to test audio and mic and motor. for the motor then I was testing GPIOs I heared one that are connected to the motor.

xromansx, 2025-03-20:
another FW uses this driver for j37:
[   12.683756] probe ok ------->jxf37
[   13.019056] -----jxf37_detect: 623 ret = 0, v = 0x0f
[   13.019473] -----jxf37_detect: 631 ret = 0, v = 0x37
[   13.019484] jxf37 chip found @ 0x40 (i2c0)
[   13.019492] sensor driver version H20200416a
[   13.169973] jxf37 stream on

and it is working fine.

xromansx, 2025-03-21:
When the image is pink and there is motion I can see some lines.

gtxaspec, 2025-03-21:
Try a different setting,
[ 11.541502] set sensor gpio as PA-high-10bit

BR2_SENSOR_PARAMS="sensor_gpio_func=0"

Or

BR2_SENSOR_PARAMS="sensor_gpio_func=1"

xromansx, 2025-03-24:
Any ideas ?

themactep, 2025-03-24:
absolutely none, lol. you have the hardware and a working combination of code. compare, find the difference. we can do very little here without such a device on hands.

gtxaspec, 2025-03-24:
regarding pink screen, have you tried toggling the ir cut? if not, does it fix it?

themactep, 2025-03-24:
https://superarbor.io/products/nexht-smart-wifi-1080p-wireless-security-camera-with-night-vision-2-way-audio-cloud-storage-auto-track-pan-tilt-zoom

i tried to order it but $99 shipping to canada said no way.

xromansx, 2025-03-25:
are there any other sensor option I can test ?
what this means:
[ 13.019056] -----jxf37_detect: 623 ret = 0, v = 0x0f
[ 13.019473] -----jxf37_detect: 631 ret = 0, v = 0x37

gtxaspec, 2025-03-25:
rotate does not work on that SoC. are you saying the image is upside down?

xromansx, 2025-03-25:
[image]

gtxaspec, 2025-03-25:
no, the hardware ( the chip) does not support  rotation.   you can only flip h/v

themactep, 2025-03-25:
open the camera, confirm image sensor orientation. that's weird to have it on a side, it's not a doorbell.

xromansx, 2025-03-25:
If software doesn't help some scotch tape and hot glue and rotation worked ) even from sensor I can see that it was side oriented. now motors left, motor pins are correct when I press left and right on the circle camera moves a bit I don't understand why it's only till 1000 even if I setup higher value and also I can move very little steps one side it can barelly move another sometimes it loses direction, steps pins or speed doesn't change anything only if I put to small steps it doesn't move at all.

xromansx, 2025-03-25:
Got motors working:
42 39 40 41 
500 3700 1500

xromansx, 2025-03-26:
/sbin/record is looking for /etc/web.config but /etc/thingino.config has all the configs so needed to create link.

themactep, 2025-03-26:
How old is your fw? Update it.

xromansx, 2025-03-26:
I compiled it this weekend.

themactep, 2025-03-26:
There were commits since then. We develop extensively and commit often.

xromansx, 2025-03-27:
I'm on the latest version now. Everything seems to be working you can add it to supported devices only mention that immage is 90d rotated.
also is there motor_pos_0="1500" option 
My working configs:

[nexht_86336.config.txt](https://github.com/user-attachments/files/19487761/nexht_86336.config.txt)
[nexht_86336.txt](https://github.com/user-attachments/files/19487764/nexht_86336.txt)
[nexht_86336.uenv.txt](https://github.com/user-attachments/files/19487762/nexht_86336.uenv.txt)
[t21z_jxf37_rtl8188ftv.txt](https://github.com/user-attachments/files/19487763/t21z_jxf37_rtl8188ftv.txt)

gtxaspec, 2025-03-28:
keeping open until we review and add to production

themactep, 2025-03-28:
@xromansx can you make it into a normal pull request?

xromansx, 2025-03-28:
To speak truly I have no idea how to do it )

themactep, 2025-03-28:
no problem, i'll teach you. 
- fork the thingino repository ('fork' button in the top right corner)
- clone the forked repository locally (git clone ...)
- make necessary changes to the local repository (add files), make sure it compiles
- commit the changes and push them to your fork
- github will offer you to create a pull request with the changes to send upstream 
- create the pull request
- we'll accept and merge it

xromansx, 2025-03-28:
I think I succeed, let me know if it worked.

themactep, 2025-03-28:
Super. PR accepted. Thanks for the contribution!

closed 2025-03-29 by xromansx