themactep/thingino-firmware #574: Upgrade firmware fails

jsissom, 2025-03-20:
I would like to upgrade the version of firmware on my camera.  Following the directions, I tried:

`sysupgrade -p`

I get this error:

`Run as /sbin/sysupgrade -p
Creating /tmp/sysupgrade directory
Updating self
curl: (22) The requested URL returned error: 404

Cannot download https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade to /tmp/sysupgrade/sysupgrade
`

I then tried it passing the name of the current firmware and it still tries to download the above file which does not exist.

`
sysupgrade -p https://github.com/themactep/thingino-firmware/releases/download/firmware-2025-03-20/thingino-wuuk_y0510_sc4336p.b
in
`

Is there a way to work around the missing file?

Thanks

themactep, 2025-03-21:
You are trying to upgrade from a very old firmware. Many things have changed since. Please do the following:

> ATTENTION! If you are trying to upgrade thingino from a really old build, you might need to update the upgrading script first. Run the following command on the camera:
> `curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade && chmod +x /sbin/sysupgrade`
> then run `sysupgrade -f` for full upgrade and reconfigure the camera from scratch.

jsissom, 2025-03-21:
Thank you for the info.  The firmware on that camera is from November so it is ~4 months old.  Can you tell me how often I need to upgrade my cameras without having to start from scratch each time I upgrade?

Thanks
Jay

themactep, 2025-03-21:
we build new images nightly. you can track updates to the firmware in [the history of commits](https://github.com/themactep/thingino-firmware/commits/master/) in this repo. 

we usually do not advise to upgrade unless something is broken or there is a new feature that you eager to have in your system. it's an actively developed firmware, and we do not only fix things, sometimes we break them, too.

themactep, 2025-03-22:
please mark the issue closed if it's resolved for you.

closed 2025-03-23 by jsissom