themactep/thingino-firmware #580: White led call by url don't work (Aoqee C1)

tranbert, 2025-03-24:
HI...

Trying to get white led on /off by url... It's Ok with web UI / Gpio page

It's ok for ir led (or blue led) :  

`http://192.168.0.62/x/json-imp.cgi?cmd=ir850&val=1 `
`{"code":200,"result":"success","message":{"time":"1742835252","command":"irled 1 ir850","result":"","daynight":"day","color":1,"ircut":1,"ir850":1}}`

But for white led, no action no results :

`http://192.168.0.62/x/json-imp.cgi?cmd=white&val=1`
`{"code":200,"result":"success","message":{"time":"1742835347","command":"irled 1 white","result":"","daynight":"day","color":1,"ircut":1,"ir850":1}}`

Do I miss something ?

in json-imp.cgi white is coded so it "should" work  : 

`    ir850 | ir940 | white)
        command="irled ${val:-read} $cmd"
        ret=$(irled ${val:-read} $cmd)
        ;;`

It's on firmware master+ac2d28e, 2025-03-19 09:15:07 +0000

tranbert, 2025-03-24:
> But for white led, no action no results :
> http://192.168.0.62/x/json-imp.cgi?cmd=white&val=1

Found a working command : 
http://192.168.0.62/x/json-gpio.cgi?n=gpio_led_w&s=0

But doesn't work to get gpio state...

themactep, 2025-03-25:
does running `irled on white` and `irled off white` on console work for you?

themactep, 2025-03-25:
```
http://root:rootpass@192.168.0.62/x/json-imp.cgi?cmd=white&val=1
http://root:rootpass@192.168.0.62/x/json-imp.cgi?cmd=white&val=0
```

tested, these do work on jooan a2r

tranbert, 2025-03-25:
> irled on white

Nope

root@pod25-3 ~# irled on white
  Please define LED GPIO pin

themactep, 2025-03-26:
The pin is defined as `gpio_led_w` which is a tiny LED indicator.
Does this camera have a white projector instead?
I don't have such a model to confirm.

Run this on the camera
```
fw_setenv gpio_white 60O
fw_setenv gpio_sed_w
```
then try to toggle the light
```
irled on white
irled off white
```

tranbert, 2025-03-26:
>The pin is defined as gpio_led_w which is a tiny LED indicator.
> Does this camera have a white projector instead?

Yes, it's a very bright white led.

And nope : 

`root@pod25-3 ~# fw_setenv gpio_white 60O
root@pod25-3 ~# fw_setenv gpio_sed_w

root@pod25-3 ~# irled on white
  Please define LED GPIO pin

root@pod25-3 ~# irled off white
  Please define LED GPIO pin
`

BTW I'm fine with http://192.168.0.62/x/json-gpio.cgi?n=gpio_led_w&s=0 

I just don't understand why it's not Ok on command line, so I'll try to find a way :-) have to get time to read and understand the cgi's.

themactep, 2025-03-26:
You neeb to make system to reread values from env. Either `rm /tmp/environment` or just reboot the camera.

tranbert, 2025-03-26:
> You need to make system to reread values from env. Either `rm /tmp/environment` or just reboot the camera.

OK. Working... And State is Ok in Home Assistant Thanks !

[image]

closed 2025-03-26 by tranbert