themactep/thingino-firmware #581: No stream on Cinnado D1 after sysupgrade -f

papajim978, 2025-03-25:
No stream on Cinnado D1 after sysupgrade -f

Upgrade proceeded normally - attempted twice with on build available today.

themactep, 2025-03-25:
T23 is broken now, downgrade or disable OSD on both streams

themactep, 2025-03-30:
Please check if it works for you now.

papajim978, 2025-03-30:
Hangs on sysupgrade -f - not able to check T23 functionality. Won't come back on power cycle. Will prepare a card to reset tomorrow. 

[image]

themactep, 2025-03-30:
Over ssh? Not uart? This is fixed in the new build. Along with upgrade from web ui.

papajim978, 2025-03-30:
Yes over SSH.

3/30 4pm edt - did an autoupdate-full.bin with today's thingino.com cinnado d1 / t23 to recover. reconfigured. stream seemed good.
Original issue appears resolved.
3/30 5pm edt - ran sysupgrade -f over SSH to test repair of sysupgrade -f - hung again - different place

                                                  cinnado_d1_t23n
Check connection to https://github.com
Upgrade from GitHub
Remote file size     8388608
Remote file offset          0
Full flash size       8388608
Available space      18874368
Download size         8388608
Download firmware from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-cinnado_d1_t23n.bin
Downloading from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-cinnado_d1_t23n.bin to thingino-cinnado_d1_t23n.bin
################################################################################################################# 100.0%
Download checksum from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-cinnado_d1_t23n.bin.sha256sum
Downloading from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-cinnado_d1_t23n.bin.sha256sum to thingino-cinnado_d1_t23n.bin.sha256sum
################################################################################################################# 100.0%
Verify downloaded file
thingino-cinnado_d1_t23n.bin: OK
Clean up
removed 'thingino-cinnado_d1_t23n.bin.sha256sum'
Relocate downloaded file
'thingino-cinnado_d1_t23n.bin' -> '/tmp/sysupgrade/fw.bin'
Bootloader detected
Stop services
Stopping ONVIF Notify service ...
- Stopping onvif_notify_server by PID from /run/onvif_notify_server.pid
stopped process in pidfile '/run/onvif_notify_server.pid' (pid 2000)
Stopping video buffer ...
- Nothing to stop
Stopping Recording service ...
Stopping ONVIF discovery ...
- Stopping wsd_simple_server by PID from /run/wsd_simple_server.pid
stopped process in pidfile '/run/wsd_simple_server.pid' (pid 1876)
Stopping Ingenic Audio Daemon ...
- Stopping iad by PID from /run/iad.pid
stopped process in pidfile '/run/iad.pid' (pid 1792)
- Killing iad by PID
stopped iad (pid 1792)
Stopping Prudynt ...
- Killing prudynt by PID
stopped prudynt (pid 1753)
- Killing prudynt by name
stopped prudynt (pid 1753)
- Killing prudynt by full process name /bin/prudynt
no /bin/prudynt found; none killed
Stopping rc.local ...
Ciao
Stopping Telegram bot daemon ...
Stopping mdnsd ...
- Stopping mdnsd by PID from /run/mdnsd.pid
stopped process in pidfile '/run/mdnsd.pid' (pid 1682)
- Killing mdnsd by PID
stopped mdnsd (pid 1682)
- Killing mdnsd by name
stopped mdnsd (pid 1682)
- Killing mdnsd by full process name /sbin/mdnsd
no /sbin/mdnsd found; none killed
sh: out of range
Stopping HTTPD ...
- Killing httpd by PID
stopped httpd (pid 8236 1647)
Stopping crond ...
- Killing crond by PID
stopped crond (pid 1626)
- Killing crond by name
stopped crond (pid 1626)
Stopping NTP daemon ...
- Killing ntpd by PID
stopped ntpd (pid 1605)
- Killing ntpd by name
stopped ntpd (pid 1605)
- Killing ntpd by full process name /sbin/ntpd
no /sbin/ntpd found; none killed
  No WireGuard IP address configured
Captive Portal ...
  Not running
Stopping network ...
run-parts /etc/network/if-down.d
- DHCPv6 on lo
Stopping odhcp6c on lo ...
- Running service with PID 1440
  kill -9 1440
  rm -f /run/odhcp6c.pid
ip addr flush dev lo
ip link set lo down
run-parts /etc/network/if-post-down.d
  No bridge ports defined
  No VLAN on lo

run-parts /etc/network/if-down.d
  DHCPv6 disabled on wlan0
ip addr flush dev wlan0 label wlan0
client_loop: send disconnect: Connection reset

Climbed the ladder and did another autoupdate-full.bin to restore life.

gtxaspec, 2025-04-01:
please verify this is fixed on the latest builds, it should be!

papajim978, 2025-04-02:
1. Reset the camera to today's firmware using autoupdate-full.bin
2. Configured the camera to my standards.
3. Ran sysupgrade -f using SSH
4. AP showed as spec'd
T23 and sysupgrade issues both resolved!
THANK YOU!!!

closed 2025-04-02 by papajim978