themactep/thingino-firmware #584: Youtube streamer

xromansx, 2025-03-26:
Hi.
 I was interested if this feature is in plans ?
Thanks

thomasfcw, 2025-03-26:
https://www.synology.com/zh-tw/tech-partner/youtube

xromansx, 2025-03-26:
Was thining maybe there would be an option directly from camera like in OpenIPC cameras usually already have h264 and AAC.

gtxaspec, 2025-03-26:
yuo can use FFMPEG, you can add it via buildroot command.

xromansx, 2025-03-26:
I thought FFMPEG is huge for this small device but would be interesting to have it )

xromansx, 2025-03-26:
with ffmpeg image is 9.3M doesn't fit.

themactep, 2025-03-27:
use an off-camera restreamer. one restreamer can handle multiple cameras.

xromansx, 2025-03-27:
I know but one more dependency and no so fun, I may change SPI to 16Mb have some left from time when was replaceing some routers internal flash for OpenWRT interesting that ffmpeg package in openwrt occupy less space. maybe prudynt or raptor will have this feature some time.

gtxaspec, 2025-03-27:
> I know but one more dependency and no so fun, I may change SPI to 16Mb have some left from time when was replaceing some routers internal flash for OpenWRT interesting that ffmpeg package in openwrt occupy less space. maybe prudynt or raptor will have this feature some time.

some folks run ffmpeg off an sd card or the network, you have options.

xromansx, 2025-03-27:
Thanks, never tried this will try.

xromansx, 2025-03-27:
strange but couldn't find binary after compilation in per-package/thingino-ffmpeg

themactep, 2025-03-27:
check `per-package/thingino-ffmpeg/target/bin/`
read compilation log, every action is reflected there.

gtxaspec, 2025-04-01:
the future is ffmpeg, coming soon.

closed 2025-04-01 by gtxaspec