themactep/thingino-firmware #596: Feature req. Add option to use SD card as temporary folder for sysupgrade

boblapo, 2025-04-01:
Hi

Xiaomi cams don't have enough flash to perfom sysupgrade natively because they fail as space checks.

I use to rely on patch to get it but it's dirty

#update upgrade script
curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade && chmod +x /sbin/sysupgrade
cp /sbin/sysupgrade /sbin/sysupgrade_bkp
cp /sbin/sysupgrade-stage2 /sbin/sysupgrade-stage2_bkp
# replace /tmp by /mnt/mmcblk0p1/tmp
sed -i -e 's/\/tmp/\/mnt\/mmcblk0p1\/tmp/g' /sbin/sysupgrade
sed -i -e 's/\/tmp/\/mnt\/mmcblk0p1\/tmp/g' /sbin/sysupgrade-stage2
# -x to avoid self update of the script which removes the patch
sysupgrade -x 

Could be great if we could have native option to specify the location for /tmp so we can workaround low space setups.

thanks

themactep, 2025-04-04:
i don't think this would work anymore because i cannot make it save checksum file to sd card partition, it just silently fails.

```
cd /mnt/mmcblk0p1
curl -fL# --url "https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cam3
_t31x_gc2053_rtl8189ftv-update.bin.sha256sum" -o "thingino-wyze_cam3_t31x_gc2053_rtl8189ftv-update.bin.sha256sum"
```

themactep, 2025-04-04:
```
root@ing-wyze-cam3-e6de mmcblk0p1# sysupgrade -x -b
Run as /sbin/sysupgrade -x -b
/mnt/mmcblk0p1 is a mountpoint
Skip self-update
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -x -b
/mnt/mmcblk0p1 is a mountpoint
Check connection to https://github.com
Change directory to /mnt/mmcblk0p1
/mnt/mmcblk0p1
Upgrade from GitHub
Remote file size     16777216
Remote file offset          0
Full flash size      16777216
Available space      48234496
Download size        16777216
Download firmware from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin
Downloading from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin to thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin
####################################################################################################################################### 100.0%
Download checksum from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin.sha256sum
Downloading from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin.sha256sum to thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin.sha256sum
####################################################################################################################################### 100.0%
Verify downloaded file
sha256sum: thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin.sha256sum: no checksum lines found
Checksum does not match!
Clean up temporary files.
root@ing-wyze-cam3-e6de mmcblk0p1# ls -al
drwxrwxrwx    2 root     root          1024 Apr  4 05:31 .
drwxr-xr-x    1 root     root             0 Mar 17 17:53 ..
-rwxrwxrwx    1 root     root             0 Apr  4 05:32 thingino-wyze_cam3_t31x_gc2053_rtl8189ftv.bin
root@ing-wyze-cam3-e6de mmcblk0p1# 
```

themactep, 2025-04-04:
i guess it is some kind of FAT partition limitations and the length of these names.

themactep, 2025-04-04:
i can only make it skip checksum validation for sd card which is kinda sucks, so it's your call, do you prefer flashing an unverified binary or remapping memory to flash the usual way.

themactep, 2025-04-04:
ok, solution found, sysupgrade improved. @boblapo, please test and confirm it works for you.

moosehadley, 2025-04-04:
It appears this caused a regression for me.

I've always updated my cameras without an SD card installed, but now I'm unable to do so. From the WebUI, at least.

```
Run as /sbin/sysupgrade -p
Create /tmp/sysupgrade directory
Update self
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade to /tmp/sysupgrade/sysupgrade

######################################################################## 100.0%
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade-stage2 to /sbin/sysupgrade-stage2

######################################################################## 100.0%
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -p
mountpoint: /mnt/mmcblk0p1: No such file or directory
Check connection to https://github.com
Change directory to /tmp
Upgrade from GitHub
Cannot get remote file size
Clean up temporary files.
# 
--- sysupgrade exit! ---

```

moosehadley, 2025-04-04:
@themactep 

```
root@thingino-t23n-850e ~# grep IMAGE_ID /etc/os-release
IMAGE_ID=cinnado_d1_t23n
```

themactep, 2025-04-04:
Your issue is not a regression in sysupgrade. Camera profile has changed its name. Edit the line to match the new name then rerun sysupgrade.

moosehadley, 2025-04-04:
That took care of it. Thank you!

themactep, 2025-04-27:
done.

closed 2025-04-27 by themactep