themactep/thingino-firmware #602: motion/motion guard not working (Wyze vdb1)

gamer06, 2025-04-04:
I'm having trouble receiving notifications when motion is detected on the Wyze Video Doorbell camera (VDB1). I have MQTT configured like this and Motion Guard is activated with sensitivity 1. Am I doing something wrong?

[image]

I checked the /sbin/motion file, and in this path I don't have "motion_alarm"... could this be why?
[image]

Thanks

themactep, 2025-04-04:
motion_alarm is the flag set by motion detection script when a motion even is started. it gets deleted when the event finishes.

gamer06, 2025-04-04:
ok... and why aren't the detection notifications working?

themactep, 2025-04-04:
no idea. but it works here. check your settings, permissions.
[image]

gamer06, 2025-04-05:
Now yes, it is working. I did everything again, reset the camera and now yes. Thank you, you can close this issue

closed 2025-04-05 by themactep