themactep/thingino-firmware #604: No instructions on how to restore settings

carefulcomputer, 2025-04-07:
Wiki doesn't have instructions on how to restore the settings backup.
Please add instruction around saving and restoring backups.

themactep, 2025-04-10:
there is no option to restore anything as the back up files are not necessarily forward compatible with the new system after upgrade. use the back up as notes to self for reconfiguring the system from scratch.

closed 2025-04-10 by themactep

carefulcomputer, 2025-04-11:
theoretically, the backup files could be versioned and based on the firmware version they could allow restoring (i.e. if versions match only then allow restore). 
It is will be immensely helpful for folks who are either resetting the camera due to errors or are setting up multiple cameras with same settings.