themactep/thingino-firmware #605: Can't set timezone properly

mcderd-coder, 2025-04-07:
In time settings it says "Start typing the name of the nearest large city in the box above, then select from available variants."

No matter what I type, nothing appears to select from

I have entered 'Etc/GMT-1'

This seems to set the time of the software correctly, however, the timestamps on the videos and the file dates are incorrect.

How can I set the time to display correctly on the videos and have the saved files have the correct time on them?

marchershey, 2025-04-08:
I'm just spitballing here, but it seems like something in your browser might be blocking the script? Do you have any type of adblock / tracking prevention extensions?

When I type in `etc`, I get [a long list of options](https://i.imgur.com/StGtzam.png) to choose from in a autocomplete widget. Maybe your browser has autocomplete disabled?

Sorry if this isn't any help, I just thought I would throw some ingredients in the thought cauldron.

virmaior, 2025-04-08:
have you rebooted since setting the time zone?

mcderd-coder, 2025-04-08:
> have you rebooted since setting the time zone?

Yes I have, thanks for mentioning that

mcderd-coder, 2025-04-08:
Thanks for the info, I see that you get a list of options in the dropdown (although it says to type in a major city?) I can type options such as Etc/GMT+2 which will have the effect of changing the time top right, but the time on the video preview (or the time on the saved files) does not change.

[image]

virmaior, 2025-04-08:
what version of the firmware are you running?

is there an option in the drop down with the same TZ in the system. Maybe the OSD system is having trouble parsing your typed option? (I live in Japan and it correctly does the TZ for me).

mcderd-coder, 2025-04-08:
Thanks

I take it I will have to input all my setup again for things like Email Alerts and Webhooks, RTSP/ONVIF Access? Is there a way to save this info and re-import it once I have updated the firmware?

virmaior, 2025-04-08:
I think you'll lose all settings as there's been quite a bit of changes between then and now...

is there a more common GMT +2 tz code you can try? that might be the easiest way to get back to functional

mcderd-coder, 2025-04-08:
> I think you'll lose all settings as there's been quite a bit of changes between then and now...
> 
> is there a more common GMT +2 tz code you can try? that might be the easiest way to get back to functional

I am not sure about other zones other than GMT, as I mentioned, whatever I change there only affects the time shown top right, the time on the video is always 1 hour earlier than the current time no matter what I change. I have 3 of these cameras, so I might just stick with the wrong time rather than having to redo all the settings. I also tried changing the time servers, but that didn't make any difference.

virmaior, 2025-04-08:
I think you're not understanding what I'm saying.
I'm saying to type in something other than "Etc/GMT-1" (got that wrong previously and assumed +2) under the theory that
1. the web interface / OS can resolve that
2. BUT prudynt (at least in that version) could not.

e.g.,
Atlantic/Cape_Verde
OR
Atlantic/Azores

themactep, 2025-04-10:
so i close it as resolved.

closed 2025-04-10 by themactep