themactep/thingino-firmware #606: Build fails with git repo in detached head state

Yarn, 2025-04-08:
the line `GIT_BRANCH="$(shell git branch | grep '^*' | awk '{print $$2}')"` in the Makefile sets GIT_BRANCH to something along the lines of `* (HEAD detached at origin/stable)`. This later seems to make it into a use of `sh -c` where it causes a syntax error.

```
-------------------------------- distclean
if [ -d "/home/ryan/output-"(HEAD"/wuuk_y0510_sc4336p" ]; then rm -rf /home/ryan/output-"(HEAD"/wuuk_y0510_sc4336p; fi
/bin/sh: -c: line 1: syntax error near unexpected token `('
/bin/sh: -c: line 1: `if [ -d "/home/ryan/output-"(HEAD"/wuuk_y0510_sc4336p" ]; then rm -rf /home/ryan/output-"(HEAD"/wuuk_y0510_sc4336p; fi'
make: *** [Makefile:264: distclean] Error 2
```

themactep, 2025-04-10:
how old is your repo? i think it was fixed last week.

themactep, 2025-04-19:
do you still have that issue with the recent HEAD?

Yarn, 2025-04-21:
tried with commit 940bcb60e6e2615755b0e4fd8315d1fb37af9cfa, same result

```
if [ 16777216 -gt 16777216 ]; then /usr/bin/figlet -t -f pagga "OVERSIZE"; fi
rm -f /root/output-junk/wuuk_y0510_t31x_sc4336p_ssv6158/images/thingino-wuuk_y0510_t31x_sc4336p_ssv6158.bin.sha256sum
echo "# wuuk_y0510_t31x_sc4336p_ssv6158" >> /root/output-junk/wuuk_y0510_t31x_sc4336p_ssv6158/images/thingino-wuuk_y0510_t31x_sc4336p_ssv6158.bin.sha256sum
echo "# "(HEAD"+"940bcb6", "2025-04-21 10:29:38 +0000"" >> "/root/output-junk/wuuk_y0510_t31x_sc4336p_ssv6158/images/thingino-wuuk_y0510_t31x_sc4336p_ssv6158.bin.sha256sum"
/bin/sh: 1: Syntax error: "(" unexpected
make: *** [Makefile:370: pack] Error 2
```

gtxaspec, 2025-05-10:
the changes won't roll back if you checkout an earlier commit, you would have to manually cherry-pick or apply the changes while in the older commit.

gtxaspec, 2025-05-11:
fixed in 
https://github.com/themactep/thingino-firmware/commit/09ec4e414d99d15ab12c63495617c352fde32aa5
please test

Yarn, 2025-05-11:
tried it and it seems to work

presumably it could still break on git branch names with other shell characters or only parens but people will probably just not do that

closed 2025-05-11 by Yarn