themactep/thingino-firmware #608: Config portal breaks with wifi password containing spaces

wltechblog, 2025-04-08:
Enter "my password" and submit, verification shows "my". Looks like the form handler is eating it.

yebo29, 2025-04-09:
I didn't test other fields, but probably worth checking those as well.

themactep, 2025-04-10:
please check with the latest build

yebo29, 2025-04-10:
Oh nice! I'll check later today and report back. Thank you!

yebo29, 2025-04-11:
Hmmm I remember seeing something regarding resetting a camera after flashing with thingino but now I can't find it. The camera I've been using doesn't want to update itself with the inserted SD card even though I'm:
1. Powering it off
2. Inserting the SD card
3. Holding the setup button while re-inserting power

There's no status light at all. Restarting when not holding the power button makes the camera boot back into thingino (well, I can only assume since I don't see the thingino wifi and since I misconfigured it originally I don't have a way to access it). I have another camera I can try. But in the meantime, any of you gents know where the doc is and/or what the process is?

yebo29, 2025-04-11:
[Found it](https://github.com/themactep/thingino-firmware/wiki/Camera:-Wyze-v2-%E2%80%90-Neos-SmartCam-%E2%80%90-ATOM-Cam-1#thingino-seems-to-be-installed-but-i-cant-see-it-on-my-wifi-how-can-i-reset-the-configuration-how-do-i-reset-the-wifi-)

yebo29, 2025-04-11:
It worked! Using the second camera, I was able to configure my wifi password with spaces and it is now connected. Thank you so much!! This can be closed.

closed 2025-04-12 by themactep