themactep/thingino-firmware #609: SD card blank after installing thingino on WUUK Y0510

blackketter, 2025-04-09:
Thanks for the recommendation for the WUUK Y0510, it's a great camera and thingino is awesome.

Unfortunately, I need to restore to the factory firmware and found that the 1GB SD card that I used to install thingino now is blank after removing and mounting on my Mac. No thingino firmware, no backup directory.

The SD card was in the camera for about a week and this was the first time I removed it.

Not sure if this an error on my part or some bug in the installer, but I'd love to recover the original firmware.

The thingino firmware I downloaded on 2025-03-27.

(I also filed this as https://github.com/wltechblog/thingino-installers/issues/17 sorry if it's in the wrong place...)

wltechblog, 2025-04-10:
Not sure where your card got blanked, but that copy of firmware is your only way back to the WUUK ecosystem. The firmware itself doesn't format the card unless you manually go in and do it, the installer just adds some files... it is possible your mac "repaired" the card when mounting it. My suggestion is to try an undelete tool, presumably all the data is still on the card but the FAT table is missing.

You will for sure want to take an image of the card before attempting any operations, and try operating on a copy of the image. If you jump on Discord and share that image we may be able to lend a hand.

blackketter, 2025-04-10:
Thanks for the tip!  I was able to use dd and testdisk to recover the backup directory.  Not sure where the corruption happened but the warning that I got from testdisk was:
```
check_FAT: Unusual number of reserved sectors 4 (FAT), should be 1.
Warning: number of heads/cylinder mismatches 8 (FAT) != 255 (HD)
Warning: number of sectors per track mismatches 32 (FAT) != 63 (HD)
```

closed 2025-04-10 by blackketter