themactep/thingino-firmware #610: Changing substream resolution causes both feeds to stop

Milkysunshine, 2025-04-10:
On a wyze v3, if I try setting the substream resolution to 960x540 ( 50% bigger than default 640x360 ) the camera will not output either stream.

I'm not sure this is a limitation of the hardware, or something else.

gtxaspec, 2025-04-10:
the default ISP memory allocation (rmem) set set to match the stock settings, any increase in either resolution will need additional memory allocated.

See the wiki
https://github.com/themactep/thingino-firmware/wiki/ISP-Reserved-Memory-(RMEM)

Milkysunshine, 2025-04-10:
Thanks for the quick reply. 

I had to use ChatGPT to determine the address since the Thingino IPC Memory Calculator doesn't have 99M as an option. https://thingino.com/ramcalc

Since there was quite a bit of free system ram, I set the rmem to 40M with this command. 

fw_setenv osmem 59M@0x0; fw_setenv rmem 40M@0x3B00000; reboot

Stream and substream both now works with higher substream res. 

I'll monitor sys mem and performance for a bit.

closed 2025-04-10 by Milkysunshine

gtxaspec, 2025-04-10:
the SoC's typically come with 64, 128, or 256mb.  your wyze v3 has 128mb of system memory, so that's the option you should choose.

Milkysunshine, 2025-04-10:
Interesting. Using the latest Thingino firmware, it was only set up to allocate 99MB total. Maybe something to look into?

gtxaspec, 2025-04-10:
the defaults are set to 99 (OS) / 29 (ISP) .   99 + 29 = 128mb.

Milkysunshine, 2025-04-11:
Thanks for the clarification. I'll adjust my command to be 88/40