themactep/thingino-firmware #611: video recorder is not working

nicolasroche898, 2025-04-11:
# repro

```
jooan a6m
autoupdate-full.bin from sdcard using 
jooan_a6m_t23n_sc1a4t_atbm6012bx
master+72583a1, 2025-04-11 08:57:23 +0000

wait for it to finish boot after initial config,
then go to: services -> video recorder
choose storage: /mnt/mmcblk0p1/
click enable
```

# expected

non-empty video files in /mnt/mmcblk0p1/*

# actual

empty video files:

```
# ls -la /mnt/mmcblk0p1/ing-jooan-a6m-654a/records/2025
0411/
drwxrwxrwx    2 root     root          8192 Apr 11 20:02 .
drwxrwxrwx    3 root     root          8192 Apr 11 19:49 ..
-rwxrwxrwx    1 root     root             0 Apr 11 19:58 20250411T195805.mp4
-rwxrwxrwx    1 root     root             0 Apr 11 19:59 20250411T195905.mp4
-rwxrwxrwx    1 root     root             0 Apr 11 20:00 20250411T200005.mp4
-rwxrwxrwx    1 root     root             0 Apr 11 20:01 20250411T200105.mp4
-rwxrwxrwx    1 root     root             0 Apr 11 20:02 20250411T200205.mp4
```

# root cause

The streamer / prudynt / openrtsp / someone has started using more memory buffers from the camera. 
log shows failed allocations. this didn't occur in older firmwares, and shouldn't occur according to the excel memory calculator,
which shows that 11M is enough for 720p.

# workaround

remove this line:
`BR2_THINGINO_RMEM_12=y`

from this file:
`jooan_a6m_t23n_sc1a4t_atbm6012bx/jooan_a6m_t23n_sc1a4t_atbm6012bx_defconfig`

this will set the default memory allocation to wasted 23M instead of optimal 12M.
```
rmem=23M@0x2900000
osmem=41M@0x0
```

nicolasroche898, 2025-04-11:
testing ispmem-calc on jooan a6m, latest firmware mentioned above:

```
osmem 41 rmem 23 OK
osmem 46 rmem 18 OK
osmem 49 rmem 15 OK
osmem 51 rmem 13 OK
osmem 52 rmem 12 FAIL (*.mov size 0) (this setting was working on older firmware)
osmem 53 rmem 11 (calculation from ingenic excel file with 720p)
```

so theoretically, you just need to increase rmem=12 to rmem=13

i increased to rmem=16 just to be safe.

themactep, 2025-04-12:
I have changed the file format from .mov to .mp4, that could be it.

nicolasroche898, 2025-04-12:
occurs both with mp4 and mov. anyway can you increase rmem to 16 in the source code? thx

jooan_a6m_t23n_sc1a4t_atbm6012bx/jooan_a6m_t23n_sc1a4t_atbm6012bx_defconfig

themactep, 2025-04-12:
done.

nicolasroche898, 2025-04-13:
fixed in new version, rmem is now 16M instead of 12M
tested with BUILD_ID="master+b6b4bdd, 2025-04-13 08:57:57 +0000".
closing issue.

closed 2025-04-13 by nicolasroche898