themactep/thingino-firmware #612: Support for Vstarcam - CS621VS

fss-hacks, 2025-04-12:
Support for Vstarcam devices 

Hello, I am about to receive a Vstarcam CS621ZS triple lens PTZ cam. 
There has been a lot of attention devoted to hacking Vstarcam devices and they are normally Ingenic chipset devices. I believe this device has a T31 in it. 

I am willing to work with you to bring support to this and other Vstarcam products - they have a pretty broad product line with a lot of different devices potentially all could be supported by Thingino.  
This would also bring Thingino support to cameras with multiple lenses. 

Can you point me in the direction of what I should do first? I am not sure which derivative of the T31 or how much NOR the camera has. I am guessing without evidence that it may work with the USB cloner but am not sure what firmware to throw at it.

gtxaspec, 2025-04-12:
First step, need to have photos of the inside of the unit, PCB, to verify processor, wifi module, etc.

Second step, once we know the hardware, then we need to get a flash dump =D

fss-hacks, 2025-04-12:
That’s the part I’m kind of afraid of. It looks like a very hard unit to crack open cuz of the 3 different cameras and shape.
I know openipc has the ipctool utility and I figured I could also get some hardware info if I was able to log in as root using the WyzeHacks sd card hack and get access to the hardware info.bin file.
I’ll also take a look to see if the FCC has any report data on it. There’s a document Vstarcam has floating around on its support website that shows this:
https://www.vstarcam.com/support/wp-content/uploads/2024/09/CS621ZS-EN.pdf?srsltid=AfmBOoqiWXIx27Wp1J731jpHbicN_1PKHyL-MmzcNO3TZEv8Y5J0gSeX

gtxaspec, 2025-04-12:
wyzehacks only works on wyze (hualai) hardware, so chances are slim (but not impossible) that it may work.  
if you have the FCCID's, you should be able to look up the hardware photos.

gtxaspec, 2025-04-12:
according to that pdf, the ram is 512mb, which, if true, would mean it would be a T40A or T41-ish cpu, not t31 (t31 max 128mb)

fss-hacks, 2025-04-13:
That would be (very, very) unfortunate. Do you think that something like wyzehacks would work on it considering the commands for it would be similar at a system level?

I also have another rebranded Vstarcam battery camera I think has a more appropriate T series chip in it. I have to check and find out what model it actually is and give you the info.

For anyone who needs it’s future reference, Vstarcam cameras that use the Okam Pro app come in two flavors. One flavor that is usually for battery cameras that does not openly have Onvif (which I believe are running T series chips), and another that is for powered cameras that does allow Onvif connections (more of a grab bag of different ones like we just found).

To use the Onvif, you have to go into device settings and set the “plain text password”, then Onvif will be reachable on an unusual port in the 1000s range.

gtxaspec, 2025-04-13:
wyzehacks relies on a specific system design (init scripts, factory scripts, etc) to gain the shell, this is not typically replicated by other vendors.

Best way to see if the device is a thingino candidate is to tear one of them down as it turns out. Sometimes the vendors change the internal hardware ( albeit identical from the outside ), so even similar units may have completely different hardware.

fss-hacks, 2025-04-13:
Gotcha ☹️ I think I’ll do it as a last resort.
Wouldn’t it be ironic if Vstarcam never changed the root passwords after all the attention placed on them with the hardware engineers physically attacking their cams?
From what I’ve seen with other brands, I wouldn’t be surprised.
I’ll find out today.

fss-hacks, 2025-04-14:
Some updates.
I’ve got Telnet running on it now.
This was NOT easy.
First, there is supposed to be a web ui thru a http port accessible with these models.
The http port changes every time the camera reboots, and the only way to get passed thru to a browser login prompt is to unbind the camera from the Okam Pro app, then use the Vstarcam Camera Finder windows app to select a static IP address.
Once that happens, you can then use a CGI attack in the browser url line, in the format of
[http://IP address:port/trans_cmd_string.cgi?user=admin&pwd=888888&cmd=2101&command=1](http://192.168.1.156:22293/trans_cmd_string.cgi?user=admin&pwd=888888&cmd=2101&command=1)

This will enable telnet on port 23, but no available Vstarcam root login/pass combos have worked to log in.

gtxaspec, 2025-04-14:
we have a list of some passwords we have gathered:
https://github.com/themactep/wiki/blob/master/hacking/research-security.md

GrumpyMeow, 2025-05-05:
@fss-hacks I'm a newbie here. 
I also have a camera which is to be used with the OKam Pro app. I've been tinkering with Wireguard to intercept and decrypt the communication. It's using the PPPP ("Little Stars protocol") and recognize the urls you've posted.
You can probably decrypt the communication using: https://github.com/magicus/pppp-dissector/blob/main/pppp_dissect.lua
The okam pro app provides two seeds `vstarcam2018`  and `vstarcam2019`. My cam is using the seed `vstarcam2018`. 

username = `admin`
password = `888888`
My current code is sending the following MSG_DRW requests and receiving the data:
* `GET /get_status.cgi?name=admin&loginuse=admin&userId=007&loginpas=bond&user={username}&pwd={password}&`
* `GET /eye4_authentication.cgi?loginAccount=007&loginToken=secret&loginuse=admin&userId=007&loginpas=bond&user={username}&pwd={password}&` 
* `GET /get_params.cgi?loginuse=admin&userId=007&loginpas=bond&user={username}&pwd={password}&`
* `GET /get_camera_params.cgi?loginuse=admin&userId=007&loginpas=bond&user={username}&pwd={password}&`
* `GET /livestream.cgi?streamid=10&substream=1&loginuse=admin&userId=007&loginpas=bond&user={username}&pwd={password}&`

I also intercepted the following urls which hopefully is for PTZ control:
* `GET /trans_cmd_string.cgi?cmd=2127&command=1&loginuse=admin&userId=007&loginpas=bond&user=admin&pwd=888888&`
* `GET /trans_cmd_string.cgi?cmd=2017&command=11&mark=212&type=2&loginuse=admin&userId=007&loginpas=bond&user=admin&pwd=888888&`

I'll try to get some more info..

GrumpyMeow, 2025-05-05:
@fss-hacks  After i enabled telnet, i've managed to login using:
username:` vstarcam2017`, password:`20170912`
Hopefully it will also work for you

gtxaspec, 2025-05-05:
@GrumpyMeow Can you share dmesg output?

fss-hacks, 2025-05-10:
Do you think that any of the streams you picked up could be those of the other multi lens cameras? I can only get the main lens (and a zoomed in version of the main lens).

I did that by discovering that there are two rtsp servers running on the camera; one on 10554, one on 10555. Both have the same stream format as discussed.

It seems completely idiotic to have built a camera that is capable of two different ports (unlike anything I have ever seen, an advancement) and then leave two separate cameras just connected via the garbage Little Stars rubbish.

fss-hacks, 2025-08-10:
> GrumpyMeow left a comment [(themactep/thingino-firmware#612)](https://github.com/themactep/thingino-firmware/issues/612#issuecomment-3172494668)
>
> I ordered the cheapest camera i could find which is advertised to be used with the o-kam pro software. It's actually really easy to get access to the camera. I got the camera fresh out-the-box and didn't connect it using the o-kam pro app.
>
> - Power-on the camera. The camera should do some self-check stuff like PTZ-calibration. The camera plays audio: Please connect to wi-fi. A wi-fi access-point will be provided by the camera with the camera ID in it's name: @IPC-xxxxxxx.
> - Connect to the camera's access-point. No password should be needed. Your device should be given an IP# in the range of 192.168.168.*. The IP of the camera itself is 192.168.168.1.
> - Verify the IP# of the camera: ping 192.168.168.1
> - Try to retrieve some data from the camera: curl ***@***.***:81/get_status.cgi. This should result in a list of var xxxx="yyyyy";. You should also now be able to check if you got scammed with a lower-resolution camera: var sensor_name="gc1084"; (i bought mine as 8mp).
> - Retrieve some more data: curl ***@***.***:81/get_params.cgi. Here i'm able to see which cpu is used: var CPU="T23-N";
> - Enable telnet: curl ***@***.***:81/trans_cmd_string.cgi?cmd=2101\&command=1. And wait for the command to finish.
> - Connect via telnet: telnet 192.168.168.1. You probably will see the default hostname of the camera veepai. Login using: username=vstarcam2017, password=20170912. Hopefully these credentials also work for your device and you should be connected.
> - You can check the hashed password: cat /etc/password . Which for my camera is: ***@***.***:etc]

Have you tested this with this model specifically? If so this is a HUGE breakthrough, thank you so much!
According to that info, the processor is indeed an Ingenic T23-N, which is indeed supported by thingino.
Only issue on my side is that I think I tried that password on mine and it didn’t work, so I can’t confirm if my sensors are the same as yours (and we all know that these manufacturers change sensor components right and left).

I’m hoping the developer of thingino can see this so things can progress forwards.

fss-hacks, 2025-08-10:
> Alexandros89 left a comment [(themactep/thingino-firmware#612)](https://github.com/themactep/thingino-firmware/issues/612#issuecomment-3172873244)
>
> I have tried, but I received this
> ~ $ curl ***@***.***:81/get_status.cgi
> var result="Auth Failed";
> Any idea?

I’m guessing because the camera has another password already. It needs to be in completely reset setup mode. I’m about to do mine myself but it’s doing its job as a security camera rn so have to wait for the right opportunity.

GrumpyMeow, 2025-08-10:
Initially i didn't get the auth-failed error. It sounds weird, but please try to open in your browser: http://admin:888888@192.168.168.1:81
In the browser you'll probably get an 404 type of error message. Then switch to your terminal and run the "curl" commands..
For me this works every time.
------
In the meantime I've tried to flash a Thingino firmware onto my camera which apparently I succeeded in doing, as the camera is now not working any more :-). So I'll have to whip out my screwdriver-set and see if I can recover it. 

If you put in an vfat-formatted sdcard in the camera, it will write automatically a backup of some parts of the flash. I myself created a backup using DD before flashing.

I managed to trigger the built-in flash-update machanism by putting a firmware file in the root of the sdcard with the (case sensitive) named
 "IN-sys-" (8mb for my cam) flash file. And then starting from the telnet session "/etc/init.d/rcS". Eventually it does not flash the firmware as it seems to check for a file-header or crc's. I then did a bold move of "dd if=/mnt/sda0/IN-sys- of=/dev/mtd6" and is now non-booting.

fss-hacks, 2025-08-11:
> Initially i didn't get the auth-failed error. It sounds weird, but please try to open in your browser: ***@***.***:81

Sorry for your loss 😔
I think we need to be careful that we are aware of all the sensor types and of the internal machinations, then we build fresh from source.

I’m gonna check if mine is still in the Amazon warranty period - pretty sure it isn’t but if it is, I’ll reset it and see what I can do based on what you’ve done so far without trying that particular fatal move.
Just making sure, your camera is the exact same as the model the original post was made for, right? Cuz I thought I read that you have a “similar” camera.

willgryan, 2025-08-20:
> Some updates.
> I’ve got Telnet running on it now.
> This was NOT easy.
> First, there is supposed to be a web ui thru a http port accessible with these models.
> The http port changes every time the camera [...]

I used the Vstarcam Camera Finder windows app to pick a static IP address and port for my camera. However, all my attempts to connect & issue CGI commands are refused. Did you enable happen to do this before or after you setup the cam with the Okam pro app?

fss-hacks, 2025-11-15:
Unfortunately I wasn’t thinking on my feet in my last response. The 4 streams that exist ARE the two ONVIF ports for the main camera. The main camera has the two ports 10554 and 10555, and each of the two ports has 2 streams each, a low quality and a high quality stream.
I found this by checking the ONVIF profiles for both the ports.

I think trying to get the side cameras is a lost cause if the device’s OS doesn’t see the side cameras.
I’m very curious if a tear down of only the side cameras might give us insight about their hardware and what they run on. I’m guessing that the O-KAM app connects to both the main and side cameras as separate devices but the interface treats them as one device, uniting them.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS.

abmateen, 2026-01-03:
@fss-hacks how to you managed to enable the Ovnif ports?

Alexandros89, 2026-09-07:
i have the 4 rstp on the original firmware