themactep/thingino-firmware #618: 360 AP1PA3 U-BOOT FAILD

shyydy, 2025-04-15:
Thingino U-Boot for Ingenic T31X SPL 2013.07 (Mar 17 2025 - 17:53:20 UTC)
Timer init
CLK stop
PLL init
pll_init:363
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_c                                                              fg.l2div = 2
nf = 116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf = 100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf = 100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq = 1392000000
mpll_freq = 1200000000
vpll_freq = 1200000000
cpu clk source: apll
ddr clk source: mpll
ddrfreq 600000000
cclk  1392000000
l2clk 696000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
DDR clk rate 600000000
SDRAM init ok
board_init_r
image entry point: 0x80100000

Thingino U-Boot for Ingenic 2013.07 (Mar 17 2025 - 17:53:20 UTC)

Architecture: XBurst@I
Platform: ISVP
Built profile: T31X
SOC Model: T31X
DRAM:  128 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 699k for U-Boot at: 83f50000
Reserving 32800k for malloc() at: 81f48000
Reserving 32 Bytes for Board Info at: 81f47fe0
Reserving 124 Bytes for Global Data at: 81f47f64
Reserving 256k for boot params() at: 81f07f64
Stack Pointer at: 81f07f48
Now running in RAM - U-Boot at: 83f50000

MMC:   msc: 0
JZSFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 16MB (5e 40 18 5e aa)
In:    serial
Out:   serial
Err:   serial
GPIO:  gpio_button: No GPIO env settings provided
RST:   Reset button undefined; skipping factory reset.
GPIO:  gpio_default: 25id 26iud
GPIO:  gpio_usb_en: No GPIO env settings provided
Net:   HW address for ETH: XXXX
Net:   HW address for WLAN: XXXX
ETH:   Searching for valid PHY
ETH:   Error: Invalid PHY configuration.
ETH:   PHY not found!
GPIO:  gpio_default_net: No GPIO env settings provided
GPIO:  gpio_ircut: 52i 53i 49i 50i 57i 58i
GPIO:  gpio_user: No GPIO env settings provided
GPIO:  gpio_motor_v: No GPIO env settings provided
GPIO:  gpio_motor_h: No GPIO env settings provided
MMC:   SD card disabled
isvp_t31#

I write the bin file with ch341 programmer.
It's stoped at here.this is all the thing from PuTTY.
What can I do? Thanks.

shyydy, 2025-04-15:
I insert 4G FAT32 sdcard:

isvp_t31# mmc rescan; mmcinfo
Device: msc
Manufacturer ID: 3
OEM: 5344
Name: SD07
Tran Speed: 0
Rd Block Len: 0
SD version 2.0
High Capacity: No
Capacity: 0 Bytes
Bus Width: 1-bit
isvp_t31#

shyydy, 2025-04-16:
I flashed the bin file agian,this is first load info:

`MMC:   msc: 0
JZSFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 16MB (5e 40 18 5e aa)
ENV:   Warning - bad CRC, using default environment
ENV:   Saving default environment...
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
In:    serial
Out:   serial
Err:   serial
GPIO:  gpio_button: No GPIO env settings provided
RST:   Reset button undefined; skipping factory reset.
GPIO:  gpio_default: 25id 26iud
GPIO:  gpio_usb_en: No GPIO env settings provided
Net:   MAC address for ETH based on device serial set
JZSFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 16MB (5e 40 18 5e 00)
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
Net:   MAC address for WLAN based on device serial set
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
ETH:   Searching for valid PHY
ETH:   Error: Invalid PHY configuration.
ETH:   PHY not found!
GPIO:  gpio_default_net: No GPIO env settings provided
GPIO:  gpio_ircut: 52i 53i 49i 50i 57i 58i
GPIO:  gpio_user: No GPIO env settings provided
GPIO:  gpio_motor_v: No GPIO env settings provided
GPIO:  gpio_motor_h: No GPIO env settings provided
MMC:   SD card disabled
isvp_t31#`

shyydy, 2025-04-16:
I flashed the bin file agian,this is first load info diffrent up:

MMC:   msc: 0
JZSFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 16MB (5e 40 18 5e aa)
ENV:   Warning - bad CRC, using default environment
ENV:   Saving default environment...
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
In:    serial
Out:   serial
Err:   serial
GPIO:  gpio_button: No GPIO env settings provided
RST:   Reset button undefined; skipping factory reset.
GPIO:  gpio_default: 25id 26iud
GPIO:  gpio_usb_en: No GPIO env settings provided
Net:   MAC address for ETH based on device serial set
JZSFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 16MB (5e 40 18 5e 00)
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
Net:   MAC address for WLAN based on device serial set
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
ETH:   Searching for valid PHY
ETH:   Error: Invalid PHY configuration.
ETH:   PHY not found!
GPIO:  gpio_default_net: No GPIO env settings provided
GPIO:  gpio_ircut: 52i 53i 49i 50i 57i 58i
GPIO:  gpio_user: No GPIO env settings provided
GPIO:  gpio_motor_v: No GPIO env settings provided
GPIO:  gpio_motor_h: No GPIO env settings provided
MMC:   SD card disabled
isvp_t31#

gtxaspec, 2025-04-16:
are you using the latest firmware?

shyydy, 2025-04-17:
> are you using the latest firmware?

I flashed 2025-02-16,It's work.

gtxaspec, 2025-04-17:
ok, i will investigate

gtxaspec, 2025-04-17:
fixed in the repo, you can build your own (update your repo to latest commit) or wait to tomorrows build

ycsongx, 2025-04-17:
my last version is 2025-04-03, it works. 
update use the latest version, the same state

shyydy, 2025-04-17:
> my last version is 2025-04-03, it works. update use the latest version, the same state

lasted firmware is worked,Thank gtxaspec very much

closed 2025-04-17 by shyydy