themactep/thingino-firmware #619: Issues with CBR on Wyze v3

Milkysunshine, 2025-04-16:
firmware: master+fbe2d60, 2025-04-15 09:35:26 +0000
hardware: wyze v3 t31zx, gc2053, 16 MB

When choosing VBR on the main stream, the bitrate is generally around the requested value. When choosing CBR, the bitrate is significantly different than the requested value. One can set the bitrate to silly values like 30000, and only get around 1000 kb/s according to VLC. There really isn't much documentation that I can find regarding CBR settings on thingino/prudynt.

there are a few bug reports when searching for bitrate or cbr on both tingino and prudynt, but none are specific to CBR.

themactep, 2026-08-19:
Fixed — root cause found and verified on the exact hardware (Wyze Cam3, `wyze_cam3_t31x_gc2053_atbm6031`, 192.168.88.127).

## Root cause

Two bugs in prudynt's encoder setup for T31/T40/T41 (the "new SDK" path in `src/video/IMPEncoder.cpp`):

1. **Bitrate unit mismatch.** After `IMP_Encoder_SetDefaultParam(..., stream->bitrate)` correctly sets the bitrate (kbps), the code overwrote the RC attr fields (`attrCbr.uTargetBitRate`, `uMaxBitRate`, `uMaxPictureSize`) with the raw **kbps** value — but those fields take **bit/s** (per `imp_encoder.h`: *"unit: bit/s"*). The encoder then targeted ~1000× less than configured and floored out at ~700-1000 kbps regardless of the setting. This is exactly the "~1000 kbps no matter what" symptom — and why VBR was also off.

2. **CBR min QP too tight.** `iMinQP = 34` is a quality ceiling that prevents the rate control from ever reaching the target bitrate (it capped output around 700 kbps). The old-SDK path uses `minQp = 15`, and that's the SDK default.

## Fix

- Drop the bitrate field writes entirely — `SetDefaultParam` owns the bitrate; the switch now only tunes QP/delta bounds.
- CBR: `iMinQP = 15, iMaxQP = 45` (matching the old-SDK path).

Committed and pushed to `themactep/prudynt-t` **stable**: `8a5e497`. Firmware pinned in `prudynt-t.mk` (commit `7aeaa44eb` on `ciao`).

## Verified on hardware (Wyze Cam3 T31X/GC2053, measured over RTSP with ffmpeg)

| Mode | Config | Measured |
|---|---|---|
| CBR | 1500 kbps | 1.7 Mbps |
| CBR | 4000 kbps | 4.3 Mbps (was 0.7 Mbps before fix) |
| CBR | 8000 kbps | 7.4 Mbps |
| VBR | 4000 kbps | 4.2 Mbps |
| CAPPED_VBR | 3000 kbps | 3.1 Mbps |

(Target ±~5-10% is expected — matches Ingenic's bitrate-control doc: actual rate ≈ uTargetBitRate ± 5%, scene-dependent.)

Also verified the SDK's `SetDefaultParam` takes kbps while the struct fields take bit/s — the classic Ingenic inconsistency; confirmed empirically (4000 kbps config produced 4.3 Mbps only after removing the wrong-unit writes, and an out-of-range value made `IMP_Encoder_CreateChn` fail).

Note: `raptor` (the master-branch default streamer) has the same unit bug in its T31 path (`thingino-raptor-hal` sets `attrCbr.uTargetBitRate = br_kbps`) plus the same `minQP=34` CBR default — I'll fix that next.

Upgrade to a `ciao` build with this fix and CBR will track the configured bitrate. Thanks for the detailed reports — the "stuck at ~1000 kbps" detail across all values is what made the unit bug obvious.

themactep, 2026-08-19:
@gtxaspec — heads-up that raptor likely has the same bug on T31/T40/T41 (new SDK). The fix for prudynt just landed, and the identical code pattern exists in `gtxaspec/raptor-hal` `src/hal_encoder.c`:

1. The RC attr fields (`attrCbr.uTargetBitRate`, `uMaxPictureSize`, etc.) take **bit/s** per `imp_encoder.h`, but raptor-hal writes `br_kbps` (`cfg->bitrate / 1000`) into them, overwriting the correct values `IMP_Encoder_SetDefaultParam()` set. Result: encoder targets ~1000× less than configured and floors out around 700-1000 kbps.
2. CBR `minQp` defaults to **34** — too tight a quality ceiling for the rate control to reach the target bitrate (caps output ~700 kbps). The old-SDK path and the SDK default use 15.

prudynt fix (verified on Wyze Cam3 T31X): `themactep/prudynt-t` `8a5e497` — dropped the bitrate field writes entirely (let `SetDefaultParam` own the bitrate), kept QP tuning, set CBR `minQp=15/maxQp=45`. After the fix: CBR 4000 config measured 4.3 Mbps (was 0.7), CBR 1500/8000 → 1.7/7.4 Mbps, VBR 4000 → 4.2 Mbps, CAPPED_VBR 3000 → 3.1 Mbps.

For raptor-hal the equivalent change would be removing the `attrCbr.uTargetBitRate/uMaxPictureSize = br_kbps` writes in the new-SDK path (`hal_encoder.c` ~line 607) and changing the `cbr_min` default from 34 to 15.

closed 2026-08-19 by themactep