themactep/thingino-firmware #627: Status of wyze cam floodlight v1?  (floodlight doesn't appear to be working)

juanboro, 2025-04-22:
Hi,
I see there is a separate specific release of the firmware for the wyzecam v3 with floodlight... but what exactly does this currently provide over the wyzecam v3?

As it is - for my wyze cam w/ floodlight - it doesn't seem to be recognizing the serial usb connection to the floodlight accessory.  The kernel boot logs are filled with messages like this:
[    1.205959] dwc2 dwc2: Sorry, SPLIT transfer is not supported!

From [this](https://github.com/gtxaspec/wz_mini_hacks/issues/172) it seems @gtxaspec provided a specific kernel build for wz_mini_hacks (I believe I used it in the past) for this.  But it isn't clear from the issue threads what specific kernel options were set.

With wz_mini_hacks I was able to use the script in [this ](https://github.com/gtxaspec/wz_mini_hacks/discussions/771) thread to control the floodlight - it's not clear there is an equivalent I've seen in the current thingino releases.

gtxaspec, 2025-04-22:
the floodlight specific release includes different kernel options to fix the issues with the USB hub in the unit.  basic floodlight control is also available in thingino.

juanboro, 2025-04-22:
Hi,
I'm sorry if I'm missing something obvious - but I don't see any of this when I install the latest floodlight firmware - and from the current sources as I understand it doesn't seem to be the case - the only thing that appears different (maybe I'm not looking everywhere corretly)  is adding the floodlight package:
```shell
git clone https://github.com/themactep/thingino-firmware.git
diff  ./thingino-firmware/configs/cameras/wyze_cam3_t31x_gc2053_atbm6031/wyze_cam3_t31x_gc2053_atbm6031 ./thingino-firmware/configs/cameras/wyze_camfl1_t31x_gc2053_atbm6031/wyze_camfl1_t31x_gc2053_atbm6031.config
diff  ./thingino-firmware/configs/cameras/wyze_cam3_t31x_gc2053_atbm6031/wyze_cam3_t31x_gc2053_atbm6031 ./thingino-firmware/configs/cameras/wyze_camfl1_t31x_gc2053_atbm6031/wyze_camfl1_t31x_gc2053_atbm6031.uenv.txt
 diff  ./thingino-firmware/configs/cameras/wyze_cam3_t31x_gc2053_atbm6031/wyze_cam3_t31x_gc2053_atbm6031_defconfig ./thingino-firmware/configs/cameras/wyze_camfl1_t31x_gc2053_atbm6031/wyze_camfl1_t31x_gc2053_atbm6031_defconfig
1c1
< # NAME: Wyze Cam v3 (T31X, GC2053, ATBM6031)
---
> # NAME: Wyze Cam Floodlight v1 (T31X, GC2053, ATBM6031)
12,13c12
< BR2_PACKAGE_WYZE_ACCESSORY_CAR=y
< BR2_PACKAGE_WYZE_ACCESSORY_SPOTLIGHT=y
---
> BR2_PACKAGE_WYZE_ACCESSORY_FLOODLIGHT=y
```

And the floodlight package appears to just add the serial port module (no basic floodlight control):
```shell
grep -A5 -e "define.*INSTALL.*FLOODLIGHT" thingino-firmware/package/wyze-accessory/wyze-accessory.mk
define WYZE_ACCESSORY_INSTALL_TARGET_CMDS_FLOODLIGHT
        $(INSTALL) -m 0755 -d $(TARGET_DIR)/etc/modules.d
        echo ch341 >> $(TARGET_DIR)/etc/modules.d/accessory
        echo snd-usb-audio >> $(TARGET_DIR)/etc/modules.d/accessory
endef
```

I'm pretty sure I installed the latest floodlight firmware - I'm only seeing the camera... no serial usb in /dev/ttyUSB0

gtxaspec, 2025-04-23:
the kernel magic is in the fixups wihch are called when `BR2_PACKAGE_WYZE_ACCESSORY_FLOODLIGHT` is selected:

```
ifeq ($(BR2_PACKAGE_WYZE_ACCESSORY_FLOODLIGHT),y)
        WYZE_ACCESSORY_INSTALL_TARGET_CMDS += $(WYZE_ACCESSORY_INSTALL_TARGET_CMDS_FLOODLIGHT)
        WYZE_ACCESSORY_LINUX_CONFIG_FIXUPS += $(WYZE_ACCESSORY_LINUX_CONFIG_FIXUPS_FLOODLIGHT)
endif
```

and

```
define WYZE_ACCESSORY_LINUX_CONFIG_FIXUPS_FLOODLIGHT
        $(call KCONFIG_SET_OPT,CONFIG_USB_SERIAL,m)
        $(call KCONFIG_SET_OPT,CONFIG_USB_SERIAL_CH341,m)
        $(call KCONFIG_ENABLE_OPT,CONFIG_SND)
        $(call KCONFIG_ENABLE_OPT,CONFIG_SND_USB)
        $(call KCONFIG_ENABLE_OPT,CONFIG_USB_DWC2_FULLSPEED_HOST)
        $(call KCONFIG_SET_OPT,CONFIG_SND_USB_AUDIO,m)
endef
```

you can verify this on device by `zcat /proc/config.gz`  So you should not be seeing `[ 1.205959] dwc2 dwc2: Sorry, SPLIT transfer is not supported!`  If you can check the kernel config, otherwise i will verify later.

your right, it looks like we never included the floodlight ctrl script, i'll add it in today.

closed 2025-04-23 by juanboro

gtxaspec, 2025-04-23:
You can always change the device ID using the `gestalt` command on the device.

juanboro, 2025-04-23:
> You can always change the device ID using the `gestalt` command on the device.

```shell
 ls /dev/ttyUSB0
/dev/ttyUSB0
```

Yay - thanks again!

GovSat1, 2025-05-10:
> the floodlight specific release includes different kernel options to fix the issues with the USB hub in the unit. basic floodlight control is also available in thingino.

Can you clarify how the "basic floodlight control" works? Is it in the webui somewhere or are you saying it just allows the USB functionality?

Otherwise - for anyone confused (like me), you need this script -> https://github.com/gtxaspec/wz_mini_hacks/discussions/771#discussioncomment-11737172

Save it as `floodlight_ctl` in /usr/sbin (chmod +x to make it executable).

juanboro, 2025-05-10:
> Save it as `floodlight_ctl` in /usr/sbin (chmod +x to make it executable).
Agree with the comment.  Also - did you get this to work?  In my case - I can't seem to deactivate the automatic PIR sensor.  I can use floodlight_ctl to turn off the floodlight momentarily... but then the automatic PIR switches it right back on.  Since I'm currently physically not present at my location with the wyze floodlight - I just had someone turn the whole thing off for now so it stops wildly operating at night.  But my plan is once I get to it to just "physically" de-activate the PIR sensor.  I _think_ it should be possible to reverse engineer the protocol for controlling the sensor - but I have my hands busy with alot of other projects right now, so am not going to try to bother myself.

GovSat1, 2025-05-10:
> Agree with the comment. Also - did you get this to work? In my case - I can't seem to deactivate the automatic PIR sensor. I can use floodlight_ctl to turn off the floodlight momentarily... but then [...]

Yep - I went ahead and disconnected the PIR module while installing the floodlight. Haven't tested much yet but it seems to work fine so far. 

The plastic around the "globe" can be pried out, its just held in with a seal. Then 4 rubber screw covers at the corners, remove screws and pull the assembly out. I disconnected the PIR ribbon cable at the circuit board (standard flip up style connecter)

juanboro, 2025-09-13:
> > > the floodlight specific release includes different kernel options to fix the issues with the USB hub in the unit. basic floodlight control is also available in thingino.
> > 
> > 
> > Can you clarify how the "basic floodlight control" works? Is it in the webui somewhere or are you saying it just allows the USB functionality?
> > Otherwise - for anyone confused (like me), you need this script -> [gtxaspec/wz_mini_hacks#771 (reply in thread)](https://github.com/gtxaspec/wz_mini_hacks/discussions/771#discussioncomment-11737172)
> > Save it as `floodlight_ctl` in /usr/sbin (chmod +x to make it executable).
> 
> i cant seem to get the scrip to work any advice? to be honest running up against my programming limits and cant get the file created would be my guess

My advice would be to just ditch the Wyze floodlight cam and instead get a better one instead.  I swapped mine for a tplink tapo floodlight and it's worked great ever since (no thingino firmware unfortunately - but it has native onvif motion events and rtsp, so good enough).  With the Wyze -- you can only at best sort of get floodlight_ctl working anyway --- you'll need to hack the hardware to disable the PIR sensor... then you'll need to re-write the functionality to turn the thing on via motion events.  At least for me it really wasn't worth the effort and time for me versus just giving up and replacing it with something that just worked right out of the box.

ELY3M, 2026-02-25:
The floodlight script seem not working on my floodlight cam.    
light do work when you move front of the pir module.      

Here is my os-release 

root@Floodlight-Camera etc# cat os-release
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
TOOLCHAIN=musl
SOC=t31
SOC_ARCH=xburst1
IMAGE_ID=wyze_camfl1_t31x_gc2053_atbm6031
BUILD_ID="stable+d10a41a, 2026-02-23 09:24:55 +0000"
BUILD_TIME="2026-02-23 09:24:55 +0000"
COMMIT_ID="stable+d10a41a, 2026-02-23 09:07:09 +0000"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
HOSTNAME=ing-wyze-camfl1
TIME_STAMP=1771838695
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=2025.11.1
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.11.1
BUILDROOT_PRETTY_NAME="Buildroot 2025.11.1"