themactep/thingino-firmware #628: Remotely accessed thingino camera not showing/streaming video preview

jinverdale, 2025-04-22:
I have an old Neos Smartcam flashed with Thingino/Wyze firmware. It is working fine and I can access it locally on LAN and can see the Web UI and video. However, when I try and connect to it remotely (on WAN, public IP of my router and port 80 or 553) it gives me the correct Web UI but no video. I can see/access all other settings apart from video preview - broken camera image appears.
To connect to this cam remotely I am using my phone on the 4G rather than wifi.

I have tried talking to my ISP, thinking that ports 80 and 553 were blocked or not allowed to stream video. but they say they are. A port check verifies these as open. Is there another port I need to open? I cannot see anywhere on the settings what port the Cam is supposed to be using, so just assumed ports 80 and 553.
Any help wold be much appreciated 

thanks
jin

daoudeddy, 2025-04-22:
Preview is a websocket as far as i know
@themactep correct me if im wrong

themactep, 2025-04-22:
yes, you need access to port 8090

themactep, 2025-04-22:
Exposing camera directly to the internet is a very bad idea. Use a VPN on your router to login to your home network. Also, use a mobile app like tinycam monitor to monitor cameras. Web UI is for settings, not monitoring.

jinverdale, 2025-04-22:
Thanks for the advice themactep, will try port 8090 and then have a go at adding my cam to tinycam for monitoring. Understand about exposing camera to the internet without a VPN. Unfortunately it looks like I cannot install a VPN on my router, it won't allow it - there's no settings page available for me to do it.

Many thanks all

gtxaspec, 2025-04-23:
you can install a VPN (wireguard) on the camera, and have it connect to an external VPN service/server.  Just an idea.

jinverdale, 2025-04-23:
> you can install a VPN (wireguard) on the camera, and have it connect to an external VPN service/server. Just an idea.

Thank you. I have been attempting to install Wireguard on the camera today but it's not as easy as I thought it'd be. This is my first try with Wireguard, so will research online and see how others have done it.

themactep, 2025-04-24:
i guess this had drifted away from the issue. maybe convert it to a discussion?