themactep/thingino-firmware #633: onboard person detection

nijhawank, 2025-04-26:
So it seems the T31x chip inside WyzeCam v3 is capable of onboard person detection as other cameras using the same chip (example Eufy) do that.

Would it be possible to enable onboard person detection on Wyze Cam v3?

daoudeddy, 2025-04-26:
it is possible only by modifying the prudynt-t binary to include the libraries provided by Ingenic SDK, i did that [before](https://github.com/daoudeddy/prudynt-t) but it will become very large it will only work on 16M or higher

nijhawank, 2025-04-26:
The Wyze Cam v3 is 16mb. If you could provide the steps, I could try to play or I could try a beta test

daoudeddy, 2025-04-26:
tbh i switch to [frigate](https://github.com/blakeblackshear/frigate) and archived the repo, you could see the changes in the commits and try to build it yourself

lcwsoft, 2025-05-14:
This would be awesome to get in place for at least the Wyze Cam v3 builds. Currently I'm looking to use 3 of these cameras with the ONVIF integration in HomeAssistant. Having the camera do the person detection would cut down on the number of snapshots my automations have to take each day and also the number of requests I am sending to Google Gemini AI to get a description of the motion. I'm using the snapshots and AI responses for better notifications, but hit up against Google API limits due to having to rely on motion events in general, which seem to be quite easy to trigger at this time (even with the sensitivity set on min).