themactep/thingino-firmware #634: Difficulty setting up with NEOS SmartCam

Quagmire312, 2025-04-26:
Hi,

I have been having issues setting up Thingino with an old NEOS smart cam. Whenever I put in the SD card with the files attached to it, and power on the camera holding reset, the colours go as follows

solid yellow as it first powers on
solid blue for 5-10 seconds
then back to solid yellow.

The video tutorials suggest waiting up to 6 minutes, but even after 15, the yellow colour did not start flashing between blue and yellow

Any ideas what could be causing this?

Additionally, I have tried using the cloner to directly upload to the NEOS camera, subsituting the USB A-A cable with a USB A-C cable since only data transfer is needed, however my computer doesn't even detect the device as connected. Any ideas as to what might be causing this, or if its an issue with the program itself?

virmaior, 2025-04-28:
the cloner is a bit complicated to use and will only work when the camera is in special mode.

the rest I'm not so sure of.

Was the camera on stock firmware before?

themactep, 2025-05-06:
Do not substitute, follow to the T.