themactep/thingino-firmware #636: MAC address cloning, or camera acting as both AP and client simultaneously and bridging traffic

jspiros, 2025-04-28:
I've been converting Cinnado D1 cameras with T23 chips to Thingino the last few days, using the SD card image installer built by @wltechblog (specifically [this one](https://github.com/wltechblog/thingino-installers/blob/7c60fa75689cde2f15d89e0c7e7f7e9117c4328e/cinnado-d1/cinnado-d1-t23.zip)). What I describe here has applied to each of the six cameras that I've flashed and configured so far.

After flashing and rebooting with the microSD card removed, I connect using my iPhone to set hostname, root password, and wifi configuration. At the end of that process, it correctly displays the MAC address of the camera in question. I quickly create a DHCP configuration entry on my DHCP server before it reboots, and, in the end, it all works fine. The camera eventually gets the intended IP, and works well.

However, as I was watching the DHCP server logs while the camera came back online for its first connection attempt to my network, I notice an unfamiliar MAC sending DHCP discover requests. Here's what I typically see, this being a log of the most recent one I converted (note: MAC addresses have been anonymized):

```
Apr 28 16:41:43 DHCPDISCOVER 8e:4a:8c:0b:44:3f
Apr 28 16:41:43 DHCPOFFER 192.168.165.153 8e:4a:8c:0b:44:3f

Apr 28 16:41:43 DHCPDISCOVER 8e:4a:8c:0b:44:3f
Apr 28 16:41:43 DHCPOFFER 192.168.165.153 8e:4a:8c:0b:44:3f

Apr 28 16:41:46 DHCPDISCOVER 8e:4a:8c:0b:44:3f
Apr 28 16:41:46 DHCPOFFER 192.168.165.153 8e:4a:8c:0b:44:3f

Apr 28 16:41:52 DHCPDISCOVER 02:76:c5:93:53:84
Apr 28 16:41:52 DHCPOFFER 192.168.42.92 02:76:c5:93:53:84
Apr 28 16:41:52 DHCPREQUEST 192.168.42.92 02:76:c5:93:53:84
Apr 28 16:41:52 DHCPACK 192.168.42.92 02:76:c5:93:53:84 cinnado-d1-06

Apr 28 16:41:55 DHCPDISCOVER 8e:4a:8c:0b:44:3f
Apr 28 16:41:55 DHCPOFFER 192.168.165.153 8e:4a:8c:0b:44:3f

Apr 28 16:42:03 DHCPDISCOVER 8e:4a:8c:0b:44:3f
Apr 28 16:42:03 DHCPOFFER 192.168.165.153 8e:4a:8c:0b:44:3f
```

I don't see this upon futher boots of the camera, only the first time it's connecting to my network. The unrecognized MAC would never respond to the DHCP offer with a DHCP request, so it would never obtain a lease.

I looked for other issues before reporting this, and took note of issue #426. That person indicated that the device that was connecting to the AP mode camera had its MAC address being reused by the camera. In my case, though, it was a different MAC each time, and oddly enough, they were MAC addresses that do not have registered OUI components, suggesting they're fake. But then I realized that my phone, upon connecting to new networks, will use a different randomized MAC address for each SSID, and each camera had a unique SSID.

So, although it's not the same as what I describe here, as their final installer screen displaying the MAC address was displaying the wrong MAC address and my final installer screens were correctly showing the `02:[...]` MAC address, it seems noteworthy that in both cases the incorrect MAC involved seems to be borrowed from the client device that previously connected to the camera acting as an AP.

I paid closer attention this most recent time, and can confirm that the unrecognized MAC corresponded to the one used by my phone when connecting to the AP mode camera. But I can also confirm that, for some reason, my phone still seemed to be connected to the AP mode camera for a few seconds even after the camera obtained its IP as a client on my network.

So, my guess is that either (a) the camera is for some reason using the MAC address of the client that had connected to it while it was in AP mode when it connects as a client itself, and somehow realizes its mistake before actually obtaining a lease, OR (b) since two more DHCP discover packets are seen after the lease is obtained, maybe my phone is still connected to the camera acting as AP even as the camera is simultaneously acting as a client, and the camera is forwarding my phone's DHCP requests on to my network's DHCP server.

Either way, ideally, I'd not expect either of those things to be happening.

I should be getting a few more of these in coming days, if there's anything anyone wants me to check. At the very least, I plan to try deactivating wifi on my phone as soon as the camera reboots, to potentially rule out possibility (b).

virmaior, 2025-04-29:
are the mac addresses sequential or nearly sequential or are they as random as what you're putting up in your anonymized example?

jspiros, 2025-04-29:
The random MAC addresses my phone generates per-SSID are seemingly entirely random, no apparent relationship to each other. The MAC addresses for the Cinnado D1 cameras all start with the `02:` byte but are otherwise non-sequential, with most of the other bytes different. There are two other patterns I see: the penultimate byte (so, the `00` in `ff:ff:ff:ff:00:ff`) is the same for each one as well, and the fourth byte (so the one before the penultimate has the first digit the same, with the second digit varying, except one where two cameras have the same fourth byte entirely.

virmaior, 2025-04-29:
Er, so the claim is that the cinnado camera is basically cloning the randomized mac of your phone?

something like 

_Cinnado as AP_
```
Cinnado = 02:be:ef:ca
Phone = fe:fe:fe:01
```

_Your Network Router Moment 1_
```
Cinnado =  fe:fe:fe:01
acquires IP 192.168.165.80 via DHCP

```

_Your Network Router Moment 2_
```
Cinnado =  02:be:ef:cb
acquires IP 192.168.165.100 via DHCP
```

?

jspiros, 2025-04-30:
Not quite, because the random MAC (`8e:4a:8c:0b:44:3f` in the example log provided) only sends out DHCP discover packets, it never sends out a DHCP request packet in response to the server's offer, so it never obtains a lease. `8e:4a:8c:0b:44:3f` is the MAC my phone generated exclusively for that Cinnado while it was in AP mode upon first boot, and the Cinnado's MAC is `02:76:c5:93:53:84`. Here's what I see, step by step, using the same MAC addresses as in the log I provded above:

1. Cinnado boots for the first time, and establishes access point `THINGINO-1234` (for example).
2. Phone generates MAC `8e:4a:8c:0b:44:3f` to connect to `THINGINO-1234` SSID, and connects.
3. I visit `172.16.0.1` using the Phone, which displays the Cinnado's Thingino configuration/installer page.
4. I configure the wifi credentials for my network, and save.
5. The configuration tool displays `02:76:c5:93:53:84` as the Cinnado's MAC address, so I quickly add that to my DHCP server, and reload the DHCP server config.
6. I watch the DHCP server's log as I wait for the Cinnado to boot.
7. I see the log above, which specifically shows:
   1. The DHCP server sees `8e:4a:8c:0b:44:3f` requesting an IP, and it offers one. The offer is not accepted. This repeats two more times.
   2. The DHCP server sees `02:76:c5:93:53:84` request an IP, offers it, the offer is accepted, and the Cinnado is now on my network with its configured IP.
   3. Two more times, `8e:4a:8c:0b:44:3f` requests an IP, is offered one, and doesn't accept.

It seems that the Cinnado is the source of the DHCP discover packets from both MAC addresses, from the perspective of my network's AP and DHCP server. I don't believe my phone is trying to connect to my network using `8e:4a:8c:0b:44:3f`, as the whole point of the randomized MAC addresses is to make correlating the same client between networks impossible. So, it seems most likely to me that one of these two things is happening:

1. The Cinnado is itself sending the discover packets from both MAC addresses, but is only paying attention to the responses coming in for its `02:76:c5:93:53:84` MAC address.
2. The Cinnado is not sending the discover packets from `8e:4a:8c:0b:44:3f`, but my phone is, and the Cinnado still has its AP mode enabled while it's connecting as a client to my network, which my phone is connected to. The Cinnado is acting as if it's an AP on my network, while also being a client, like a "WiFi Extender". Though, if this is the case, it's not a perfect bridge, as it doesn't seem likely that it's forwarding the DHCP offers back to my phone, as otherwise why isn't my phone accepting it?

I'm not sure which is more likely, but I assume either one wouldn't be desirable.

I got more cameras in today, so I'll look into this in a little bit, either today or tomorrow. I'll try turning my phone's wifi off immediately upon saving the credentials, so that when the Cinnado reboots, there's no chance my phone will be connected/reconnect to the Cinnado if it is indeed still acting as an AP.  If I still see the DHCP discover packets coming in from two MACs at that point, we'll know that it's entirely coming from the Cinnado on its own. Perhaps I'll also use a third device to scan for WiFi APs continuously, to confirm whether the Cinnado is offering an AP simultaneously as it acts as a client on my network.

nicolasroche898, 2025-04-30:
You're using an iPhone... iPhones (and also some other Apple products) have a quirk where they send DHCP requests of the last few networks they were connected to, into the the new network it's connected right now, as a way to speed up connectivity (and contrary to standard protocol). I don't see how it's relevant to Thingino.

jspiros, 2025-04-30:
> You're using an iPhone... iPhones (and also some other Apple products) have a quirk where they send DHCP requests of the last few networks they were connected to, into the new network it's connected right now, as a way to speed up connectivity (in complete violation to all standard protocols). I don't see how it's relevant to Thingino.

Yep, I came to that same conclusion last night. As I planned, I disabled wifi on my phone immediately to see if the behavior changed, and it did, indicating the phone was sending the DHCP discover packets. Oddly, the phone would still sometimes show itself as connected to the `THINGINO-1234` SSID at the moment those DHCP discover packets were coming in, so it still seemed like the camera was potentially acting as an "extender". But, I decided to monitor my AP's logs at the same time to be sure, and indeed, my phone was connecting to my AP, using the correct MAC for that SSID, but then sending the DHCP discovers using the MAC it had been using for the `THINGINO-1234` SSID.

I had considered that the least likely option, as the MAC randomization feature exists precisely so your activity can't be correlated between networks, but this behavior makes it possible to do exactly that.

It was pretty late, so I decided to wait until today when I could better check to see if this was known behavior or not, as it seems like a potential privacy issue and I wanted to report it to Apple if it was otherwise unknown. But, as you know about it, I won't worry about that. Still, given that it undermines that feature, I'm baffled, it almost seems intentional.

I do wonder if issue #426 might be the same issue, but what the reporter in that issue described (the randomized MAC being confirmed while SSH'd in to the camera running commands) seems to preclude it. At any rate, if it weren't for issue #426, I would have been more skeptical of the notion that Thingino might have had anything to do with this, so if anyone else is reading this in a similar situation, make sure it's not your Apple device acting weird.

Anyway, sorry for the trouble! Issue closed.

closed 2025-04-30 by jspiros