themactep/thingino-firmware #637: /etc/os-release does not update automatically..

matiasdelellis, 2025-04-29:
I updated two cameras to the latest firmware and they report a firmware hash from March 12.

```
root@192.168.1.90's password: 

   \\   _______ _     _ _____ __   _  ______ _____ __   _  _____
   )\\     |    |_____|   |   | \  | |  ____   |   | \  | |     |
  (  /     |    |     | __|__ |  \_| |_____| __|__ |  \_| |_____|
  / /
                             personal_campan_t31x_gc2053_atbm6031
                        master+34482f3, 2025-03-12 09:13:36 +0000

                                       02:6b:98:68:66:dc as wlan0
                                   fe80::6b:98ff:fe68:66dc (IPv6)
                                              192.168.1.90 (IPv4)

              17:37:22 up 15 min,  load average: 1.89, 1.83, 1.18

root@thingino-interior ~# cat /etc/os-release 
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
IMAGE_ID=personal_campan_t31x_gc2053_atbm6031
BUILD_ID="master+34482f3, 2025-03-12 09:13:36 +0000"
BUILD_TIME="2025-03-12 09:13:36 +0000"
COMMIT_ID="master+34482f3, 2025-03-12 07:42:45 +0000"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
HOSTNAME=ing-personalcam-
TIME_STAMP=1733653000
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-gb4758c95
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"
```

I wasted a lot of time thinking it was failing to update, but it seems to actually be updated since for example I have the latest changes to prudynt.. 😅

closed 2025-05-07 by themactep

matiasdelellis, 2025-05-08:
Hi, I understand that you don't want to touch the overlay to keep it network, preferences and and everything else, But shouldn't the user see the current firmware version somewhere?

gtxaspec, 2025-05-08:
so long as `os-release` does not exist in overlay, it will always pull the latest one from `/rom`