themactep/thingino-firmware #640: Stream regularly crashes when used with Frigate

benjani85, 2025-05-01:
I have a Wyze Cam V3 with Thingino firmware which is stable until I start using the RTSP stream with Frigate, causing the stream to crash anywhere between a few minutes to a few hours. Restarting the stream resolves the issue temporarily until the next crash happens.

I only recently moved over to Thingino so should be running the latest version of the firmware: master+55a3254, 2025-04-27 09:46:35 +0000

The stream configuration settings are running with the defaults but I have tried playing around with the different modes to no effect.

Did anyone else encounter this?

alan5ab, 2025-05-01:
Are you using timeout=30? There is a frigate section in the wiki.

davidebeatrici, 2025-05-01:
I've been fighting with the issue ever since I set up Frigate, thinking it was a bug of that.

For reference:

https://github.com/blakeblackshear/frigate/discussions/17905
https://github.com/AlexxIT/go2rtc/issues/1713

I'm going to follow the guide in the wiki and check if anything changes.

davidebeatrici, 2025-05-01:
No luck unfortunately...

```
2025-05-01 21:27:36.745143455  [INFO] Preparing Frigate...
2025-05-01 21:27:36.856906352  [INFO] Starting Frigate...
2025-05-01 21:27:39.862611469  [2025-05-01 23:27:39] frigate.util.config            INFO    : Checking if frigate config needs migration...
2025-05-01 21:27:39.893793827  [2025-05-01 23:27:39] frigate.util.config            INFO    : frigate config does not need migration...
2025-05-01 21:27:39.951610170  [2025-05-01 23:27:39] frigate.util.services          WARNING : Did not detect hwaccel, using a GPU for accelerated video decoding is highly recommended
2025-05-01 21:27:42.371273166  [2025-05-01 23:27:42] frigate.app                    INFO    : Starting Frigate (0.15.0-cea210d)
2025-05-01 21:27:42.387413195  [2025-05-01 23:27:42] peewee_migrate.logs            INFO    : Starting migrations
2025-05-01 21:27:42.388348357  [2025-05-01 23:27:42] peewee_migrate.logs            INFO    : There is nothing to migrate
2025-05-01 21:27:42.406326034  [2025-05-01 23:27:42] frigate.app                    INFO    : Recording process started: 379
2025-05-01 21:27:42.406938993  [2025-05-01 23:27:42] frigate.app                    INFO    : Review process started: 380
2025-05-01 21:27:42.413155914  [2025-05-01 23:27:42] frigate.app                    INFO    : go2rtc process pid: 96
2025-05-01 21:27:42.433398191  [2025-05-01 23:27:42] detector.coral                 INFO    : Starting detection process: 395
2025-05-01 21:27:45.046733665  [2025-05-01 23:27:42] frigate.app                    INFO    : Output process started: 414
2025-05-01 21:27:45.054599826  [2025-05-01 23:27:42] frigate.detectors.plugins.edgetpu_tfl INFO    : Attempting to load TPU as usb
2025-05-01 21:27:45.055114256  [2025-05-01 23:27:42] frigate.app                    INFO    : Camera processor started for cam1: 438
2025-05-01 21:27:45.055304705  [2025-05-01 23:27:45] frigate.detectors.plugins.edgetpu_tfl INFO    : TPU found
2025-05-01 21:27:45.055788462  [2025-05-01 23:27:42] frigate.app                    INFO    : Camera processor started for cam2: 443
2025-05-01 21:27:45.055857633  [2025-05-01 23:27:42] frigate.app                    INFO    : Capture process started for cam1: 445
2025-05-01 21:27:45.055959782  [2025-05-01 23:27:42] frigate.app                    INFO    : Capture process started for cam2: 465
2025-05-01 21:27:45.056031102  [2025-05-01 23:27:42] frigate.api.fastapi_app        INFO    : Starting FastAPI app
2025-05-01 21:27:45.056114996  [2025-05-01 23:27:42] frigate.api.fastapi_app        INFO    : FastAPI started
2025-05-01 21:28:12.596770077  [2025-05-01 23:28:12] watchdog.cam2                  INFO    : No frames received from cam2 in 20 seconds. Exiting ffmpeg...
2025-05-01 21:28:12.597072806  [2025-05-01 23:28:12] watchdog.cam2                  INFO    : Waiting for ffmpeg to exit gracefully...
2025-05-01 21:28:12.960579256  [2025-05-01 23:28:12] frigate.video                  ERROR   : cam2: Unable to read frames from ffmpeg process.
2025-05-01 21:28:12.961476796  [2025-05-01 23:28:12] frigate.video                  ERROR   : cam2: ffmpeg process is not running. exiting capture thread...
2025-05-01 21:28:22.572689256  [2025-05-01 23:28:22] watchdog.cam1                  INFO    : No frames received from cam1 in 20 seconds. Exiting ffmpeg...
2025-05-01 21:28:22.573075791  [2025-05-01 23:28:22] watchdog.cam1                  INFO    : Waiting for ffmpeg to exit gracefully...
2025-05-01 21:28:22.943331874  [2025-05-01 23:28:22] frigate.video                  ERROR   : cam1: Unable to read frames from ffmpeg process.
2025-05-01 21:28:22.944331053  [2025-05-01 23:28:22] frigate.video                  ERROR   : cam1: ffmpeg process is not running. exiting capture thread...
2025-05-01 21:28:22.971180864  [2025-05-01 23:28:22] watchdog.cam2                  ERROR   : Ffmpeg process crashed unexpectedly for cam2.
2025-05-01 21:28:22.971266627  [2025-05-01 23:28:22] watchdog.cam2                  ERROR   : The following ffmpeg logs include the last 100 lines prior to exit.
2025-05-01 21:28:22.971347435  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [h264 @ 0x555bffb9a1c0] error while decoding MB 101 34, bytestream -25
2025-05-01 21:28:22.971427842  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [vost#0:0/copy @ 0x555bffbcab00] Non-monotonic DTS; previous: 2388, current: 2388; changing to 2389. This may result in incorrect timestamps in the output file.
2025-05-01 21:28:22.971488813  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [h264 @ 0x555bffbcc500] error while decoding MB 101 34, bytestream -25
2025-05-01 21:28:22.971554897  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [vist#0:0/h264 @ 0x555bffbb8440] [dec:h264 @ 0x555bffbcaf80] corrupt decoded frame
2025-05-01 21:28:22.971630761  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [vost#0:0/copy @ 0x555bffbcab00] Non-monotonic DTS; previous: 21893, current: 21893; changing to 21894. This may result in incorrect timestamps in the output file.
2025-05-01 21:28:22.971686524  [2025-05-01 23:28:22] ffmpeg.cam2.detect             ERROR   : [vost#0:0/copy @ 0x555bffbcab00] Non-monotonic DTS; previous: 21894, current: 21894; changing to 21895. This may result in incorrect timestamps in the output file.
[... cut ...]
2025-05-01 21:31:03.660485942  [2025-05-01 23:31:03] ffmpeg.cam2.detect             ERROR   : [vost#0:0/copy @ 0x6369236d16c0] Non-monotonic DTS; previous: 84576, current: 84576; changing to 84577. This may result in incorrect timestamps in the output file.
2025-05-01 21:31:03.660549366  [2025-05-01 23:31:03] ffmpeg.cam2.detect             ERROR   : [vost#0:0/copy @ 0x6369236d16c0] Non-monotonic DTS; previous: 84577, current: 84576; changing to 84578. This may result in incorrect timestamps in the output file.
2025-05-01 21:31:03.660621474  [2025-05-01 23:31:03] ffmpeg.cam2.detect             ERROR   : [rtsp @ 0x6369236921c0] CSeq 6 expected, 0 received.
```

alan5ab, 2025-05-01:
Some think running go2rtc on the camera sd card helps, I run it on all of mine. It also has a section in the wiki

davidebeatrici, 2025-05-01:
Both cameras are wired though... ironically at another location I have a Wi-Fi camera running OpenIPC which is working flawlessly with Frigate.

All of them are Ingenix T31X.

alan5ab, 2025-05-01:
Do the streams show errors on vlc

davidebeatrici, 2025-05-02:
Just checked. Audio is intermittent and sometimes a proper single frame is shown.

It wasn't that bad until yesterday and I've already rebooted both cameras... not sure what changed.

gtxaspec, 2025-05-03:
if you disable audio from all streams, does the problem still occur?

davidebeatrici, 2025-05-03:
The substream is disabled on both cameras.

I disabled audio, unfortunately there appears to be no change whatsoever.

benjani85, 2025-05-03:
It tried leaving the stream up without Frigate and it turns out the stream still crashes.

I did see this in the logs but not sure if it is related:

```
[218896.486685] gc2053 stream on
[218896.615271] cgu_set_rate, parent = 1392000000, rate = 12288000, n = 3625, reg val = 0x22000e29
[218896.615283] codec_codec_ctl: set sample rate...
[218896.615582] codec_codec_ctl: set device...
[218897.073918] codec_set_device: set device: MIC...
[251816.852228] codec_codec_ctl: set CODEC_TURN_OFF...
[251817.044479] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[251817.044479] 00000018 
[251817.044501] epc = 771d7920 in libc.so[77165000+b8000]
[251817.044524] ra  = 771d7904 in libc.so[77165000+b8000]
[251817.044534] 
[251817.143566] gc2053 stream off
```

```
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource audio destructed, encoder channel:0
May  3 20:21:02 prudynt[26859]: [DEBUG:RTSP.cpp]: Stop RTSP Server.
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:0
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
May  3 20:21:02 prudynt[26859]: [DEBUG:main.cpp]: join rtsp thread
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPAudio.cpp]: IMPAudio::deinit()
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPAudio.cpp]: IMP_AENC_UnRegisterEncoder(&handle)
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPAudio.cpp]: IMP_AENC_DestroyChn(0)
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPAudio.cpp]: IMP_AI_DisableChn(1, 0)
May  3 20:21:02 prudynt[26859]: [DEBUG:IMPAudio.cpp]: IMP_AI_Disable(1)
May  3 20:21:02 prudynt[26859]: [DEBUG:AudioWorker.cpp]: AudioWorker destroyed for channel 0
```

trickk1337, 2025-05-03:
Hi,

I've encountered similar issues, but I strongly suspect this isn't related to Frigate. Instead, check your camera's uptime—some newer firmware versions have problems with nightly crontab tasks.

You might want to verify whether you can manually toggle night mode or IR LEDs on/off. In my case, my cameras restart after these changes, and occasionally, they fail to reconnect until physically power-cycled. This behavior aligns with what you're describing in Frigate.

I deactivated nightly crontab, afterwards the cams are stable since 3 days

davidebeatrici, 2025-05-04:
No difference here.

alan5ab, 2025-05-04:
> No difference here.

What wired cams are you trying to get to work?

davidebeatrici, 2025-05-05:
Both are Enzhi/Vanhua Z55.

alan5ab, 2025-05-05:
I have 3 of those, never experienced any streaming issues. Still on older firmware though.

benjani85, 2025-05-05:
My cam is wireless however didn't seem to have streaming issues when using the Wyze firmware. The signal looks good based on what my router is reporting. It looks like you can add PoE to the camera but no point is doing so if the issue is related to something else.

gtxaspec, 2025-05-05:
how soon after booting the cam does it crash? can you share full diagnostic logs?

benjani85, 2025-05-05:
Anywhere between 5 minutes to a few hours.

https://tb.thingino.com/nbk5/

davidebeatrici, 2025-05-14:
The streamer has been much more stable for the past few days, without me touching anything.

Errors in Frigate's log are now sporadic and the last happened 5 hours ago!

davidebeatrici, 2025-05-14:
And you didn't perform any changes either?

lcwsoft, 2025-05-19:
I'm also noticing the same type of issue with HomeAssistant on all of my wyze cam v3s (all t31x). If I start a live view from the preview on the main dashboard (cameras added with RTSP through the ONVIF integration), the stream will run for anywhere between 15 seconds to a few minutes, then lock up. Exiting and starting again works. From what I can see, HA it also using ffmpeg for handling the streams, so this could be related.

davidebeatrici, 2025-05-20:
I can confirm that as well.

SpawnSP2, 2025-05-22:
Same problem here. I have a Xiaomi 2K magnetic mount integrated into Home Assistant via Generic Camera integration. I have an automation that receives the MQTT topic that I've configured on the camera when motion is detected, and then records a 30 second video (probably using ffmpeg).

Eventually, the stream goes down. On dmesg there is this error (posting two different times the crash occured):

```[ 9873.241247] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[ 9873.241247] 682e3e02 
[ 9873.241268] epc = 77af0bc8 in libc.so[77a85000+b8000]
[ 9873.241301] ra  = 77a38634 in libimp.so[77975000+107000]
[ 9873.241316] 
[ 9873.352967] jxq03p stream off
[ 9873.513743] codec_codec_ctl: set CODEC_TURN_OFF...

[27155.379936] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[27155.379936] 00000000 
[27155.379958] epc = 00000000 inra  = 77062780 in libimp.so[7701f000+107000]
[27155.379992] 
[27155.501223] jxq03p stream off
[27155.739238] codec_codec_ctl: set CODEC_TURN_OFF...
```

Also, on logread there are those 3 events, repeating forever every 10 seconds:

```
May 22 21:18:02 prudynt[2025]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
May 22 21:18:02 prudynt[2025]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
May 22 21:18:02 prudynt[2025]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
```

If I disable the secondary stream, those repeating messages go away, but then motion detection doesn't work. It makes sense that motion detection uses the lower res secondary stream, in order to save resources.

davidebeatrici, 2025-05-30:
Updated both cameras to the latest firmware, the issue is still there.

On a fresh restart the RTSP stream is completely unusable, after a few minutes it becomes somewhat stable... until it crashes a few days later.

gtxaspec, 2025-06-05:
> Yeah, I can confirm... please note that the RTSP stream is also severely unstable when connecting directly through media players such as VLC and MPV.
> 
> The preview on Thingino's web UI often shows the broken camera image instead of the stream.

Can you share logs (or confirm) which model soc you have on your z55?

davidebeatrici, 2025-06-06:
```
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
TOOLCHAIN=musl
SOC=t31
SOC_ARCH=xburst1
IMAGE_ID=vanhua_z55_t31x_gc4653_eth
BUILD_ID="master+3ca64c8, 2025-05-30 09:11:06 +0000"
BUILD_TIME="2025-05-30 09:11:06 +0000"
COMMIT_ID="master+3ca64c8, 2025-05-29 19:42:53 +0000"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
HOSTNAME=ing-vanhua-z55
TIME_STAMP=1742234000
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-ge9be0d14
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.05-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.05-git"
```

Here's the `dmesg` from both cameras:

```
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (buildroot@buildroot) (buildroot-gcc-14.2.0) #1 PREEMPT Mon Mar 17 17:53:20 UTC 2025
[    0.000000] CPU0 RESET ERROR PC:801ADC20
[    0.000000] [<801adc20>] 0x801adc20
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0039f000 @ 00010000 (usable)
[    0.000000]  memory: 00031000 @ 003af000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803aaa70, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 16256
[    0.000000] Kernel command line: mem=64M@0x0 rmem=64M@0x4000000 console=ttyS1,115200n8 panic=10 root=/dev/mtdblock4 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),32k(env),224k(config),1248k(kernel),4736k(rootfs),-(rootfs_data),15872k@0x80000(upgrade),16384k@0(all)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 60564k/65536k available (3090k kernel code, 4972k reserved, 617k data, 196k init, 0k highmem)
[... 243 lines cut ...]
[    5.490760] cgu_set_rate, parent = 1392000000, rate = 2048000, n = 10875, reg val = 0x01002a7b
[    5.490767] cgu_enable,cgu_i2s_mic reg val = 0x21002a7b
[    5.893311] dma dma0chan24: Channel 24 have been requested.(phy id 7,type 0x06 desc a239d000)
[    5.893608] dma dma0chan25: Channel 25 have been requested.(phy id 6,type 0x06 desc a127f000)
[    5.893932] dma dma0chan26: Channel 26 have been requested.(phy id 5,type 0x04 desc a239c000)
[    5.921982] @@@@ avpu driver ok(version H20220825a) @@@@@
[    6.118454] @@@@ tx-isp-probe ok(version H20221206a), compiler date=Dec  6 2022 @@@@@
[    6.537205] Bus Mode Reg after reset: 0x00020101, cnt=0
[    8.803241] libphy: 0:00 - Link is Up - 10/Full
[   10.651268] codec_codec_ctl: set repaly channel...
[   10.651307] cgu_set_rate, parent = 1392000000, rate = 4096000, n = 10875, reg val = 0x22002a7b
[   10.651313] codec_codec_ctl: set sample rate...
[   10.651397] codec_codec_ctl: set device...
[   10.883123] codec_set_device: set device: speaker...
[   11.363710] -----sensor_detect: 1057 ret = 0, v = 0x46
[   11.364322] -----sensor_detect: 1065 ret = 0, v = 0x53
[   11.364330] gc4653 chip found @ 0x29 (i2c0)
[   11.364330]  sensor drv version H20211211a
[   11.623769] gc4653 stream on
[   11.712587] cgu_set_rate, parent = 1392000000, rate = 12288000, n = 3625, reg val = 0x22000e29
[   11.712600] codec_codec_ctl: set sample rate...
[   11.712923] codec_codec_ctl: set device...
[   12.163093] codec_set_device: set device: MIC...

```

gtxaspec, 2025-06-06:
which chip version is physically in the device? can you run `soc`

davidebeatrici, 2025-06-06:
I see `t31x` for both.

gtxaspec, 2025-06-06:
Thx, I have a matching module. I'll see if I can reproduce any issues locally

davidebeatrici, 2025-06-06:
Same 48V PoE switch for both, power appears to be stable.

As for prudynt, these are my current settings:

[image]
[image]
[image]

Substream is disabled.

davidebeatrici, 2025-06-06:
I can't at the moment because the cameras are at a remote location, but once I'm there I can try to power at least one of the two directly using a 12V power supply.

I would expect the camera to fully crash and reboot if unstable power was the issue, though.

alan5ab, 2025-06-07:
Doubt it's power either, just the process of elimination. Could be network issues, You can try dropping the frame rate to 5 and the the bitrate to 200 (or as low as you can go)  and see  if the rtsp stream is more solid.  How long did  you bench test these cameras before installing them?

davidebeatrici, 2025-06-07:
Not too long, honestly. However, at the beginning I only had one camera and it was stable.

I then added a second one with a newer version of Thingino and upgraded the hardware Frigate runs on. I'm pretty sure both cameras were working reliably until I updated the firmware.

alan5ab, 2025-06-08:
> Yeah, I can confirm... please note that the RTSP stream is also severely unstable when connecting directly through media players such as VLC and MPV.
> 
> The preview on Thingino's web UI often shows the broken camera image instead of the stream.

Those are symptoms of network problems, the camera pics you posted are the same as the cameras I have.

alan5ab, 2025-06-10:
benjani85 on May 3:

> It tried leaving the stream up without Frigate and it turns out the stream still crashes.
>
> I did see this in the logs but not sure if it is related:
>
> ```
> [218896.486685] gc2053 stream on [...]

I recently encountered this problem except I got a stream stop. In my case it was triggered by weak wifi strength. I had just mounted the camera, adjusted the settings and attached it to the NVR. All was good for an hour until I realized it was running last years firmware, installed current firmware and the streamer would not come back on. 
  17.905426] do_page_fault() #2: sending SIGSEGV to Encoder-0 for invalid read access from
[   17.905426] 00000000 
[   17.905444] epc = 77beab1c in libimp.so[77ba4000+ef000]
[   17.905472] ra  = 77beab10 in libimp.so[77ba4000+ef000]
[   17.905484] 
[   17.943351] sc2336 stream off
[   18.136698] wait stop  q->num_buffers=2 q->done_count=1
[   18.328680] [atbm_log]:tso xmit(2962)
[   18.536678] streamoff stop！！
[   18.556661] wait stop  q->num_buffers=2 q->done_count=1
[   18.956711] streamoff stop！！

alan5ab, 2025-06-10:
Yes galayou - IMAGE_ID=galayou_y4_t23n_sc2336_atbm6062
BUILD_ID="master+41f8330, 2025-06-08

Moved the camera 2 feet horizontal, went from -62dbm to -48dbm and it has been OK.

lcwsoft, 2025-06-12:
I'm using Wyze v3s with Frigate too (configured with go2rtc as per the thingino wiki) and noticed that the streams are unstable in general (live view cutting out within seconds sometimes). At first I suspected a network issue, but I have other devices within a few feet of the camera and they can get 300+ up and down on speedtest (.net). Out of ideas at the time, I changed the main stream to be 1280 x 720 instead of the default 1920 x 1080 (everything else, including the bitrate, remains unchanged). So far, it's been over 20 minutes and working good.

alan5ab, 2025-06-12:
lcwsoft: try go2rtc on the camera end and see if there is an improvement.

boereworswizard, 2025-06-14:
I am also using Wyze Cam V3 with constant errors in frigate. I am going to give the go2rtc SD card method a go.

boereworswizard, 2025-06-14:
> The sd card method takes about 5 minutes to try. Your camera url will be the same except the port will be 8554.https://github.com/themactep/thingino-firmware/wiki/go2rtc

Thanks, I managed to get go2rtc running on the camera. I am just not sure how to update my frigate config accordingly (there are a lot of different streams available). Can you let me know if I've done it right?

I removed the camera from the go2rtc section and updated the camera paths to the following:

cameras:
  driveway_camera:
    ffmpeg:
      inputs:
        - path: rtsp://thingino:thingino@192.168.3.14:8554/ch0?mp4
          roles:
            - record
        - path: rtsp://thingino:thingino@192.168.3.14:8554/ch1
          input_args: preset-rtsp-restream
          roles:
            - detect

duggasco, 2025-06-14:
getting pretty frequent instability from go2rtc from pre-compiled image... also getting i/o timeout errors pretty frequently.

alan5ab, 2025-06-14:
What camera? The wyze V3 are the only cameras I use it on. i/o timeout errors are prob network issues, some recommend a timeout of 30 but that just masks the problems.

duggasco, 2025-06-14:
> What camera? The wyze V3 are the only cameras I use it on. i/o timeout errors are prob network issues, some recommend a timeout of 30 but that just masks the problems.

DC5L which is a T31 - the i/o timeouts were on the go2rtc side on the camera itself... so I think it might be an issue on prudynt's end

alan5ab, 2025-06-14:
Does that cam have 128mb of memory? What were the original issues?

duggasco, 2025-06-14:
> Does that cam have 128mb of memory? What were the original issues?

64mb - looks like the rmem needed an adjustment for prudynt to stop crashing.  will forgo go2rtc on cam as I've got even less osmem to spare now!

alan5ab, 2025-06-14:
It doesn't work on the 64mb cameras, added a note to the wiki.

lcwsoft, 2025-06-21:
Disabling day/night cron jobs on the camera helped in my case. I then adjusted the quality of the main and substreams as follows, as my wifi isn't great (and the live view stuttered if I used higher values):

- main stream: 1280x720, 1500 bitrate, 15 fps
- sub stream: 640x360, 500 bitrate, 15 fps

davidebeatrici, 2025-07-29:
In my case the issue turned out to be a defective SFP+ DAC cable that connected the server to the switch. Steady 10 Gb/s in one direction and unstable 10 Mb/s in the other, sometimes reaching 0 (!!!) with iperf.

It's been a month since I replaced the cable and Thingino has been rock solid since then, confirming it was in fact purely a network issue. I apologize for not realizing the cause right away, the fact that other services running on the same server seemingly worked just fine tricked me!

m4mmon, 2025-08-02:
To give an update on my case.
I kind of have solved the issue. I was using, on both ch0 and ch1 streams, OSD. No logo, no uptime. Only time, and user text. Those last days, I noticed that if I disabled user text, the stream would not crash. My user text was not "%hostname" but something else  directly set in prudynt.cfg. The text was "CAM1" and so on for the various cams. Also, setting back the user text to "%hostname" and setting it on again does not crash either.
I did not notice anything in OSD.cpp that would explain that, but maybe there is, elsewhere, some out of bounds write or something, that has visible consequences (i.e. process crashing) when the conditions I described above are met.
This goes way beyond what I can try to debug.

lifeishard, 2025-08-06:
My V2 crashes regularly also, I disabled the day/night cron and it is a little bit more stable, but still can't stay up for more than 2 days. Got any tips to help debug?

lifeishard, 2025-08-08:
> Well, I have the same issues, stream is being reconstructed after a few seconds.
>Aug 7 10:21:27 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:0 Aug 7 10:21:27 prudynt[1809]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. Aug 7 10:21:27 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:0 Aug 7 10:21:27 prudynt[1809]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000 Aug 7 10:22:00 crond[1682]: USER root pid 4758 cmd daynight Aug 7 10:23:00 crond[1682]: USER root pid 4781 cmd daynight Aug 7 10:23:03 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:0 Aug 7 10:23:03 prudynt[1809]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. Aug 7 10:23:03 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:0 Aug 7 10:23:03 prudynt[1809]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000 Aug 7 10:24:00 crond[1682]: USER root pid 4804 cmd daynight Aug 7 10:24:23 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:0 Aug 7 10:24:23 prudynt[1809]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. Aug 7 10:24:23 prudynt[1809]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:0

Is the stream being reconstructed normal behaviour? I just assumed it was normal for me.

Rococo88, 2025-08-08:
I don't think so, sink clogged message does not look like typical stuff, it happens less frequently when I lower bitrate and fps. I had the same issue with t23n based camera but gave up further troubleshooting.

lifeishard, 2025-08-24:
> To give an update on my case. I kind of have solved the issue. I was using, on both ch0 and ch1 streams, OSD. No logo, no uptime. Only time, and user text. Those last days, I noticed that if I disabled user text, the stream would not crash. My user text was not "%hostname" but something else directly set in prudynt.cfg. The text was "CAM1" and so on for the various cams. Also, setting back the user text to "%hostname" and setting it on again does not crash either. I did not notice anything in OSD.cpp that would explain that, but maybe there is, elsewhere, some out of bounds write or something, that has visible consequences (i.e. process crashing) when the conditions I described above are met. This goes way beyond what I can try to debug.
> 
> edit: actually, I tried to dig into it, and, in my use case, the crashes seem to have vanished. See [https://github.com/gtxaspec/prudynt-t/pull/86](url)

I had frequent crashes, like usually every day or 2 days max, but after I disabled the user text from the gui menu the crashes stopped and the camera was stable for weeks. There is likely something odd with the user text.

lcwsoft, 2025-10-19:
For anyone still having issues with Frigate and random stream failures on the latest Thingino stable release (2025-10-18): if you are using go2rtc in your Frigate config with just plain RTSP, try switching it to a "ffmpeg" source like this:

```
go2rtc:
  streams:
    camera1:
      - ffmpeg:rtsp://{RTSP_USER}:{RTSP_PASSWORD}@192.168.1.121:554/ch0

cameras:
  camera1:
    enabled: true
    ffmpeg:
      inputs:
        - path: rtsp://127.0.0.1:8554/camera1
          input_args: preset-rtsp-restream-low-latency
          roles:
            - record
```
Testing it locally now and it seems to be helping. Previously the ffmpeg processes used by Frigate would crash every few minutes and take the streams offline as a result. After making that change, its been 30 minutes so far and not a single error being reported.

I tried everything recommended by Frigate devs, including increasing the shm size, to no avail. This is the first change I've made that's shown promise.

Anaerin, 2026-02-27:
It seems that the Wifi connection and driver built-in to the Wyze cams uses SDIO to connect to the CPU, which isn't the best in terms of bandwidth, and Thingino doesn't appear to have any extra buffering or anything, causing dropouts. I've seen recommendations to use PoE ethernet adapters over USB, or even just a USB Wifi adapter instead, but that seems less than optimal to me at least.

olivluca, 2026-03-25:
I think, in my case, the problem is the backchannel: if I don't negotiate a backchannel prudynt doesn't crash (or at least not that often), but with the backchannel on it crashes after a few connections. In fact my log in the comment above complains about the BackchannelSink.

gtxaspec, 2026-05-17:
anyone try master branch with the new raptor streamer with frigate? note that the streamer has not been integrated into the UI, so UI config won't work.

themactep, 2026-08-19:
Triaging this one. Unlike the bandwidth-driven stream issues (cf. #678), this report has a genuine crash signature: `do_page_fault() #2: sending SIGSEGV to prudynt` with `ra` inside libimp.so/libc.so — an encoder-path defect in the streamer as it stood in April–May 2025 (`master+55a3254`).

Since then the streamer has been fully rewritten (prudynt-t on the ciao branch): new encoder pipeline with per-channel workers and buffer pools, plus targeted fixes in exactly this area — an encoder watchdog that force-cycles on consecutive polling timeouts, DMA-buffer fixes for low-RAM T31 devices, T31+ bitrate control, jitter-buffer pacing, and a batch of teardown/thread-race fixes.

Can anyone still reproduce stream crashes / SIGSEGV on a current ciao build? If it's gone, this can be closed; if it still crashes, please reopen with the build hash, camera model, encoder settings, and the dmesg excerpt — it would be actionable. Leaving open pending verification.

davidebeatrici, 2026-09-07:
`ciao+2c3eca5` seemed to work perfectly on Vanhua Z55 cameras, until one's stream randomly stopped working:

```
[152648.901692] gc4653 stream off
[152649.078052] codec_codec_ctl: set CODEC_TURN_OFF...
[152649.271728] codec_codec_ctl: set CODEC_TURN_OFF...
```

```
D/Encoder(  649): 20:00:13.354 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=0
D/Encoder(  649): 20:00:13.355 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=0
E/Encoder(  649): 20:00:13.371 [EndEncoding] /home/user/git/proj/sdk-lv3/src/imp/video/alcodec/lib_encode/Com_Encoder.c(1565):errorCode = 88, eProfile = 4000000, uEncWidth = 2560, uEncHeight = 1440
```

Seems like it can be reproduced somewhat reliably by toggling between day/night mode manually:

```
D/Encoder(29662): 03:17:46.469 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=1
D/Encoder(29662): 03:17:46.828 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=1
D/Encoder(29662): 03:18:35.866 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=0
D/Encoder(29662): 03:18:36.028 [do_day_night_change] do_day_night_change(664):enc_chn->inStat.is_day=0
E/Encoder(29662): 03:18:54.643 [EndEncoding] /home/user/git/proj/sdk-lv3/src/imp/video/alcodec/lib_encode/Com_Encoder.c(1565):errorCode = 88, eProfile = 4000000, uEncWidth = 2560, uEncHeight = 1440
```

Another one has had random green artifacts at the bottom of the image for a long time now, but I suspect it's a hardware problem:

[image]