themactep/thingino-firmware #642: wyze vdb doorbell_ctrl segfault

davlaw, 2025-05-04:
root@ing-wyze-vdb1-16be ~# doorbell_ctrl -p 77-C2-88-95
Serial port configured to 115200 baud, raw mode.
Pairing command sent to /dev/ttyS0
Segmentation fault

os-release
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
TOOLCHAIN=musl
SOC=t31
SOC_ARCH=xburst1
IMAGE_ID=wyze_vdb1_t31x_sc4236_rtl8189ftv
BUILD_ID="master+70d7cb9, 2025-05-04 09:23:54 +0000"
BUILD_TIME="2025-05-04 09:23:54 +0000"
COMMIT_ID="master+70d7cb9, 2025-05-04 01:55:11 +0000"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
HOSTNAME=ing-wyze-vdb1
TIME_STAMP=1742234000
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-ge9be0d14
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.05-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.05-git"

gtxaspec, 2025-05-04:
Thx I'll take a look

davlaw, 2025-05-07:
would have been nice to point out my error...

gtxaspec, 2025-05-07:
i verified that the program does segfault when characters other than colon's are used.  this will be fixed in the next doorbell_ctl release.

closed 2025-05-11 by davlaw

gtxaspec, 2025-05-11:
fixed in
https://github.com/themactep/thingino-firmware/commit/1b8e2fd3a295eab4fd9fd087fbde0f5b15383311