themactep/thingino-firmware #643: Support for: Besder A22YQQ 4-lens (rebranded vstarcam)

GrumpyMeow, 2025-05-06:
Shop link: https://nl.aliexpress.com/item/1005007948660872.html
Serial number/DeviceID/UID: VE5237xxxRXRD / VSTH3781xxxWVLKX
Firmware: `10.53.120.18`
App version: `ZH161.8.53.31`
App: O-kam pro
PTZ: Yes on main+zoom camera
Onvif: Yes, only PTZ main+zoom camera streams. No lighting control. 
PPPP: seed: `vstarcam2018`, hash: `[0x2c,0xd4,0x60,0x06]`
Telnet: (activation needed) `vstarcam2017` password: `20170912`
CPU: `T23-N`
Sensors: `gc2083` (1920x1080)
uname -a: `Linux veepai 3.10.14__isvp_pike_1.0__ #192 PREEMPT Wed May 22 15:19:04 CST 2024 mips GNU/Linux`
cat /proc/mtd:
```
dev:    size   erasesize  name
mtd0: 00040000 00008000 "boot"
mtd1: 00158000 00008000 "kernel"
mtd2: 00360000 00008000 "root"
mtd3: 00298000 00008000 "system"
mtd4: 00030000 00008000 "www"
mtd5: 00040000 00008000 "param"
mtd6: 00800000 00008000 "all"
```
cat /proc/version:
```
Linux version 3.10.14__isvp_pike_1.0__ (fanwangxing@vstarcam) (gcc version 5.4.0 (Ingenic Ingenic r3.3.0-gcc540 Smaller Size 2023.05-22) ) #192 PREEMPT Wed May 22 15:19:04 CST 2024
```
lsmod:
```
exfat 94064 0 - Live 0xc05ef000 (O)
motor_spi 16688 0 - Live 0xc05c8000 (O)
spi_device 720 0 - Live 0xc05bb000 (O)
spi_gpio 4352 0 - Live 0xc05b4000
sensor_gc2083s1_t23_double 10848 0 - Live 0xc04be000 (O)
sensor_gc2083_t23_double 10448 0 - Live 0xc04b1000 (O)
tx_isp_t23_double 856768 2 sensor_gc2083s1_t23_double,sensor_gc2083_t23_double, Live 0xc03d5000 (O)
sinfo 7744 0 - Live 0xc02d0000 (O)
audio 39248 2 - Live 0xc02b9000 (O)
aic8800_fdrv 324960 0 - Live 0xc01c1000 (O)
aic_load_fw 63392 1 aic8800_fdrv, Live 0xc012b000 (O)
logdev 526112 6 - Live 0xc008c000 (O)
```

gtxaspec, 2025-05-06:
4 lens? does this have 4 different sensors? do you have photos of the internals?

GrumpyMeow, 2025-05-06:
Sorry.. No photo's (yet) of the internals. I didn't want to break the device right away.

GrumpyMeow, 2025-05-08:
I think my camera might be somewhat the same as the O-Kam QC3. As i understand it's important to get the GPIO correct. I don't think the GPIO pins are an exact match.

I start out with the config of the QC3 and will replace the questionmarks if/when i find a matching GPIO for my cam.. So work-in-progress...
```
gpio_ircut=57 58     =>  ?
gpio_ir850=61     =>  ?
gpio_mmc_cd=59     =>  ?
gpio_white=50     =>  ?
gpio_wlan=62o     =>  ?
```

GrumpyMeow, 2025-05-09:
I've started to try and flash thingino to my camera, as expected unsuccessfully.
I had to decompile tfUpgrade. I found `CheckTfHaveUpgradeFile`  which tries to find files and determines which type of upgrade it will be. It does this by setting bit-flags which (i think) maps to mtd partitions:
```
u-boot-with-spl   => 0x01      => mtd0
uImage => 0x02   =>  mtd1
rootfs_64k_T23 => 0x04  => mtd2
jz_system => 0x08  => mtd3
jz_www => 0x10  => mtd4
IN-sys- => 0x20   => mtd5
JZ_FLASH_ => 0x100  => mt6    (all!)
```
So i think that would mean i would have to put the thingino firmware in the root called `JZ_FLASH_`.

GrumpyMeow, 2025-05-10:
Main cpu: Ingenic T23 4803221910613A20-N 191016824GB709 (cooled via glued on metal plate)
Uart pads:  RXD, TXD, RST, GND (next to CPU)
Wifi board: AICB800Dx40L 2433-3 TRRR77.00 FA020009
[image]

[image]

GrumpyMeow, 2025-05-10:
Output via uart pads. Rebooting due to missing other camera's.
Finally i do see the modelnumber: `A22QQ`
```
<break>
<break>
<break>
<0xfe><0xff><0xff>

U-Boot SPL 2013.07 (Nov 03 2024 - 18:06:52)
Board info: T23N

apll_freq = 1404000000 
-pll_freq = 1200000000 
sdram init start
DDR clk rate 600000000
image entry point: 0x8000000 

U-Boot 2013.07 (Nov 03 2024 - 18:06:52)

Board: ISVP (Ingenic XBurst T23 SoC)
DRAM:  64 MiB
TOp of RAM usable for U-Boot at: 84000000
Reserving 444k for U-Boot at: 83f90000
Reserving 32784j<break>
for malloc() at: 81f8c000
Reserving 32 Bytes for Board Info at: 81f8bfe0
Reserving 124 Bytes for Global Data at: 81f8bf64
Reserving 128k for boot params() at: 81f6bf64
Stack Pointer at: 81f6bf48
Now rqnning an RAM - T-Boot at: 83f90000
<0xff>MMC:   msc: 0
the manufacturer 20
SF: Detecded XM25QH64C
<0x0c>
*** Warning - bad CBC, us)ng defa5lt environment

In:    serial
Out:   serial
Err:   serial
watchdog open!
<0xff><0xff><0xf7><0xff><0xfb><0xff><0xff><0xff><0xff><0xf7><0xff><0xff><0xff><0xf7><0xfb>===QC11-V1-T23N-2<0x10>240102===
Net:   <0xfd><0xf7>====>phy 0:0x937c-0x4024 found
SPEED:0, DUPLEX:0
Jz4775-9161
the mantfacturer 20
SF: Depected XM21QH64C

<0xff><0xff><0xff><0xff><0xff><0xfd><0xef><0xff><0xff><0xff><0xff><0xf7>Hit any key to stop autoboot:  1 <0xff><0xff><0xff><0xff><0xff><0xff><0xff><0xff><0xef><0x7f><0x08><0x08><0x08> 0 
the majufacdurer 20
SF: Detected XM25QH64C

--->probe spend 4 ms
<0xff><0xff>SF: 2097152 bytes @ 0x4000<0x10> Read: OK
--->raad spand 674 ms
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14__isvp_pikE_1.0_[
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1368450 Bytes = 0.3 MiB
   Load <0x01>ddress: 80010000
   Entry Point:  80334f90
   Verifying Checksum ... OK
   UnckmpRes3)ng Kernel Image ... OK

Starting iernel ...

<0xff><0xff><0xf7><0xff><0xfb><0xff><0xdf><0xff><0x7f><0xff><0xfb>mder is ok.....,<0x08><0xf7><0xff><0xf7>open sfc fd error 
<0x1b>[1;36;40m[D][1970/01/01/08:00:00][tfUpgrade.c<KernelCheckVersIon<0x1e>:590][    0.044031<0x1d> vp_kErnel_version:10.52.0.9
<0x1b>[0m<0x1b>[1;36;40m[D][1970/01/01/08:00:00][tfUpgrade.c<KernelCheckVersion>:598]CheckVersion8<break>
10.52.0.9
<0x1b>[0m<0x1b>[1;33;40m[W][1970/01/01/08:00:00][tfUpgrade.c<KernelCheckVersikn>:603]-------)--------------------------
[... 1889 lines cut ...]
          RX bytes:116007 (113.2 KiB)  TX bytes:116007 (113.2 KiB)

wlan0     Link encap:Ethernet  HWaddr 6C:05:D3:1C:A7:53  
          inet addr:192.168.1.116  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4462 errors:0 dropped:1871 overruns:0 frame:0
          TX packets:303 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:837387 (817.7 KiB)  TX bytes:14810 (14.4 KiB)

<0x1b>[1;31;40m[E][2025/05/10/09:03:28][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:29][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;36;40m[D][2025/05/10/09:03:30][mqtt.c:2036][Mqtt_Pth]mqtt wait internet connect...
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:30][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;32;40m[M][2025/05/10/09:03:30][cputotol.c:396][Cputotol_System_pth]MEM: used: 27 Mb, free: 12 Mb, CPU-Idle: 73%, SdTotol = 0 Mb, SdFree = 0 Mb
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:31][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:32][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;36;40m[D][2025/05/10/09:03:33][mqtt.c:2036][Mqtt_Pth]mqtt wait internet connect...
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:33][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;33;40m[W][2025/05/10/09:03:33][sensorManage.c:168][gotoWatchPreset]NoSet watchPreset
<0x1b>[0m<0x1b>[1;33;40m[W][2025/05/10/09:03:33][sensorManage.c:469][alarmTiming]Goto backPos
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:34][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:03:35][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m[   97.527783] Restarting system.
```

GrumpyMeow, 2025-05-10:
With SDcard and upgrade-file `IN-sys-`:
```
<0x1b>[0m<0x1b>[1;32;40m[M][2025/05/10/09:21:01][cgiapp.c:32][CgiAppInit]CgiUpdateNoteThread
<0x1b>[0mfunc = CgiUpdateNoteProc, thread tid   840
func = MemCheckProc, thread tid   839
<0x1b>[1;32;40m[M][2025/05/10/09:21:01][cgireboot.c:230][CgiRebootProc]func = CgiRebootProc, thread tid   836
<0x1b>[0m<0x1b>[1;31;40m[E][2025/05/10/09:21:01][gpio_v300.c:1148][LightModeWork]DoubleChip did not connect.
<0x1b>[0m<0x1b>[1;36;40m[D][2025/05/10/09:21:01][mqtt.c:2036][Mqtt_Pth]mqtt wait internet connect...
<0x1b>[0m<0x1b>[1;32;40m[M][2025/05/10/09:21:01][cgiupdate.c:2516][SD_CheckUpgradeFile]Have Upgrade File: [IN-sys-]
<0x1b>[0m<0x1b>[1;32;40m[M][2025/05/10/09:21:01][cgiupdate.c:2471][GetCurrentFwVersion]current fw version is [www.veepai.com-10.53.120.18
]
<0x1b>[0m<0x1b>[1;33;40m[W][2025/05/10/09:21:01][cgiupdate.c:2560][SD_CheckUpgradeFile]CheckVersion: make.version [<0xaa><0x02>], current.version [www.veepai.com-10.53.120.18
]
<0x1b>[0m<0x1b>[1;33;40m[W][2025/05/10/09:21:01][cgiupdate.c:2565][SD_CheckUpgradeFile]Need Upgreade: [IN-sys-]
<0x1b>[0m<0x1b>[1;36;40m[D][2025/05/10/09:21:01][cgiupdate.c:1480][setWebUpgradeFileMaxLen]upgradeFirmWareBufSize [13910016]
<0x1b>[0m[   35.999940] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.030517] FAT-fs (mmcblk0p1): Filesystem has been set read-only
[   36.055126] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.092751] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.138278] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.159429] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.169064] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.182151] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.192008] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.203214] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
[   36.212741] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 473269)
```

GrumpyMeow, 2025-05-10:
In the MTD5 i find these strings of interest:
```
vstarcam2017:khNnIDmF6UD8.:0:0:Administrator:/:/bin/sh
vstarcam2017:/89DNegAJszV2:0:0:Administrator:/:/bin/sh
vstarcam2017:QEfWyD9l/oIgo:0:0:Administrator:/:/bin/sh
vstarcam2017:GpPxnqQg9ZqJQ:0:0:Administrator:/:/bin/sh
vstarcam2017:AYnAqH6Wa0g7.:0:0:Administrator:/:/bin/sh
vstarcam2017:TC0hYAQ8Yxe.o:0:0:Administrator:/:/bin/sh
vstarcam2017:68a4cMmw.vjVw:0:0:Administrator:/:/bin/sh
vstarcam2017:K2rGr4q4uxQmk:0:0:Administrator:/:/bin/sh
vstarcam2017:/cHejlZYG9UAc:0:0:Administrator:/:/bin/sh
vstarcam2017:5PAqy5wr3MlHk:0:0:Administrator:/:/bin/sh
vstarcam2017:R4hSASVPOGxkY:0:0:Administrator:/:/bin/sh
vstarcam2017:b.btD.WQZJr5I:0:0:Administrator:/:/bin/sh
vstarcam2017:K.WJWS4962vZk:0:0:Administrator:/:/bin/sh
vstarcam2017:wgbDtJj5ehJYQ:0:0:Administrator:/:/bin/sh
motosit.bin
```

No idea why there would be that much entries.

GrumpyMeow, 2025-05-10:
So holding the button during powering-on and holding it, it seems to do an upgrade differently. Unfortunatly still only the mtd5 partition:
`UpgradeType [32]`
```
<0x1b>[1;31;40m[E][1970/01/01/08:00:11][tfUpgrade.c<main::1617MPifx[b:21] has preqs down
<0x1b>[0m<0x1b>[1;36;40m[D][19<0x17>0/01/01/08:<0x10>0:11][tfUpgrade.c<InitTFCard>:1576]mount [/dev/mmcblk0p1] /mnt/sda0 vfat
<0x1b>[0e<0xf7><0xff><0xff><0xff><0xff><0xff><0x7f><0xff><0xff><0xff><0x1b>[1;32;40m[M][1970/01/01/08:00:12][tfUp'rade.c<CheckTfHaveUpgradeFile>:731]/mnt/sda0/ TF tryTimes [1]
<0x1b>[0m<0xfb><0xf7><0xfb><0xff><0xff><0x1b>[1;32;40m[M][1970/01/<0x10>1/08:00:13][tfUpgrade.c<CheckTfHaveUpgradeFile>:748]Get [/mnt/sda0/IN-sys-]
<0x1b>[0m<0x1b>[1;32;40m[M][1970/01/01/08:00:<0x11>3][tfUpgrade.c<CheckTfHaveUpgradeFile>:794]UpgradeType [32]
<0x1b>[0m<0x1b>[1;32;40m[M][1970/01/01/08:00:13][tfUpgrade.c<main><0x1a>1666Ytfupgrade jz t23 8m version 3.0 20231219--UpgradeType [20]
<0x1b>[0mopen sfc fd error 
<0xf7><0xff><0xff><0xff><0xff><0xf7><0x7f><0xff><0xef><0xfb><0xff><0xfb><0xff><0xf7><0xff><0x7f><0xf7><0xff><0xff><0xff><0xff><0xff><0xff><0xfd><0xff><0x1b>[1;32;40m[M][1970/01/01/08:00:16][socjet.c<tcp_sarVice>:119]Accept 5s timeout.

<0x1b>[0m<0x1b>[1;32;40m[M][1970/01/01/08:00:16][sockat,c<tcp_service>:3<0x16>9]upgread tcp serv%p exit ...
<0x1b>[0m<0xff><0xf7><0xf7><0xff><0x7f><0xff><0xef><0x1b>[1;32;40m[M][1970/01/01/08:00:17][socket.c<cendWmsg>:39<0x18>] dev 0 Tipe 7   data 0 

<0x1b>[0m<0x1b>[1;32;40m[M][1970/01/01/08:00:17][socket.c<send_msg>2419]send 7 arr...sendWret -1 

<0x1b>[0m<0x1b>[1;31;40m[E][1970/01/01/08:00:17][tfUpgrade.c<main>:1676]upgrade ... UpgradeType [32]
<0x1b>[0m<0x1b>[1;36;40m[DM[1)70/01/01'08:00:17][tfUpgrade.c<setWebUpgradeFileMaxLen>:925]upgradeFirmWareBufSize [2<0x16>2144U
<0x1b>[0m<0x1b>[1;33;40m[W][116 /01/0<0x11>/ 8:00:17][tfUpgr[   17.092393] watchdog watchdog0: watchdog did not stop!
ade.c<UpgradeInfoRead>:846]/system/param/upgradeinfo.ini not exist
<0x1b>[0m<0x1b>[1;32;40m[M][1970/01/01/08:00:17][tfUpgr!de.c<upg2adE[firmware>21362]*fread fileSize[262144]
<0x1b>[0mbuf[0]
```

GrumpyMeow, 2025-05-10:
Same flash chip: https://discord.com/channels/1086012062649565286/1369052037303504966

endertable, 2025-05-18:
Reminds me alot of [this camera](https://www.temu.com/solar-surveillance-camera-g-601099575665611.html?refer_page_name=goods&refer_page_id=10032_1747548977309_8253cgdipd&refer_page_sn=10032), which I have 4 of them..  They are kind of nice because they have both a wide angle and zoom lens (although i think the 10x is a stretch, maybe more 5x or 6x).  I believe there is alot of Vstar  in these as well although it seems they may have secured them a little bit more.  I believe they have a T31 CPU.  I am more into Wyze which uses AWS, where as these straight up use chinese cloud endpoints.
I mostly have them offline, but like all cloud cams, they get dysfunctional when they can't reach their p2p endpoints.  :(

GrumpyMeow, 2025-05-18:
Welp.. By desoldering the flash chip, i took of the connection pads. Maybe the info on the issue is of some use to somebody..

closed 2025-05-18 by GrumpyMeow

gtxaspec, 2025-05-18:
You can try to repair by restoring the traces...any photos?

GrumpyMeow, 2025-05-18:
Nah.. I butchered them. Too impatience

GrumpyMeow, 2026-01-03:
Hi! 
I think I did share the instructions somewhere, probably on another issue.
I did it through cgi. I'll try to find the issue and link it here.

In this thread I described it:
https://github.com/themactep/thingino-firmware/issues/612#issuecomment-2850871809

Hopefully it works for you

vubranko, 2026-11-12:
Can you help me? I have a22qq camera. After update on icsee hi automaticly detected new sowtvare and update. But then camera stop working correctly.
It shows x3-we2q  V5.02.R02.000A06CD.10000 do you have firmware that I culd flash.