themactep/thingino-firmware #648: How to permanently set color mode/IR filter/IR LED?

m4mmon, 2025-05-16:
Hello,

I have disabled the Day/Night Mode Control, and would like the camera to always be in B&W, IR Filter "removed", and no IR LED.

Those settings made in the preview page of the webui do not survive when the streamer is restarted, or the camera rebooted.
The camera reverts back to Color Mode "color", IR filter in the light path.
From there, I have to toggle the "Color mode" button twice for the video to turn B&W, then disable the IR Filter.

I had a look at some files in /etc but have not guessed what to modify.

PS: many thanks for this very cool project :)

daoudeddy, 2025-05-17:
set the gain in the **Day/Night Mode Control** to a point that the script thinks is always at night and change the **Actions to perform** to the thing you want

m4mmon, 2025-05-17:
yes! so obvious 🤦‍♂️...
thank you

closed 2025-05-17 by m4mmon