themactep/thingino-firmware #650: Camera won't upgrade after Curl Sysupgrade update

micahp2, 2025-05-22:
I have the Wuuk 0510 running some pretty old firmware. I followed the instructions on another ticket that asked to update the sysupgrade script via Curl, so I did so. Unfortunately, that did not solve the problem, as I'm still unable to upgrade, either via the ssh command or via a newly flashed SD card.

The error I get from sysupgrade -f is as follows:

Check connection to https://github.com
Change directory to /tmp
Upgrade from GitHub
**Cannot get remote file size. Camera profile name may have changed. Please run gestalt and select the correct profile.**
Clean up temporary files.

Are there other steps that I am missing?

micahp2, 2025-05-22:
I found an inelegant workaround for anyone who finds this issue later:

1. Upgrade to a newer sysupgrade:  curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade && chmod +x /sbin/sysupgrade
2. Manually copy the image for your camera onto an SD card. In my case it was this one: https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wuuk_y0510_t31x_sc4336p_ssv6158.bin
3. Insert the SD card into the camera
4. Log into the camera using SSH and cd into the SD card: cd /mnt/mmcblk0p1
5. Manually run this: sysupgrade thingino-wuuk_y0510_t31x_sc4336p_ssv6158.bin
6. After a reboot, connect to the WiFi of the camera, do the WiFi setup and password bits, then wait for another reboot
7. Now the camera is updated

I couldn't find this set of steps anywhere, so it was trial and error to come up with them here. Hopefully this will help someone else if they run into this situation later.

closed 2025-05-22 by micahp2

gtxaspec, 2025-05-22:
for future reference, the proper procedure is to run the `gestalt` program as indicated in the syspugrade message, and select your model device.  then you can upgrade normally.

micahp2, 2025-05-22:
FWIW, I did try that but there was no gestalt program found anywhere on the OS build that I had.

gtxaspec, 2025-05-22:
ah, gotcha, good point, older builds don't have that.  it should download it along with the sysupgrade, will see about implementing this.

PyroghostX, 2025-06-09:
On version gc2053, can't auto update, looks like ill have to sd card each of them?

                                                 wyze_c3_t31x_rtl
                        master+a1f868e, 2025-01-20 19:30:19 -0800

                                       02:77:37:2b:2d:74 as wlan0
                                                       ::1 (IPv6)
                                             192.168.50.52 (IPv4)

               08:18:53 up 5 min,  load average: 1.85, 1.36, 0.61

root@wyzew-2d74-8dcc ~# curl -sL -o /sbin/sysupgrade curl -sL -o /sbin/sysupgrad
e https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/maste
r/overlay/lower/usr/sbin/sysupgrade && chmod +x /sbin/sysupgrade
root@wyzew-2d74-8dcc ~# sysupgrade -p
/sbin/sysupgrade: line 1: 404:: not found
root@wyzew-2d74-8dcc ~# sysupgrade -p
/sbin/sysupgrade: line 1: 404:: not found
root@wyzew-2d74-8dcc ~# gestalt
Gestalt selector - Fetching available profiles:
Error: gestalt list empty. Please try again later or check your network connection.
root@wyzew-2d74-8dcc ~#