themactep/thingino-firmware #656: Wyze Doorbell v1 main stream issue with substream enabled

Tediore, 2025-05-28:
When I enable the substream on the Wyze Doorbell v1 and have both the main stream and substream rotation set to 2 (to bring both in portrait orientation as designed), the main stream goes solid green in both the web preview and the RTSP stream. Both streams are set to 3:4 aspect ratios (1536x2048, 480x640). The substream functions correctly. OSD on/off for both streams makes no difference.

When the main stream rotation is set to 2 and the substream rotation is set to 0, the issue with the main stream goes away.

prudynt logs (same for both scenarios):
```
[INFO:Config.cpp]: Loaded configuration from /etc/prudynt.cfg
[INFO:main.cpp]: PRUDYNT-T Next-Gen Video Daemon: Mar 17 2025 17:53:20_e9be0d14
[INFO:main.cpp]: Starting Prudynt Video Server.
[INFO:IMPSystem.cpp]: LIBIMP Version IMP-1.1.6
[INFO:IMPSystem.cpp]: SYSUTILS Version: SYSUTILS-1.1.6
[INFO:IMPSystem.cpp]: CPU Information: T31-X
[INFO:IMPSystem.cpp]: Sensor: sc4236
---- FPGA board is ready ----
  Board UID : 30AB6E51
  Board HW ID : 72000460
  Board rev.  : 5DE5A975
  Board date  : 20190326
-----------------------------
[INFO:Opus.cpp]: Encoder bitrate: 40000
[INFO:WS.cpp]: Server started on port 8089
warn: shm_init,53shm init already
[INFO:IMPAudio.cpp]: Audio In: format:OPUS, vol:80, gain:25, samplerate:16000, bitwidth:16, soundmode:1, frmNum:30, numPerFrm:640, chnCnt:1, usrFrmDepth:30
set jpeg streamMngCtx suceess
[INFO:RTSP.cpp]: stream 0 available at: rtsp://10.0.4.71/ch0
[INFO:RTSP.cpp]: stream 1 available at: rtsp://10.0.4.71/ch1
```

Happy to share any other relevant logging.

Tediore, 2025-06-04:
Closing since upon further review, it seems like rotating both streams is probably too much for the device to handle. The load average is already pretty high with rotating only the main stream at 1152x1728.

gtxaspec, 2025-06-05:
digging into this further, for future reference, rotation is only supported on one channel at a time on T31.

Tediore, 2025-06-05:
Oh nice, good to know. Thank you

ELY3M, 2026-01-11:
I am trying to rotate the video.  but I looked in the logs.... 

Dec 11 16:43:26 : [INFO:Config.cpp]: Loaded configuration from /etc/prudynt.json
Dec 11 16:43:26 : [ERROR:Config.cpp]: invalid config value. stream0.rotation = 2
Dec 11 16:43:26 : [INFO:main.cpp]: PRUDYNT-T Video Daemon: prudynt-t 08d71521d3 (Dec 11 2025 22:43:20)

I looked through the wiki.  What are valid rotations for this doorbell v1?