themactep/thingino-firmware #668: WebSocket error in OBS browser

TurboLed, 2025-06-05:
OBS browser gives the following error:
Uncaught DOMException: Failed to construct 'WebSocket': The URL '//192.168.0.27:8089?token=***' is invalid.
    at http://***@192.168.0.27/x/preview.cgi:199:10

Fixed by adding ws:// prefix in the WebSocket constructor.

https://github.com/TurboLed/thingino-firmware/pull/1

gtxaspec, 2025-06-09:
try updating to OBS 31.0.3, works for me now without any changes.

jspiros, 2025-06-13:
> we should probably add a toggle to restore compatibility, or have it as a dev option somewhere, i like this idea.

Not sure there's any reason for it to be a toggle. Since the webui already conditionally determines the port in the URL based on whether HTTPS is in use, the URL's protocol can be determined the same way, as I implemented in the pull request I just submitted (#672). The main reason support for protocol-less URLs was added to browsers was to allow the browser to figure that out, it has no other benefit that I'm aware of.

themactep, 2025-06-16:
fixed in #672