themactep/thingino-firmware #676: Backquotes in thingino.config breaks completely the UI

potens1, 2025-06-21:
If a backquote appears in the `thingino.config` file, it breaks completely the UI whith "ERR_EMPTY_RESPONSE"

i.e:

```
telegrambot_description_0="Start conversation`"

```
or
```
mqtt_password="asdWA257E0%Fw*+rd9&MT6*`ASDFqweerasdF"
```

To replicate just edit the thingino.config file and add a backquote in some field, the UI breaks instantly, remove the backquote, it works again (no need to reboot)

PS: the usecase is usually it can be in passwords, not texts, but it appears that if it is in the file, anywhere, the result is the same, but I did not found anything in the logs (through ssh, since the UI was broken)

themactep, 2025-09-06:
Prudynt config format has changed in favor of json. Please check if the issue is still actual.

potens1, 2025-10-01:
I did not had the time to check, I will try this weekend. That said the bug what more pernicious since the config file was "sourced" so, a lot could happen there. Now I guess this is not the case anymore (I guess) so it should solve it (but I will try to check soon)