themactep/thingino-firmware #678: Flashed 3 Different Wyze V3 Cameras (t31x t31al and t31zx) RSTP stream shows broken camera image

zenterprisecloud, 2025-06-22:
So I have 3 different versions of the Wyze V3 Cameras 
2 of them are the standard v3
1 of them is the floodlight v1

1 of the standard V3s have been running with the latest build with no issues (t31al)
1 of the standard V3s after about 10 minutes stream dies but im able to ping the device and enter the web portal with no issues able to ping and everything. (t31zx)
Same with the floodlight v1, stream randomly dies but everything else is working as it should. (t31x)

I have a dedicated AP just for these cameras (They're like less than 3 feet away from the AP)

What should I be looking for to get the stream to be properly working? Rebooting solves it but stream dies again after a couple minutes.

All have been recently flashed over again with the latest build so not sure what else could be done.

Thank you!

potat0man, 2025-06-24:
I have same issue with Wyze v3, made automation in Home Assistant if camera has ping but offline in Frigate, restart streamer with shell command: curl -k "http://root:pass@ip/x/config-streamer.cgi?do=restart"

I think part of the issue is reception, it was happening on my furthest camera very often, it had -65db reception, then did external antena mod, now it has -55db reception and strem breaks about once a day.

Now my cameras show 3 days uptime, but automation to restart stream triggered 5 times.

Another Issue is smeared image, like half of the image becomes grey or pink, tried playing with GOP, disabling audio and all suggested things in discord but it still happens.

potat0man, 2025-06-27:
> What's the streamer/OSD settings set to?

My setting was like yours but 25fps and 3000bitrate, I tried many encoding/gop settings, and it always happend so I got back to default, but after seeing your post I realized my bitrate was default or lower, today I set bitrate 10000 on 3 cameras and it didn't happen for 8 hours now.

gtxaspec, 2025-06-28:
visual artifacts are typical of bandwidth issues, keep in mind that stock firmware had the bitrate set to _under_ 1000kbps, while thingino sets it to 3mbps as the default, though, theres always the possibility of bugs, so more reports and information are helpful.  I have a batch (10) of t31al/t31x wyze v3's, and have no issues here locally on the latest builds.

potat0man, 2025-06-28:
maybe it's too early but 24 hours after setting 10000 bitrate seems like image glitches 90% disappeared, also cameras did not disconnect from frigate since

prudynt.cfg shows bitrate = 10000; but in frigate, cameras still show about 300 MiB / hour

themactep, 2025-07-24:
Smear happens when a large key frame is failed to deliver while smaller P-frames keep updating the image changes. It can be anything from networking to wrong mode and bandwidth issues. Set gop to 30 for more frequent key frame generation.

themactep, 2026-08-19:
Closing as stale. This was filed in June 2025 against the old prudynt generation and the thread converged on a bandwidth/bitrate/gop diagnosis rather than a streamer defect (key-frame delivery on weak links, per the July 2025 guidance: gop 30, bitrate tuning).

The streamer has since been fully rewritten (prudynt-t on the ciao branch — new config/JSON API, per-channel encoder workers, adaptive bitrate) and T31+ bitrate control was just fixed again in 8a5e497. Multiple t31x cameras on current builds are in normal service here.

If anyone still sees streams dying or smearing on a current ciao build, please open a fresh issue with the build hash, camera model, encoder settings (bitrate/gop/fps), and network link quality (RSSI).

closed 2026-08-19 by themactep