themactep/thingino-firmware #697: Add Wyze 4 Cams

richardhughes260, 2025-07-17:
This is a copy of the e-mail I sent you yesterday.

I have been thinking that you need to meet and collaborate.

https://thingino.com/ is providing open source firmware for commercially available cameras.

The Flashback Team is reading firmware from chips.

https://www.youtube.com/watch?v=nruUuDalNR0

I asked if the Wyze 4 cameras were on the roadmap for open source firmware & Paul said he needed the encryption keys they use. 

Can Flashback Team extract the encryption keys?

Have a great day gentlemen!

-Rich

wltechblog, 2025-07-17:
The signing key we need is not in the firmware. Extraction is simple enough, we do firmware extraction on every new device we support.

gtxaspec, 2025-07-18:
to be precise, we need the key that is used to sign the u-boot SPL.  The chip validates the SPL against the public key embedded in the chip.