themactep/thingino-firmware #703: Problem with motion detector on Xiaomi MJSXJ03HL

vladimirdyskin, 2025-07-26:
Hi everyone! I have Xiaomi MJSXJ03HL, i try flash many version firmware thingino, but on all version i get error  prudynt[2008]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1 and after that cam freeze and reboots. Maybe someone has encountered this situation?

vladimirdyskin, 2025-07-26:
Jul 26 17:27:00 192.168.68.37 crond[1881]: USER root pid 3164 cmd daynight
Jul 26 17:27:03 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:27:08 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:27:08 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:27:09 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Jul 26 17:27:17 192.168.68.37 prudynt[2008]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1
Jul 26 17:27:19 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:27:22 192.168.68.37 prudynt[2008]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1
Jul 26 17:27:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:27:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:27:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Jul 26 17:27:34 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:27:35 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:27:35 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:27:35 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Jul 26 17:27:39 192.168.68.37 prudynt[2008]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1
Jul 26 17:27:46 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:27:48 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:27:48 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:27:49 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Jul 26 17:27:56 192.168.68.37 prudynt[2008]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1
Jul 26 17:27:59 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:28:00 192.168.68.37 crond[1881]: USER root pid 3198 cmd daynight
Jul 26 17:28:02 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:28:02 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:28:02 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Mar 17 17:53:31 192.168.68.37 odhcp6c[1739]: Failed to send RS (Address not available)
Mar 17 17:53:31 192.168.68.37 odhcp6c[1743]: (re)starting transaction on wlan0
Mar 17 17:53:31 192.168.68.37 odhcp6c[1743]: Starting SOLICIT transaction (timeout 4294967295s, max rc 0)
Mar 17 17:53:31 192.168.68.37 odhcp6c[1743]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Mar 17 17:53:31 192.168.68.37 crond[1881]: crond (busybox 1.37.0) started, log level 8
Mar 17 17:53:31 192.168.68.37 mdnsd[1938]: mdnsd 0.12 starting.
Mar 17 17:53:31 192.168.68.37 /etc/init.d/S94rc.local: Ciao
Mar 17 17:53:32 192.168.68.37 odhcp6c[1743]: Got a valid ADVERTISE after 1089ms
Mar 17 17:53:32 192.168.68.37 odhcp6c[1743]: IA_NA 0001 T1 43200 T2 69120
Mar 17 17:53:32 192.168.68.37 odhcp6c[1743]: 2a0d:6fc0:2328:bb00::100d preferred 86400 valid 86400
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: Starting REQUEST transaction (timeout 4294967295s, max rc 10)
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: Send REQUEST message (elapsed 0ms, rc 0)
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: Got a valid REPLY after 12ms
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: IA_NA 0001 T1 43200 T2 69120
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: 2a0d:6fc0:2328:bb00::100d preferred 86400 valid 86400
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: T1 43200s, T2 69120s, T3 86400s
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: entering stateful-mode on wlan0
Jul 26 17:29:05 192.168.68.37 odhcp6c[1743]: Starting <POLL> transaction (timeout 43200s, max rc 0)
Jul 26 17:29:06 192.168.68.37 : [INFO:Config.cpp]: Loaded configuration from /etc/prudynt.cfg
Jul 26 17:29:06 192.168.68.37 : [INFO:main.cpp]: PRUDYNT-T Next-Gen Video Daemon: Mar 17 2025 17:53:20_e9be0d14
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [DEBUG:Logger.cpp]: Logger Init.
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [INFO:main.cpp]: Starting Prudynt Video Server.
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [DEBUG:IMPSystem.cpp]: IMPSystem::init()
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [DEBUG:IMPSystem.cpp]: IMP_OSD_SetPoolSize(1048576)
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [INFO:IMPSystem.cpp]: LIBIMP Version IMP-1.1.6
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [INFO:IMPSystem.cpp]: SYSUTILS Version: SYSUTILS-1.1.6
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [INFO:IMPSystem.cpp]: CPU Information: T31-L
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [DEBUG:IMPSystem.cpp]: IMP_ISP_Open() = 0
Jul 26 17:29:06 192.168.68.37 prudynt[2008]: [INFO:IMPSystem.cpp]: Sensor: jxq03p
[... cut ...]
Jul 26 17:29:22 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video destructed, encoder channel:1
Jul 26 17:29:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPServerMediaSubsession.cpp]: Create Stream Source. 
Jul 26 17:29:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPDeviceSource.cpp]: IMPDeviceSource video constructed, encoder channel:1
Jul 26 17:29:23 192.168.68.37 prudynt[2008]: [DEBUG:IMPAudioServerMediaSubsession.cpp]: createNewRTPSink: OPUS, 48000
Jul 26 17:29:32 192.168.68.37 crond[1881]: time disparity of 188616 minutes detected

Anacletus, 2026-01-17:
It happens to me also with newst firmware, camera is ok as long as I don't activate motion.  Many times (but not always), when I activate it, it  shows:
prudynt[1455]: [WARN:Motion.cpp]: IMP_IVS_PollingResult error: -1
then freezes and reboot

On top, free memory without motion activate shows 800-1200K free.  Maybe a low memory issue?

themactep, 2026-08-15:
Fixed — Motion.cpp now handles IMP_IVS_PollingResult error by logging and continuing instead of crashing. Retest on a fresh build; if it reappears, open a new ticket.

closed 2026-08-15 by themactep