themactep/thingino-firmware #706: firmware install on LSC 3215672

TomaszCwik, 2025-07-29:
Hi i install with success thingino-lsc_3215672_t23n_sc2331_atbm6012bx.bin on my LSC camera.
but after that i don't see communication between firmware and camera sensor.
Main view
[image]
logs:
root@ing-lsc-3215672-c572 ~# lsmod
Module                  Size  Used by    Tainted: G
atbm6012bx            557136  0
sensor_sc2331_t23      10224  0
tx_isp_t23            841440  1 sensor_sc2331_t23
gpio_userkeys            768  0
audio                  38816  1
exfat                  93488  1
jzmmc_v12              13328  0
root@ing-lsc-3215672-c572 ~# dmesg | grep -i sensor
root@ing-lsc-3215672-c572 ~# dmesg | grep -i isp
[    7.148997] cgu_isp can not set to 153000000, will change!!!
[    7.186355] @@@@ tx-isp-probe ok(version H20240204a), compiler date=Feb  4 2024 @@@@@
[   13.909904] cgu_isp can not set to 153000000, will change!!!
root@ing-lsc-3215672-c572 ~# ls /proc/jz/
audio     clock     codecs    ddr       debug     gpio      isp       mmc0      reset     watchdog
root@ing-lsc-3215672-c572 ~#

i cannot find any information about exact model of camera on board:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gtxaspec, 2025-07-29:
Run `sensor info`

TomaszCwik, 2025-07-29:
ChatGPT help me with some basic diagnostic (maybe it will help
root@ing-lsc-3215672-c572 ~# uname -a
Linux ing-lsc-3215672-c572 3.10.14__isvp_pike_1.0__ #1 PREEMPT Mon Mar 17 17:53:20 UTC 2025 mips GNU/Linuxissue
cat /etc/os-release 2>/d
root@ing-lsc-3215672-c572 ~# cat /proc/version
Linux version 3.10.14__isvp_pike_1.0__ (buildroot@buildroot) (buildroot-gcc-14.2.0) #1 PREEMPT Mon Mar 17 17:53:20 UTC 2025
root@ing-lsc-3215672-c572 ~# cat /etc/issue
Welcome to Thingino firmware!
root@ing-lsc-3215672-c572 ~# cat /etc/os-release 2>/dev/null
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
TOOLCHAIN=musl
SOC=t23
SOC_ARCH=xburst1
IMAGE_ID=lsc_3215672_t23n_sc2331_atbm6012bx
BUILD_ID="master+8d30d45, 2025-07-28 09:07:18 +0000"
BUILD_TIME="2025-07-28 09:07:18 +0000"
COMMIT_ID="master+8d30d45, 2025-07-25 09:53:47 +0000"
BOOTLOADER=isvp_t23n_sfcnor
HOSTNAME=ing-lsc-3215672
TIME_STAMP=1742234000
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-ge9be0d14
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.05-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.05-git"
root@ing-lsc-3215672-c572 ~#
root@ing-lsc-3215672-c572 ~# lsmod
/ingenic/
ls /usr/lib/modules/$(unModule                  Size  Used by    Tainted: G  ame -r)/extra/ 2>/dev/null
atbm6012bx            557136  0
sensor_sc2331_t23      10224  0
tx_isp_t23            841440  1 sensor_sc2331_t23
gpio_userkeys            768  0
audio                  38816  1
exfat                  93488  1
jzmmc_v12              13328  0

cat /root@ing-lsc-3215672-c572 ~# ls /usr/lib/modules/$(uname -r)/ingenic/
etc/modules
for f in /etc/modules.d/*; do echo "### $f"; cat "$f"; done
audio.ko              gpio-userkeys.ko      pwm_core.ko           pwm_hal.ko            sensor_sc2331_t23.ko  tx-isp-t23.ko
root@ing-lsc-3215672-c572 ~# ls /usr/lib/modules/$(uname -r)/extra/ 2>/dev/null
exfat.ko      hal_apollo    wireguard.ko
root@ing-lsc-3215672-c572 ~# cat /etc/modules
exfat
root@ing-lsc-3215672-c572 ~# for f in /etc/modules.d/*; do echo "### $f"; cat "$f"; done
### /etc/modules.d/audio
audio spk_gpio=63 spk_level=-1
### /etc/modules.d/gpio-userkeys
gpio-userkeys gpio_config="28,6,1"
### /etc/modules.d/isp
tx_isp_t23 direct_mode=0 isp_memopt=1 print_level=1
### /etc/modules.d/sensor
sensor_sc2331_t23
root@ing-lsc-3215672-c572 ~#
root@ing-lsc-3215672-c572 ~# dmesg > /mnt/dmesg.log
 | grep -i isp
cat /mnt/dmesg.log | grep -i sensor
cat /mnt/dmesg.log | grep -i atbm
[... 91 lines cut ...]
-sh: can't create /sys/class/gpio/gpio63/value: nonexistent directory
root@ing-lsc-3215672-c572 ~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@ing-lsc-3215672-c572 ~# LD_PRELOAD=/lib/libimp_control.so /bin/prudynt -- -d
[INFO:Config.cpp]: Loaded configuration from /etc/prudynt.cfg
[INFO:main.cpp]: PRUDYNT-T Next-Gen Video Daemon: Mar 17 2025 17:53:20_e9be0d14
[INFO:main.cpp]: Starting Prudynt Video Server.
[INFO:IMPSystem.cpp]: LIBIMP Version IMP-1.1.0
[INFO:IMPSystem.cpp]: SYSUTILS Version: SYSUTILS-1.1.0
LIBIMP_CONTROL Version: Mar 17 2025_17:53:20_ dlsym
[INFO:IMPSystem.cpp]: CPU Information: T23-N
[INFO:IMPSystem.cpp]: Sensor: gc2053
[ERROR:IMPSystem.cpp]: IMP_ISP_AddSensor(&sinfo) = -1
terminate called after throwing an instance of 'std::invalid_argument'
  what():  error initializing the imp system.
Aborted
root@ing-lsc-3215672-c572 ~#

gtxaspec, 2025-07-29:
do you have a dump of the stock firmware? also please share `thingino-diag` after running `sensor-info`

wltechblog, 2025-09-01:
If everything is working, we can close this issue