themactep/thingino-firmware #708: Wyze Cam Pan V1 installer

unomar, 2025-07-31:
While attempting to install thingino on a Wyze Cam Pan V1 I kept getting failures.  Looking at the logs I noticed the following lines:
```bash
Starting hardware check for filerename on Wyze V2 and V3
+ /sbin/soc -f
+ '[' t20 '=' t20 ]
+ '[' jxf22 '=' jxf23 ]
+ '[' jxf22 '=' jxf22 ]
+ '[' -f /sdcard/thingino-wyze_cam2_t20x_jxf22_rtl8189ftv.bin ]
+ '[' jxf22 '=' jxf22 ]
+ '[' -f /sdcard/thingino-wyze_campan1_t20x_jxf22_rtl8189etv.bin ]
+ echo 'Error: No matching firmware file found for t20, skipping rename.'
Error: No matching firmware file found for t20, skipping rename.
```

A quick search revealed that file is renamed to `autoupdate-full.bin` in the `do_wyze_cp1` function of make_installers.  I downloaded the file using the url from the aforementioned function and dropped it on my sdcard next to the renamed autoupdate-full.bin and the install succeeded.  I haven't dug into the actual code, but I suspect it should be looking for the renamed version.

Hopefully this is enough detail to figure out the issue and/or it helps someone else fix their install.

wltechblog, 2025-08-09:
This was my fault, I updated the Wyze installer to do online installs via uniflasher and hadn't updated the Pan cams yet. The fix is in place now.

unomar, 2025-08-13:
This appears to have fixed the installer issue.  Thanks!

closed 2025-08-13 by unomar