themactep/thingino-firmware #714: Wyze V3 do_page_fault crash

plat2on1, 2025-08-04:
I've discovered a(the?) cause of these crashes. I was having some construction done and had to remove two of my cameras temporarily. When I put them back up I switched their positions, and as it happens one of them was mounted upside-down. Now the camera that would crash multiple times per day is rock solid and the rock solid one is now crashing.

The cause is the vertical/horizontal flip. Effects both t31zx with atbm and t31x with rtl9189ftv.

I'm guessing this a bug in the ingenic SDK version thingino is using?

```
[ 3571.647734] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[ 3571.647734] 00000000 
[ 3571.647751] epc = 00000000 inra  = 7793373c in libimp.so[778f0000+107000]
[ 3571.647777] 
[ 3571.686282] gc2053 stream off
[ 3571.862418] codec_codec_ctl: set CODEC_TURN_OFF...

[  867.781795] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[  867.781795] 74702b90 
[  867.781811] epc = 76f6d6e0 in libimp.so[76f2a000+107000]
[  867.781834] ra  = 76f6d694 in libimp.so[76f2a000+107000]
[  867.781843] 
[  867.833114] gc2053 stream off

[  101.188524] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[  101.188524] 688ff334 
[  101.188540] epc = 77822bdc in libc.so[777b7000+b8000]
[  101.188564] ra  = 7776a634 in libimp.so[776a7000+107000]
[  101.188574] 
[  101.229426] gc2053 stream off
```

plat2on1, 2025-08-07:
I was able to get thingino to build with the 1.1.5 SDK and as of right now have 1 day 3 hours uptime on the flipped camera.

themactep, 2025-08-31:
just to clarify, you say that flipping the image makes the camera crash with invalid read access? i am trying to reproduce it on multiple cameras and so far no dice. did not try to flip though.

plat2on1, 2025-09-30:
I updated my cams to BUILD_ID="stable+851db40, 2025-09-28 09:11:38 +0000" and noticed some changes to the OSD so I renabled it and so far no crashes in 2 days.

themactep, 2025-10-04:
closing as resolved then

closed 2025-10-04 by themactep