themactep/thingino-firmware #724: "Flash new firmware image" requires internet connectivity

jsn-0, 2025-08-08:
My cameras are on a VLAN with no output internet access. I am unable to update my cameras locally due to the following limitation:
```
# /sbin/sysupgrade /tmp/fw-web.bin

Run as /sbin/sysupgrade /tmp/fw-web.bin
Downloading firmware into /tmp, there is enough free space.
Create /tmp/sysupgrade directory
Upgrade from a local file /tmp/sysupgrade/fw.bin
Found /tmp/fw-web.bin
'/tmp/fw-web.bin' -> '/tmp/sysupgrade/fw.bin'
Update self
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade to /tmp/sysupgrade/sysupgrade
curl: (7) Failed to connect to raw.githubusercontent.com port 443 after 30 ms: Error

chmod: /tmp/sysupgrade/sysupgrade: No such file or directory
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade-stage2 to /sbin/sysupgrade-stage2
curl: (7) Failed to connect to raw.githubusercontent.com port 443 after 12 ms: Error

Re-run script from /tmp/sysupgrade
/sbin/sysupgrade: exec: line 436: /tmp/sysupgrade/sysupgrade: not found
-sh: /sbin/sysupgrade /tmp/fw-web.bin: not found
# 
--- sysupgrade exit! ---
```

alan5ab, 2025-08-09:
Upgrade with an sd card or serve the file from a local server.

sstepansky, 2025-08-09:
Use `sysupgrade -x`

themactep, 2025-08-09:
not an issue, that's how it works.