themactep/thingino-firmware #726: [Tapo C110] Video stream has still image

abhas, 2025-08-11:
First of all, thank you for this project and for all the hard work.

Since the Tapo C110 is supported, I thought I would give it a try. The model I have is "**Tapo C110 IN/3.8**". I flashed a few-days old release binary. And then compiled a binary using the docker-based build system.

In all these attempts, however, the video stream appears to be frozen. It only shows a still image captured when the camera boots up. It does not show any further updated to the video stream.

I'm attaching a picture of the board, if that proves helpful. I would love to help debug and fix this issue.

[image]

sstepansky, 2025-08-12:
The output of `dmesg` is somehow truncated. Can you run `dmesg` and post the full output?

Are both main stream and substream frozen?
Have you tried playing the rtsp stream in a video player?

Have you tried changing the streamer settings? Different resolution, fps, bitrate...

Run these on your camera and monitor the output for errors:
```
killall prudynt
prudynt
```

themactep, 2025-08-16:
prudynt cannot start the stream. run `thingino-diag` and share the link.

sstepansky, 2025-09-18:
@abhas set buffers >= 2 if you want to use the sensor's native resolution.

paintballkev812, 2026-01-15:
I can confirm setting buffers = 2 resolves this issue.

half-duplex, 2026-04-20:
Manual buffers=2 is still necessary on `tplink_tapo_c110_t23n_sc3336_wq9001` `[stable+0400131, 2026-04-06 23:35:10 +0000]`. Diag in #1171

themactep, 2026-08-19:
Heads-up: automatic DMA buffer detection landed in prudynt and is now on the stable branch — this should remove the need for the manual `buffers=2` workaround on this camera.

The framesource now auto-calculates the buffer count (fps-scaled, capped at ~15% of total RAM) and enforces minimums:
- **T23 platforms (this camera, T23N): `min_bufs = 2`** — the T23 encoder requires double-buffering, previously the 15% RAM budget could collapse it down to 1 buffer, freezing the stream
- frames > 4MB (large native-resolution frames): also ≥2 buffers
- explicit `buffers` in the config still overrides the auto value

If you (or anyone with a Tapo C110) can verify on a current stable build with `buffers` left at its default, this can be closed. If it still freezes with defaults, that's a real remaining bug worth reopening with fresh diagnostics.