themactep/thingino-firmware #735: OnVIF never sends a close connection - Home Assistant Breaks

Semag, 2025-08-20:
Hey all-

The latest update to Home Assistant Broke Connections to Thingino. After doing some troubleshooting, it was found that Thingino doesn't send a Connection: Close.  See here:

_"The problem is the camera drops the connection after each post request without sending the Connection: close header so when the next request is attempted the connection has already been lost. We have a decorator for cameras that do this that needs to be applied in one more place. I am working on a patch.

It would be nice if someone reported the issue to Thingino so they can either support HTTP/1.1 keepalive or send Connection: close when the camera is going to drop the connection so the client doesn't have to guess what to do."_

https://github.com/home-assistant/core/issues/150735#issuecomment-3208084737

themactep, 2025-08-21:
@roleoroleo can you please take a look at that? Thank you.

themactep, 2025-09-17:
@roleoroleo please look at my fork https://github.com/themactep/thingino-onvif.
onvif server does not respond because it crashes a few times on the way to the response due to memory corruption. i fixed what i found but also refactored a lot of stuff while doing that, so it won't be easy to isolate changes for a pr. [here](https://github.com/themactep/thingino-onvif/commit/c4d90ef9fbfd959fdc25486e3842d2f062be3e84) is the breakpoint cleaning part, so you can get an idea what a jorney it was. please see what you can cherry-pick back to your repo.

roleoroleo, 2025-09-17:
Ok, I'll check it this weekend.
Thank you.

roleoroleo, 2025-09-21:
I tried again following your commits (https://github.com/themactep/thingino-onvif/commit/c4d90ef9fbfd959fdc25486e3842d2f062be3e84), with no luck.
- I removed all log features.
- I used static buffer for conf_file var.
- I used static buffer for input var.

Nothing changes: when I start the wizard with HA 2025.8.2, it doesn't work.
The behavior is the same posted above (wireshark dump). The cam replies to the GetCapabilities request and doesn't reply to the GetNetworkInterfaces request.
The packet arrives to the network interface, but http doesn't start device_service CGI.

I don't know what I could check.

themactep, 2025-09-22:
I will try to restore debugging breakpoints onto your latest version and reproduce the error locally. Did you intercept the request that does not get a reply? The raw xml payload.

roleoroleo, 2025-09-22:
Here it is.
The 1st is the GetCapabilities request, followed by the response.
Then the GetNetworkInterfaces request, without response.
Saved from a tcpdump capture executed on the cam.
The client is ha.

[Test.zip](https://github.com/user-attachments/files/22473840/Test.zip)

roleoroleo, 2025-09-25:
I'm sorry, but unfortunately I didn't fix the problem.
Even using your Docker container didn't work.
It's probably my fault, but I don't know what it is.
I solved the problem by updating HA to a later version. After updating, onvif_simple_server works with and without your changes.
Thanks anyway.

themactep, 2025-10-04:
should be fixed in the next release. https://github.com/themactep/thingino-onvif/commit/2a27e1238766162d831325177b88b8386ff58e70
please update tomorrow, test, and report back.

roleoroleo, 2025-10-04:
I can confirm that this patch works with my cam and HA 2025.8.2
Thanks.