themactep/thingino-firmware #738: send2telegram clip arrives empty

mateuszdrab, 2025-08-22:
Hi

I just deployed the latest master (`master+c918ad3, 2025-08-22 09:03:06 +0000`) onto my Wyze Cam v2 and configured the telegram bot and send2telegram. 

Seems that the snap functionality works well; however, clips arrive empty. They are unplayable files of about 7KB.

The vbuffer service is running and restarting it makes send2telegram throw the `Video buffer not found` error.

Not sure what else I can test as I'm not familiar with how the vbuffer service works.

themactep, 2025-08-24:
openRTSP is pain but that is the best we can offer for now

mateuszdrab, 2025-08-24:
Okay, at least it works with those adjustments.
Question remaining, how to make it respect the chosen framerate of the substream?
I reduced it to 5 fps and the openRTSP process runs with `-f 25` so the recordings are speeded up

mateuszdrab, 2025-08-24:
I know that, but I am trying to understand how this parameter is inferred and passed to openRTSP when the vbuffer service starts as it seems to be unaware of the frame rate changes in the webui.

themactep, 2025-08-24:
https://github.com/themactep/thingino-firmware/blob/master/package%2Fprudynt-t%2Ffiles%2FS96vbuffer

Check this out. Particularly `read_fps` function.

mateuszdrab, 2025-08-26:
Unfortunately, the log mentions 25 fps despite the substream being set to 5fps.

```
D/System( 1886): system_bind(): bind DST-Encoder-0(1.0.0) to SRC-Framesource-0(0.0.0)
I/Framesource( 1886): [chn0]: width = 1920 height = 1080
E/VBM( 1886): VBMCreatePool()-0: w=1920 h=1080 f=842094158
E/VBM( 1886): VBMCreatePool()-0: pool->config.fmt.fmt.pix.sizeimage=3110400 sizeimage=3133440
E/VBM( 1886): VBMCreatePool()-0: sizeimage=3133440
I/VBM( 1886): PoolId:0, frame=0x75075550, frame->priv=0x75075578, frame[0].virAddr=766eb000, frame[0].phyAddr=735a000
I/VBM( 1886): PoolId:0, frame=0x75075580, frame->priv=0x750755a8, frame[1].virAddr=769e8000, frame[1].phyAddr=7657000
E/Framesource( 1886): IMP_FrameSource_GetChnAttr(): chnAttr was not set yet
D/Encoder( 1886): channel-1 buffer malloc size=1049472 addr=0x74213290
I/Encoder( 1886): chn-1 NCU Info vaddr=75c46200 paddr=068d7200 size=117760 backv=75c62e00 backp=068f3e00 outv=75c9c600 outp=0692d600 total_size=588800
D/System( 1886): system_bind(): bind DST-Encoder-1(1.1.0) to SRC-Framesource-1(0.1.0)
I/Framesource( 1886): [chn1]: width = 640 height = 360
E/VBM( 1886): VBMCreatePool()-1: w=640 h=360 f=842094158
E/VBM( 1886): VBMCreatePool()-1: pool->config.fmt.fmt.pix.sizeimage=345600 sizeimage=353280
E/VBM( 1886): VBMCreatePool()-1: sizeimage=353280
I/VBM( 1886): PoolId:1, frame=0x75075930, frame->priv=0x75075958, frame[0].virAddr=76db8000, frame[0].phyAddr=7a27000
I/VBM( 1886): PoolId:1, frame=0x75075960, frame->priv=0x75075988, frame[1].virAddr=76e0e400, frame[1].phyAddr=7a7d400
D/Encoder( 1886): channel-2 buffer malloc size=3114624 addr=0x73cdf2d0
I/Encoder( 1886): framePriv->i_fps_num=25, framePriv->i_fps_den=1
D/IVS_MOVE( 1886): move->param.sense[0]=1
D/System( 1886): [ignored]read /proc/cpuinfo ret is NULLD/System( 1886): system_bind(): bind DST-IVS-0(3.0.0) to SRC-Framesource-1(0.1.1)
I/Encoder( 1886): framePriv->i_fps_num=25, framePriv->i_fps_den=1
```

themactep, 2025-08-26:
run `grep -B5 fps /etc/prudynt.cfg` on the camera, show the output.

themactep, 2026-08-15:
Closing — the vbuffer service this referenced no longer exists. The motion pipeline now records via /run/prudynt/mp4ctl and waits for the clip to finish writing before handing it to send2telegram (which re-checks size stability in copy_video). If empty clips reappear on a current build, open a new ticket.

closed 2026-08-15 by themactep